iNavFlight/inav #2358: iNav 1.7.3 Fishbowl on SPRacingF3EVO

VoicOfReason, 2017-10-17:
I'm having fishbowl effect on this setup (see title). It happened first a year ago on iNav with different frame, etc. Now, I took same FC and put on brand new frame and it's happening again. 6" props in both cases. Rest of specs:

Emax 2205-2400 redbottom
Littlebee 30a ESC's
Different GPS unit in both cases
Different Vtx
Different frames
Soft mounted motors and FC

I get the fishbowling when I switch to pos and altitude both on. It gets wider and wider and eventually leads to a crash.

teckel12, 2017-10-18:
@VoicOfReason I had this issue as well with a similar model and the same FC.  For me the issue was two-fold.

First, I did a proper magnetometer calibration in the field before launch (left stick up/right and right stick down/center).  Secondly, I changed the PIDs for GPS Navigation Position XY to: P=25, I=180, D=10.  This can also be done with the following via CLI:

```
set nav_mc_pos_xy_p = 25
set nav_mc_pos_xy_i = 180
set nav_mc_pos_xy_d = 10
```

Toilet bowl went totally away after this.  I had thought it was to do with the on-board magnatometer on the SP Racing F3 FCs (I had both the EVO and non-EVO with similar toilet bowl issues).  But, it was simply a field magnetometer calibration and PID tweak that fixed everything.  I'm not running F4 FCs with external magnetometers and the position hold is just as good as with the SP Racing F3 FC.

Hope this helps!

VoicOfReason, 2017-10-18:
Wow! I’m going to try tonight.

VoicOfReason, 2017-10-18:
We’re you running 6” props? Curious how well pid settings will translate
for me.

teckel12, 2017-10-18:
@VoicOfReason I was using 5" 3 blade props and a 220mm frame.   But the specs are similar (same motor, same ESC, etc.).  These PIDs would probably be very similar to yours.

VoicOfReason, 2017-10-18:
Well..... no luck. I changed the settings to what you said and it wouldn't even take off. The props sputtered and then I had to increase to full throttle to just get it off the ground. So, I changed the "p" setting to default and left "i" where it was and this caused so little power to control the craft that I crashed. Any other suggestions?

DzikuVx, 2017-10-18:
@VoicOfReason in 99% of cases, this is caused by faulty magnetometer, magnetometer calibration, magnetometer orientation or magnetometer too close to power cables.

teckel12, 2017-10-18:
@VoicOfReason Are you sure you were changing the GPS Navigation Position XY PIDs?  It sounds like you were changing the flight PIDs.  At launch, it's not in GPS navigation mode, so it should have zero effect till you engage position hold.

Really sounds like you changed the wrong PIDs to me.  To be clear, the PIDs to change are under GPS Navigation Position XY.

VoicOfReason, 2017-10-18:
I copied and pasted your values into the cli.

teckel12, 2017-10-18:
Very strange. Well, the proper field magnetometer calibration should do the trick.

VoicOfReason, 2017-10-18:
I tried that also and sometimes it seems like the quad hovers fine and
then I will change direction and it will start doing why the toilet bowls
So even if infield calibration works, it only works sporadically.

teckel12, 2017-10-18:
@VoicOfReason That's exactly what I experienced. Try to position hold 4 times with 90 degree turns between. I bet one will be really good, while the others not so good or even really bad. After field calibration of the gyros and magnatometer it's rock solid (well, not quite DJI rock solid). Also, adjusting the PIDs really worked well for me, it made it more stable for sure.

VoicOfReason, 2017-10-18:
So, you do the stick command, then rotate the craft all around and then it will beep again after 30 seconds to say it's done, or do you do the stick command again to end calibration?

VoicOfReason, 2017-10-18:
.... so I tried in field calib. again and it didn't change anything. Maybe it's time for a new and different FC, like the Omnibus F4.

teckel12, 2017-10-18:
You'll also likely need a new GPS module as well as most FCs don't include a magnatometer.

Did you do a **correct** INAV 6-point accelerometer calibration? Are you doing the magnatometer calibration away from any interference (I don't even have my TX or phone with me)?

Did you try to hover it in each direction (90 degrees apart)? If one direction is much better than the others, it's your calibration and not the sensor, balance, level, etc. If it's bad in all directions it could be you haven't calibrated level.

It's easy to blame the FC, but I had very similar issues as you and by simply doing all the calibrations and leveling correctly it worked perfectly.

BTW, my Omnibus F4 with external magnatometer is no more stable in position hold than my SP Racing F3.  I wouldn't give up yet, if you don't properly calibrate a new FC you'll have the same issue.

stronnag, 2017-10-18:
Are you using the onboard mag or an external one. Can you post a CLI `diff` please.

VoicOfReason, 2017-10-18:
Teckel12, it's not simply a matter of hovering level, but fishbowl which results in a crash. Yes, I did the 6 point calibration as well as the other calibrations. It does seem to hover level in some directions but shifting directions 90 degrees, e.g. to the south, I get fishbowl. Not sure what other calibrations I can do since I've done them all several times.

teckel12, 2017-10-18:
There's another potential issue, magnatometer orientation. I've seen Chinese knockoff SP Racing F3 boards using a knockoff magnatometer where the orientation is different than an actual SP Racing F3 board.

From the bench, can you point your quad N S E W and each stay there and are correct? You could have a knockoff with a flipped or rotated magnatometer. VERY bad things can happen if you try to engage position hold with the orientation wrong.

VoicOfReason, 2017-10-18:
I checked and the degrees are correct. Pointing East is 90 degrees, etc.

VoicOfReason, 2017-10-18:
If I got a different Flight controller, what's a good one that has a built in Mag?

stronnag, 2017-10-18:
There is none. Built in mags are the main source of 'toilet bowling'

VoicOfReason, 2017-10-18:
Ahhhh, ok. So, what GPS modules have a mag built in and how would I wire that to my SPRacingF3Evo, via SCL/SDA?

stronnag, 2017-10-18:
yes. The SPF3 EVO is a good board for this, as it has a rare on-board mag, so you can override it  software. It's one of my favourite boards for iNav:
````
# INAV/SPRACINGF3EVO 1.7.4 Oct 15 2017 / 11:38:59 (1ec29937)

### This is the GPS mag.
set mag_hardware = HMC5883
````

VoicOfReason, 2017-10-18:
Ok, that's the exact board I'm having trouble with... so you set your mag to HMC5883 and it works with that setting?

stronnag, 2017-10-18:
With the external mag in the GPS, yes. Perfect navigation functionality.

VoicOfReason, 2017-10-18:
OK, so which GPS do you use. I'm using the MN-880 right now. Do you know if that has a mag built in? If not, what do you use for GPS with built-in Mag?

teckel12, 2017-10-18:
None really. The industry has moved away from on-board magnatometers (for good reason).

Don't just check one direction, because if the sensor is flipped one way will work perfect.

What you're describing sounds exactly like what I experienced, and I finally worked it out. Posting your CLI `diff` may be helpful.

stronnag, 2017-10-18:
The one I mentioned two posts up :)

https://www.banggood.com/UBLOX-NEO-M8N-BN-880-Flight-Control-GPS-Module-Dual-Module-Compass-p-971082.html

VoicOfReason, 2017-10-18:
2017-10-18 @ 15:07:39 -- Running - OS: Windows, Chrome: 61.0.3163.100, Configurator: 1.7.3
2017-10-18 @ 15:07:45 -- Serial port successfully opened with ID: 86
2017-10-18 @ 15:07:46 -- MultiWii API version received - 2.0.0
2017-10-18 @ 15:07:47 -- Flight controller info, identifier: INAV, version: 1.7.3
2017-10-18 @ 15:07:47 -- Running firmware released on: Sep 9 2017 10:21:40
2017-10-18 @ 15:07:47 -- Board: SPEV, version: 0
2017-10-18 @ 15:07:48 -- Unique device ID received - 0x3100325533571220303834
2017-10-18 @ 15:08:08 -- EEPROM saved
2017-10-18 @ 15:08:08 -- Device - Rebooting
2017-10-18 @ 15:08:11 -- Unrecoverable failure of serial connection, disconnecting...
2017-10-18 @ 15:08:11 -- Serial port successfully closed
2017-10-18 @ 15:08:12 -- Serial port successfully opened with ID: 87
2017-10-18 @ 15:08:14 -- MultiWii API version received - 2.0.0
2017-10-18 @ 15:08:14 -- Flight controller info, identifier: INAV, version: 1.7.3
2017-10-18 @ 15:08:16 -- Running firmware released on: Sep 9 2017 10:21:40
2017-10-18 @ 15:08:16 -- Board: SPEV, version: 0
2017-10-18 @ 15:08:16 -- Unique device ID received - 0x3100325533571220303834
2017-10-18 @ 15:09:13 -- EEPROM saved
2017-10-18 @ 15:09:17 -- Unrecoverable failure of serial connection, disconnecting...
2017-10-18 @ 15:09:17 -- Serial port successfully closed
2017-10-18 @ 15:09:18 -- Serial port successfully opened with ID: 88
2017-10-18 @ 15:09:18 -- MultiWii API version received - 2.0.0
2017-10-18 @ 15:09:18 -- Flight controller info, identifier: INAV, version: 1.7.3
2017-10-18 @ 15:09:18 -- Running firmware released on: Sep 9 2017 10:21:40
2017-10-18 @ 15:09:18 -- Board: SPEV, version: 0
2017-10-18 @ 15:09:18 -- Unique device ID received - 0x3100325533571220303834
2017-10-18 @ 15:12:23 -- EEPROM saved
2017-10-18 @ 15:12:23 -- Device - Rebooting
2017-10-18 @ 15:12:26 -- Unrecoverable failure of serial connection, disconnecting...
2017-10-18 @ 15:12:26 -- Serial port successfully closed
2017-10-18 @ 15:12:27 -- Serial port successfully opened with ID: 89
2017-10-18 @ 15:12:27 -- MultiWii API version received - 2.0.0
2017-10-18 @ 15:12:29 -- Flight controller info, identifier: INAV, version: 1.7.3
2017-10-18 @ 15:12:29 -- Running firmware released on: Sep 9 2017 10:21:40
2017-10-18 @ 15:12:29 -- Board: SPEV, version: 0
2017-10-18 @ 15:12:29 -- Unique device ID received - 0x3100325533571220303834
2017-10-18 @ 15:21:03 -- EEPROM saved
2017-10-18 @ 15:21:03 -- Device - Rebooting
2017-10-18 @ 15:21:06 -- Unrecoverable failure of serial connection, disconnecting...
2017-10-18 @ 15:21:07 -- Serial port successfully closed
2017-10-18 @ 15:21:07 -- Serial port successfully opened with ID: 90
2017-10-18 @ 15:21:08 -- MultiWii API version received - 2.0.0
2017-10-18 @ 15:21:08 -- Flight controller info, identifier: INAV, version: 1.7.3
2017-10-18 @ 15:21:08 -- Running firmware released on: Sep 9 2017 10:21:40
2017-10-18 @ 15:21:09 -- Board: SPEV, version: 0
2017-10-18 @ 15:21:09 -- Unique device ID received - 0x3100325533571220303834
2017-10-18 @ 15:22:15 -- Unrecoverable failure of serial connection, disconnecting...
2017-10-18 @ 15:22:15 -- Serial port successfully closed
2017-10-18 @ 15:22:31 -- Serial port successfully opened with ID: 91
2017-10-18 @ 15:22:31 -- MultiWii API version received - 2.0.0
2017-10-18 @ 15:22:31 -- Flight controller info, identifier: INAV, version: 1.7.3
2017-10-18 @ 15:22:31 -- Running firmware released on: Sep 9 2017 10:21:40
2017-10-18 @ 15:22:31 -- Board: SPEV, version: 0
2017-10-18 @ 15:22:31 -- Unique device ID received - 0x3100325533571220303834
2017-10-18 @ 15:24:06 -- Accelerometer calibration started
2017-10-18 @ 15:24:08 -- Accelerometer calibration finished
2017-10-18 @ 15:24:19 -- Accelerometer calibration started
2017-10-18 @ 15:24:21 -- Accelerometer calibration finished
2017-10-18 @ 15:24:32 -- Accelerometer calibration started
2017-10-18 @ 15:24:34 -- Accelerometer calibration finished
2017-10-18 @ 15:24:40 -- Accelerometer calibration started
2017-10-18 @ 15:24:42 -- Accelerometer calibration finished
2017-10-18 @ 15:24:52 -- Accelerometer calibration started
2017-10-18 @ 15:24:54 -- Accelerometer calibration finished
2017-10-18 @ 15:25:06 -- Accelerometer calibration started
2017-10-18 @ 15:25:08 -- Accelerometer calibration finished
2017-10-18 @ 15:25:14 -- Magnetometer calibration started
2017-10-18 @ 15:25:31 -- Unrecoverable failure of serial connection, disconnecting...
2017-10-18 @ 15:25:31 -- Serial port successfully closed
2017-10-18 @ 15:25:32 -- Failed to open serial port
[... 176 lines cut ...]
set max_throttle = 2000
set min_command = 980
set motor_pwm_rate = 1000
set motor_pwm_protocol = ONESHOT125
set failsafe_procedure = RTH
set vbat_warning_cell_voltage = 10
set gps_provider = UBLOX7
set gps_sbas_mode = AUTO
set nav_mc_bank_angle = 45

# profile
profile 1

set mc_i_pitch = 35
set mc_i_roll = 35
set mc_i_yaw = 50
set mc_p_level = 70
set max_angle_inclination_rll = 600
set max_angle_inclination_pit = 600
set nav_mc_pos_xy_i = 160
set roll_rate = 40
set pitch_rate = 40
set yaw_rate = 40

#

VoicOfReason, 2017-10-18:
So on the BN-880, do I connect the SDA/SCL wires on the GPS to the I2C port?

teckel12, 2017-10-18:
BN-880 maybe? That model has a magnatometer. If it has both Tx/Rx and SCL/SDL pins it has a magnatometer.

VoicOfReason, 2017-10-18:
That's what I have, the BN-880. So, I have to connect the SCl/DSL pins to somewhere on the F3evo board. Where do I connect them?

stronnag, 2017-10-18:
To the I2C port on the EVO boad.

VoicOfReason, 2017-10-18:
Ahhhhhhh... ok. I'm going to do that. Currently I'm using onboard MAG on SPEvo board.

stronnag, 2017-10-18:
So here's a (low res) picture of my evo board. As you look at it, IIRC, the left hand populated socket is the I2C. Don't forget to recalibrate the new mag.

[image]

VoicOfReason, 2017-10-18:
Amazingly helpful. Thank you. Maybe this will solve a year old problem.

VoicOfReason, 2017-10-18:
So, related to this, I should be able to plug in a mn-880 into a OMNIBUS F7
V2 board and configure for external mag also?

W. Peter Freund
Adjunct Instructor - Philosophy
Northwest Arkansas Community College
----
“Hold no knowledge in contempt, blush to learn from no man, once you have
attained knowledge, look down on no-one.” (St. Victor)

teckel12, 2017-10-18:
A couple things bother me. 

set mag_hardware = AK8963

2017-10-18 @ 15:25:14 -- Magnetometer calibration started
2017-10-18 @ 15:25:31 -- Unrecoverable failure of serial connection, disconnecting...

Looks like the wrong magnatometer for the SP Racing F3 board? Also, looks like you powered off in the middle of calibration?

teckel12, 2017-10-18:
Just a heads up, you may not be able to use an external magnatometer with your SP Racing board if the on-board and external magnatometers use the same I2C ID (which is common).

VoicOfReason, 2017-10-18:
Yeah, I just tried to fly with the I2C plugged in and it crashed and may have broken the FC. OK, so I just want an iNav setup that works. Been trying for a year. What has been verified that actually works? Has an SP Racing F3 Evo with BN-880 been verified to work? Above, it sounded like using the BN-880's MAG is the way to go, but I just tried it, and it tried to fly off into the sunset.

teckel12, 2017-10-18:
The SP Racing F3 boards are discontinued and I believe that all are now clones so I would avoid. Also, an on-board mag is a bad idea, and you can't have 2 mags connected. So for several reasons I wouldn't do a SP Racing F3.

I'd suggest either an Omnibus F4 Pro or Omnibus F4 with baro option. But, only from a reputable seller (like https://www.readytoflyquads.com)  because there's lots of problems with the clones.

VoicOfReason, 2017-10-18:
How about an Omnibus F7 V2?

teckel12, 2017-10-19:
Just have no experience with the F7. I opted for the F4 because of some reason that escapes me. The F4 is more mature, and I questioned F7 compatibility, so that could be it. But I'm no expert on F7 support with INAV.

VoicOfReason, 2017-10-19:
Is calibrating an external compass/mag the same process as an internal one within iNav?

teckel12, 2017-10-19:
Yes, but you can't attach an external magnatometer if you have one on your FC. Also, you probably should select `Auto` for magnatometer so it can find the correct model.

VoicOfReason, 2017-10-19:
Does the Omnibus f7 v2 board have an onboard mag? So you know?
-- 
Peter Freund
Sent with Airmail

DzikuVx, 2017-10-31:
I belive this issue can be closed now. Thanks everyone

closed 2017-10-31 by DzikuVx