iNavFlight/inav #2360: v1.8 won't launch

teckel12, 2017-10-18:
I tried to test v1.8 tonight and it was stuck in Failsafe mode and wouldn't launch (Omnibus F4 Pro target).  I was getting strange flight status information too (headfree, headhold & horizon) none of which were set (I launch in angle mode and headfree/headhold were off).  I went back to my previous compile and it worked correctly.  At first I thought it was a telemetry issue (reporting 54122 for Tmp1 S.Port), but I think it's something in v1.8.  Dark and too late to test further.

digitalentity, 2017-10-18:
Please make a reset of settings and reconfigure from scratch. My OBF4 PRO target works fine (SBUS/IBUS receiver). Probably not resetting a configuration created a buggy settings.

teckel12, 2017-10-18:
@digitalentity I'll try it all again tonight (with a betaflight full chip erase step) and report back.  Unless full chip erase works now in configurator 1.8.0?

digitalentity, 2017-10-18:
Full erase works now in 1.8.0.

teckel12, 2017-10-18:
I did a full erase, flashed with v1.8-RC1 (downloaded not a private build) and again I get a failsafe and it refuses to launch.  Rolled back to 1.7.3 as well as a dev 1.7.4 build and it all works correctly.

I'm getting the same `54122` for Tmp1 on S.Port, so it's reporting failsave, headfree, headhold & horizon (none of which are correct).  Before you ask, here's my `diff`

# diff

# version
# INAV/OMNIBUSF4PRO 1.8.0 Oct 17 2017 / 12:31:54 (0e5cdd03)

# resources

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature LED_STRIP
feature AIRMODE
feature PWM_OUTPUT_ENABLE

# beeper
beeper -MULTI_BEEPS

# map

# name

# serial
serial 0 64 115200 38400 0 115200
serial 5 2 115200 38400 0 115200
serial 30 32 115200 38400 57600 115200

# led
led 0 0,0::CW:1
led 1 0,1::CW:1
led 2 0,8::G:0
led 3 0,14::CW:1
led 4 0,15::CW:1
led 5 15,0::CW:1
led 6 15,1::CW:1
led 7 15,8::G:2
led 8 15,14::CW:1
led 9 15,15::CW:1

# color
color 12 312,0,255

# mode_color
mode_color 2 5 2
mode_color 3 0 6
mode_color 3 5 6
mode_color 6 0 2
mode_color 6 1 1
mode_color 6 7 1

# aux
aux 0 0 4 1525 2100
aux 1 1 0 900 1300
aux 2 1 0 1700 2100
aux 3 2 0 1300 1700
aux 4 26 4 1525 2100
aux 5 4 7 1450 1550
aux 6 5 2 1300 2100
aux 7 6 2 1700 2100
aux 8 3 1 1300 2100
aux 9 3 3 1300 1700
aux 10 9 1 1700 2100
aux 11 9 3 1300 1700
aux 12 8 3 1700 2100
aux 13 21 5 1525 2100
aux 14 11 6 2000 2100
aux 15 12 6 1800 2100
aux 16 17 4 1525 2100

# adjrange

# rxrange

# master
set looptime = 500
set gyro_sync = ON
set gyro_sync_denom = 4
set gyro_hardware_lpf = 256HZ
set acc_hardware = MPU6000
set acczero_x = 150
set acczero_y = 11
set acczero_z = -665
set accgain_x = 4085
set accgain_y = 4071
set accgain_z = 4022
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set mag_declination = 700
set magzero_x = 62
set magzero_y = -182
set magzero_z = -54
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 13
set rssi_scale = 30
set serialrx_provider = SBUS
set blackbox_rate_denom = 2
set min_throttle = 1030
set max_throttle = 1852
set motor_pwm_rate = 8000
set motor_pwm_protocol = MULTISHOT
set failsafe_procedure = RTH
set align_board_roll = 17
set align_board_pitch = 1
set battery_capacity = 1630
set vbat_scale = 112
set vbat_min_cell_voltage = 34
set current_meter_scale = 327
set current_meter_offset = 86
set disarm_kill_switch = OFF
set inav_w_z_baro_p =  0.350
set nav_disarm_on_landing = ON
set nav_auto_speed = 500
set nav_manual_climb_rate = 300
set nav_landing_speed = 300
set nav_land_slowdown_minalt = 300
set nav_land_slowdown_maxalt = 1500
set nav_rth_climb_ignore_emerg = ON
set nav_rth_tail_first = ON
set nav_rth_altitude = 2300
set nav_mc_hover_thr = 1405
set nav_mc_auto_disarm_delay = 500
set telemetry_inversion = ON
set frsky_unit = IMPERIAL
set frsky_vfas_precision = 1
set smartport_fuel_percent = ON
set ledstrip_visual_beeper = ON
set osd_video_system = 2
set osd_units = IMPERIAL
set osd_rssi_alarm = 38
set osd_cap_alarm = 1400
set osd_alt_alarm = 400
set osd_main_voltage_pos = 2103
set osd_rssi_pos = 2415
set osd_flytimer_pos = 2432
set osd_ontime_pos = 86
set osd_flymode_pos = 2403
set osd_throttle_pos = 194
set osd_current_draw_pos = 2408
set osd_mah_drawn_pos = 2097
set osd_gps_speed_pos = 2441
set osd_altitude_pos = 2087
set osd_pid_roll_pos = 97
set osd_pid_pitch_pos = 129
set osd_pid_yaw_pos = 161
set osd_power_pos = 321
set acc_task_frequency = 1000
set attitude_task_frequency = 1000

# profile
profile 1

set mc_p_pitch = 58
set mc_i_pitch = 50
set mc_d_pitch = 45
set mc_p_roll = 43
set mc_i_roll = 40
set mc_d_roll = 45
set mc_p_yaw = 70
set max_angle_inclination_rll = 400
set max_angle_inclination_pit = 400
set dterm_lpf_hz = 80
set dterm_notch_hz = 260
set dterm_notch_cutoff = 160
set nav_mc_pos_xy_p = 25
set nav_mc_pos_xy_i = 180
set thr_expo = 50
set roll_rate = 45
set pitch_rate = 45
set yaw_rate = 30

teckel12, 2017-10-18:
[image]

[image]

@digitalentity Pics show the constant `54122` for Tmp1 which makes no sense and Tmp2 which shows 3D GPS fix with 11 satellites.  Would not arm.  Rolled back to dev build of v1.7.4 from just a couple nights ago and everything works correctly again.

I believe there was some failsafe reworking, that's my best guess of the source of the problem.  I can build/flash a branch if anyone has any ideas.

digitalentity, 2017-10-19:
@teckel12 Are you using a PPM receiver? 'cause this is what you have configured for.
Firmware doesn't "see" the receiver - therefore all channels are in middle causing activation of HORIZON, HEADING HOLD and HEADFREE exactly as configured. FAILSAFE mode is indicated per no receiver connection.

digitalentity, 2017-10-19:
Also, mind the release notes section: https://github.com/iNavFlight/inav/wiki/1.8.0-Release-notes#receiver-type-changes

teckel12, 2017-10-19:
@digitalentity I read the release notes before upgrading but didn't restore a dump, I just upgraded to 1.8.  So, the release notes didn't set off any red flags for me as I was just flashing a new version and didnt restore a dump from  previous version like the warning states. 

I'd suggest that the warning is a bit more clear (as I didn't catch it as I was upgrading not restoring a dump). Something more like "you **WILL** need to manually setup your receiver when upgrading". As it's not specific to just restoring a dump from a previous version but **ALL** upgrades will need manual intervention.

I'll test things in the morning.

digitalentity, 2017-10-19:
Good idea. I'll amend the release notes.

teckel12, 2017-10-19:
@digitalentity Haven't flown v1.8-RC1 but it arms now that I reset my transmitter settings. I'll fly with v1.8-RC1 later today. Is there a bug with RTH height correction in v1.8-RC1 that I should be aware of?

Anyway, closing this issue as an updated note about losing transmitter settings when upgrading should be enough. Now I need to rewire the LEDs on my Omnibus F4 Pro quads :/

closed 2017-10-19 by teckel12