iNavFlight/inav #2361: v1.8 - When in bootloader mode, buzzer is constantly on

teckel12, 2017-10-18:
I hope it's unintentional as it prevents me from doing anything with the wife in the house.  It also drives me slightly mad.  Okay, more than slightly.  Please save my marriage!

But seriously, when I go into bootloader mode it goes to a constant buzzer.  Never happened before I tried the v1.8 build.  Omnibus F4 Pro FC

teckel12, 2017-10-18:
Confirmed on two Omnibus F4 Pro FCs.  Constant buzzer when in bootloader mode.  I now have the desire to club a baby seal.  Save me from myself!

digitalentity, 2017-10-18:
As far as I can recall buzzer was always beeping when in bootloader mode on those targets. I'll check.

Pairan, 2017-10-18:
pretty normal ... had the same on Matek F405 OSD while in Bootloader mode (after that it had v1.7.3)

DzikuVx, 2017-10-18:
@digitalentity it is "hardware" thingy. All targets with `#define BEEPER_INVERTED` behave like that, buzzer driver is also an inverter after all.
I'm closing it since I see no way how to fix it besides putting buzzer port HIGH in bootloader mode. Is that even doable?

closed 2017-10-18 by DzikuVx

digitalentity, 2017-10-18:
According to ST's docs GPIO configuration registers are in reset state when in DFU mode. Reference manual states that after reset the alternate functions are not active and the I/O ports are
configured in input floating mode. 

On Omnibus F4 boards beeper happens to be connected to PB4 pin. **After reset (and in DPU mode)** PB4 (NJTRST) is configured internal with pull-up activated (at high logical level) which happens to be a level which is used to make a beeper... em... beep.

Therefore this is a hardware issue, which can't be mitigated in firmware.

fiam, 2017-10-18:
Yeah, can't be fixed unfortunately. I just keep a bit of tape around my workbench and put it over the buzzer when flashing at intempestive hours.

teckel12, 2017-10-18:
But...  This never happened until I tried the latest development branch and v1.8.  I'll try flashing an old build to verify.