iNavFlight/inav #2362: IRC Tramp Lua script not working SPRACINGF4 EVO

markandkymward, 2017-10-18:
Properly configured, checked with cleanflight firmware.  Dashes are reported on Taranis 9D.

DzikuVx, 2017-10-18:
@markandkymward this was never really confirmed working. @teckel12 I remember you had some luck with it

teckel12, 2017-10-18:
@markandkymward The betaflight Lua script for adjusting PIDs works, I've been using this for about a month now without an issue.  I really need to finish my INAV fork of the script to tweak a few things for INAV.  You can get the script here:

https://github.com/betaflight/betaflight-tx-lua-scripts

The script has a VTX panel, but I don't have the hardware to test it.  Not sure what script you're using, but maybe try the one above which I know works (at least with the PIDs).

markandkymward, 2017-10-18:
The PID and Servo rate changes worked a treat. The VTX channel page was not reporting data.

teckel12, 2017-10-19:
I don't even see the VTX page, so you got further than me. When I reworked S.Port telemetry in INAV I never did anything with VTX as I always use the same channel.

teckel12, 2017-10-19:
My guess is that VTX control from Taranis via INAV doesn't work yet. And to clarify, it never has nor has it claimed to. So this is not a bug, but a feature request. I'd love to help as I have with the PID controls, but I lack the hardware and the need to ever switch VTX channels to help. I know there's a desire for this, and the framework is set. But adding this is in someone else's court.

markandkymward, 2017-10-26:
Tried with ANYFCF7_EXTERNAL_BARO, RC1 and RC2 - works great!

teckel12, 2017-10-26:
@markandkymward Was that the Tramp Lua script or the PID Lua script?

markandkymward, 2017-10-26:
The script I run is for both. BF.lua. Can’t recall source

teckel12, 2017-10-26:
@markandkymward That's what I figured. This issue is about the Tramp Lua script, not BF. Lua.

Did the VTX panel work or just the PIDs? Because that could isolate the problem to INAV or the Tramp script.

markandkymward, 2017-10-26:
AFAIK both. Tramp could change channels but I did attempt to set PIDs. I’ll give it a look when I get back to the bench.  BTW, it is the same script I was having problems with earlier using 1.7.3 firmware for SPRACINGF4 evo.

closed 2018-05-27 by stale[bot]