iNavFlight/inav #2372: MATEK F405-OSD with 1.8RC  version matekf405OSD  LEDstrip

artusmuc, 2017-10-18:
I have tested this firmware with on a new standalone Matek F405OSD connected to a FSHUB-6S via Ribbon cable.

I have made my test first step by step ony connect one of the parts listed:
GY-63 Baro (MS6511) I2C - connect to UART3 -- works fins
GPS-NAZA-Clone - connect to UART1 --- GPS is marked as aktiv - Magno is marked as activ - first test with mago seems to work but "slow reaction" 
FrSky XRSB connected to SBUS - works fine - i have to make a new test with telemetry 

But LEDstrip WS2812 is not working in the moment. To see what happens you dont need to connect the strip. 
(Before you do the next step - please save your configuration) 

with Configurator go to Configuration Tab - scroll down to "other features" and actived their "multi-color-Led strip" with this switch - "save&reboot". Board is rebooting now - but blue LED is not blinking. Board is connected to Com and Com is activated  - but Configuator is not able to connect to the board. 
You have to reflash the board

Pairan, 2017-10-18:
Greetings to you in Munich!

1) What setup do you fly ? FW or MC ?
2) Where did you connect the ws2812?

artusmuc, 2017-10-18:
I flashed the version INAV/MATEKF405OSD 1.8.0 Oct 16 2017 12:40:38  Stronnag posted a link to rcgroups. 
As I described there is nothing else connected - you can test this effect without and new flashed board without changing anything and nothing connected. 

My setup ist only (as discribed) a MATEK F405-OSD with a MATEK FSHUB-6S connected with the ribbon cable. But i have devided this both to see the issue is caused by the FSHUB - without success. 

The WS2812 is not connected anywhere to this board - cause i dont now where to connect. The see my described issue - only actives "multi-led-strip support" in configuration software - i have a short discussion with stronnag in rcgroups and he confirms this issue and said to me a should post here.

I have tested this 2 nights long - also go back to betaflight - and make a test with this - and there all works fine. I make 5 or more test - all with the same effect. 

So - a naked board on my desk - connected with nothing - new flashed - nothing changed - only switch on the LED support in Configurator causes this issue. 

I make this tests with everything i want to connect with no combinations first - so i have not connected for example Baro with GPS - ony baro - baro deconnect - then GPS - GPS deconnect - then reciever. The next step will be the connect them all.

If you want a know an idea what can cause this issue - i will write this. But i am not an hardware expert and dont know anything about programming a flight controller - i am only a user of hard & software :)

artusmuc, 2017-10-18:
Hello again Parian :) sorry i only explain - and dont answer your question correct 

1:) no setup using - but i plan to setup a quadcopter (4 motors)
2:) WS2812 is connected nowhere - cause i dont exactly know where 

But - this is independent of causing this error of the board. 

Thank you a lot for your support

digitalentity, 2017-10-19:
So, to summarize. Once I activate LEDSTRIP on Matek F405 OSD - the board hangs on boot. Right?

artusmuc, 2017-10-19:
Yes - exactly

Pairan, 2017-10-20:
Just flashed 1.8RC1 to a fresh Matek F405 OSD, activated Multi-Color RGB LED strip in configuration and saved/rebooted ..

But my board works ... anything else ?

artusmuc, 2017-10-20:
I dont have an "official" release of f405OSD - i have only the F405-AIO. Stronnag has postet an version for OSD. And he confirms that he has the same problem.  Can i have this version you have flashed ? Thanks for your help.

Pairan, 2017-10-20:
sure ... simply flash the Rc1 from the Configurator onto you board! (hit the switch "show unstable releases")

stronnag, 2017-10-20:
So you're using the MATEKF405 target, not the MATEKF405OSD target firmware. That's not the subject of the PR.

artusmuc, 2017-10-20:
Yepp - this one works .... but this is for AIO - i am not able to select the version for the OSD - my information is there is a special one for the OSD board - as discribed "slightly different firmware for the OSD Board. Stronnag posted a link for a special firmware in regroups - with is able to use I2C over UART3. I have asked if there is a special one for the OSD board.

artusmuc, 2017-10-20:
Thanks a lot for your help stronnag :))

Pairan, 2017-10-20:
confusing!

stronnag, 2017-10-20:
Unfortunately so, but not much we can do about it.

Pairan, 2017-10-20:
no. thats just how it is- as long as we can get v1.8 on the Matek Im fine :D

artusmuc, 2017-10-20:
Maybe it helps a litte to solve  - i have seen a picture in another forum - someone has soldered direct to one the I2C pins of the CPU - and the other one he use was the pin marked of LED on the port to use the I2C port. Maybe if you activate LEDs strips an the board is booting there will be a collision between the soft-I2C and the WS2812.

digitalentity, 2017-10-22:
I can't reproduce the issue. 
Board is fully erased and flashed with MATEKF405OSD firmware (1.8-RC1). Once I activate LED_STRIP board correctly boots and I see valid ledstrip data on S5. Closing.

artusmuc, 2017-10-22:
Nice - can i have your version please ?

digitalentity, 2017-10-22:
https://github.com/iNavFlight/inav/releases/download/1.8-RC1/inav_1.8.0_MATEKF405OSD.hex

stronnag, 2017-10-22:
Looks like a compiler version issue. gcc 7 firmware locks up at boot with LED_STRIP enabled,

artusmuc, 2017-10-22:
Thanks a lot for your fast reaction - yes the board is booting correctly with this version after activated LEDstrip in Configuration tab. It's is now reachable with Configurator. But to fly we have connect to mainpower. Did you try what happens when the board is connected to mainpower. Yes LEDs are working as i want. Then i want to activate my other components  and after connected with USB the board is not  reachable for Configurator. Disconneting mainpower - rebooting with USB - board is reachable (sometimes) 

Maybee your compiler is better you think and detecteds some conflicts automatically  :)

artusmuc, 2017-10-22:
So - i have tested a little bit - and HEUREKA :)
UART1 is setting to MSP by default after a full-erase flash ( i have there the NAZA Cone GPS connected). Maybee it will be helpful after first flashing UART1 not setting to MSP as default. 

But i was first focused by testing I2C and LEDs. But i dont give up. I deactived MSP rebooting - and with connected mainpower - the board is running and configureable via Configurator :)  so - connecting the other stuff - all do their job now.

You can close this

digitalentity, 2017-10-22:
@artusmuc MSP on UART1 is intentional - in case you break USB you will be able to continue using the board through UART1 and FTDI converter. 

You can always disconnect GPS, reconfigure to use GPS on UART1 and then connect GPS - this configuration should persist across upgrades (unless you do a full erase or configuration reset).

@stronnag can you confirm that GCC7 generates faulty binaries?

stronnag, 2017-10-22:
I can.  I've just reverted to 6.3 and the firmware survives LED_STRIP without any problem. So this issue is an artefact of the compiler (7.2 used previous to create the bad image).

digitalentity, 2017-10-22:
Ok, I guess we can close this. We might want to open a new issue to track this gcc7 fluke

closed 2017-10-22 by digitalentity

artusmuc, 2017-10-22:
Yes - its only a problem when MSP on UART1 is activated and there is connected a GPS for a fresh-flashed board, in this case the board is not reachable after rebooting for Configurator. If there is nothing connected all is ok :) Sorry for the trouble  and many thanks for your support :)

adis1313, 2018-02-16:
I can also confirm that ARM Toolchain 7 breaks LED functionality in iNav on Omnibus F4 Pro V3. After revert to v6 it works properly.

stronnag, 2018-02-16:
Interesting, on the Matek, it's not completely dead, just somewhat zombie like and useless. For example, VCP is recognised, but dysfunctional:
````
$ dmesg .
[83244.318385] usb 3-3.3: new full-speed USB device number 16 using xhci_hcd
[83244.396186] usb 3-3.3: New USB device found, idVendor=0483, idProduct=5740
[83244.396190] usb 3-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[83244.396191] usb 3-3.3: Product: MatekF4
[83244.396193] usb 3-3.3: Manufacturer: INAV
[83244.396194] usb 3-3.3: SerialNumber: 0x8000000
[83244.397854] cdc_acm 3-3.3:1.0: ttyACM0: USB ACM device
[83256.446745] cdc_acm 3-3.3:1.0: failed to set dtr/rts
[83284.095524] cdc_acm 3-3.3:1.0: failed to set dtr/rts
### No communications possible ....
````
whereas normally ....
````
[83526.688849] usb 3-3.3: new full-speed USB device number 19 using xhci_hcd
[83526.766371] usb 3-3.3: New USB device found, idVendor=0483, idProduct=5740
[83526.766373] usb 3-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[83526.766374] usb 3-3.3: Product: MatekF4
[83526.766374] usb 3-3.3: Manufacturer: INAV
[83526.766375] usb 3-3.3: SerialNumber: 0x8000000
[83526.766911] cdc_acm 3-3.3:1.0: ttyACM0: USB ACM device
````
Manufacturer INAV, nice touch.

So I guess the damage happens after VCP initialisation (which is earlier in 1.9)

````
# INAV/MATEKF405 1.9.0 Feb 15 2018 / 20:01:39 (2bcdc237)
````
````
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (Arch Repository) 7.3.0
````