iNavFlight/inav #2373: 1.8-RC1 - full throttle after some flying in ACRO+ALTHOLD

krzysztofmatula, 2017-10-18:
[blackbox_log_2017-10-18_225150.TXT](https://github.com/iNavFlight/inav/files/1396430/blackbox_log_2017-10-18_225150.TXT)

See the attached log (2/4) at 2:35.
I was flying ACRO+ALTHOLD at ~35m and everything was fine. Suddenly the motors started to be commanded full-throttle, likely by altitude controller. Few seconds later the quad was at ~100m and I commanded RTH. Later I tried to reproduce the issue and it happened again (should be later in the log).

I think I tried ACRO+ALTHOLD on 1.7.3 and it was fine.

digitalentity, 2017-10-19:
Have a look at your accelerometer graphs - specifically Log entry 2, time 02:34 to 02:38.
[image]

What you see is a well known case of accelerometer resonating when vibration from motors exceed certain level. Can't be fixed in software. It was a mere coincidence that 1.7.3 worked. Fix that is known to work is soft-mounting the FC.

closed 2017-10-19 by digitalentity

krzysztofmatula, 2017-10-19:
OK, thanks for pointing this out!
Now I realized I probably was flying on 1.7.3 with different props. That could make the difference.
Sorry for false bug report...