iNavFlight/inav #2377: channels AETR1234 on

argileh, 2017-10-20:
I got the Omnibus F3 AIO board and flashed it with INAV 1.7.3.  In the GUI, I set the receiver type to to PPM. However, in the Receiver tab, the default setting shows AETR5678 (not 1234). I am using a FlySky FS-i6s radio and AETR are fixed to channels 1234. 

I flashed the same board with Cleanflight and it showed AETR1234. I also had used INav with a SP Racinh F3 board and there two it showed AETR1234. It is only the combination of Omnibus F3 with INAV that is producing AETR5678.

Any clue?  How can I get inav with Omnibus to have AETR1234.

giacomo892, 2017-10-20:
Upgrade to 1.8-RC1

argileh, 2017-10-20:
Thank you giacomo. Could please provide some details on what RC1 is (pls excuse my ignorance) and how to upgrade. Thanks

giacomo892, 2017-10-20:
You can upgrade as you installed 1.7.3 :) with the option "full chip erase" to reset settings. 

Read release notes https://github.com/iNavFlight/inav/releases/tag/1.8-RC1

And please close this issue since github is not a real support channel. Join slack channel for that.

digitalentity, 2017-10-20:
@fiam @DzikuVx we might have a bug in Configurator. Are we supporting pre-1.8 channel maps?

argileh, 2017-10-20:
I did the “full chip erase” but did not solve the issue.

argileh, 2017-10-20:
I flashed version 1.8 as suggested by giacomo (with "full chip erase") and I am still getting AETR5678 when selecting Default channel map.

Anyway, I will close the topic as requested.

fiam, 2017-10-20:
@argileh It's just cosmetic. Configurator 1.8 will show e.g. `AETR5678` because it refers to actual channel numbers. Earlier versions of the configurator will show `AETR1234` because they refer to AUX channel numbers. The channel mapping will be correctly written to the flight controller regardless of the INAV version installed in it.

I'm keeping this open because there's been some confusion. We need to tweak the UI a bit to make more obvious that the numbers now refer to actual channel numbers.

argileh, 2017-10-20:
@fiam, It is still not working. There is no response when I move the transmitter sticks. Weird thing is that I do not have this issue with iNAV+ SP Racing F3, neither with the Omnibus F3 board+Cleanflight. The issue appears only with the combination of Omnibus F3+iNAV. It looks more than just cosmetic, although I am hoping I am wrong and there is a simple fix.

digitalentity, 2017-10-22:
Tested Omnibus F3 + INAV 1.8-RC1 + PPM receiver. No issues. Please make sure you have S.Bus jumper open and PPM jumper closed.

closed 2017-10-22 by digitalentity

markandkymward, 2017-10-26:
This problem also exists with ANYFCF7_EXTERNAL_BARO, stable, RC1 and RC2.

digitalentity, 2017-10-26:
@markandkymward For version 1.8 this is normal behavior. Version > 1.8.0 changed the notation of the map. It now shows actual channel numbers, not AUX channel numbers. Therefore AETR5678 should be interpreted as "Aileron, Elevator, Throttle, Rudder, CH5, CH6, CH7, CH8".

markandkymward, 2017-10-26:
That makes sense when put that way. Problem is I’m not getting any signal from a known good receiver. Ports mapped and sbus (frisky x8R) inverted. ANYFCF7_EXTERNAL_BARO, stable, RC1 or RC2 firmware, iNav 1.8 configurator.

digitalentity, 2017-10-27:
Please provide more details about your configuration. Full "dump" is recommended.

svhirst, 2018-04-04:
I'm having the same problem with Inav1.9.2 and a Sp F3 board AETR Channels 5678 and getting no response from a proven receiver.

digitalentity, 2018-04-04:
Since 1.9 mapping changed. AETR5678 (1.9+) = AETR1234 (1.8 and before).

svhirst, 2018-04-04:
Yes I see that now, I have it connected now, Inav doesn't list the channels for the sticks, but every thing works.

maRCnh33, 2018-09-04:
My problem is that i am busy for days now.... short story version.... i have only AETR and no AETR5678 or AETR1234... i have a INAV F4 Flightcontroller (a furious fpv F35 clone). I can change AETR in RTEA if i want... that works but when i type behind AETR its all RED... it doesnt work, i have used different firmware versions, different versions of INAV GUI... but nothing worked! This FC is an upgrade for my rc fpv plane wich had a NAZE F1 board, there is the CHANNEL MAP : 56AETR78 and if i connect that board i can change it in whatever i want!! but not with the f4 fc and i dont know why!! i hope someone can help me!