iNavFlight/inav #2379: Omnibus F7 V2 configurator not recognizing board on 1.8-RC1

VoicOfReason, 2017-10-20:
I have betaflight 3.2.1 successfully installed on Omnibus F7 V2. I can also configure it via USB on PC. But after installing any version, including 1.8 of iNav on the F7, the configurator for iNav will NOT recognize the board. I have tried using the bootloader, using DFU with Zadig, etc. I CAN flash the board and it says successful, but it the COM port (in this case COM13) will not show up and so I can't connect to it. I have tried multiple times and then have gone back to betaflight 3.2 with no problems whatsoever.

teckel12, 2017-10-21:
The Omnibus F7 v2 is not currently supported on INAV. The FC is brand new. http://www.readytoflyquads.com is a good source for FCs as they show which FCs are compatible with INAV and the Omnibus F7 v2 is list as compatible with cleanflight and betaflight but not INAV. The Omnibus F4 Pro is probably a better option, more mature (so fewer problems), and works with INAV. I believe I tried to convince you to get a Omnibus F4 Pro on another support issue.

VoicOfReason, 2017-10-21:
F7v2 is listed as a newly supported board under unavoidable 1.8RC1. Do you
have direct link to the f4 pro board you’re referring t9? Online I see V2
pro boards, pro boards with gel mounted components, a corner board, and a
v3 board. So I’d like to order the f4 pro but not sure which one you’re
recommending,

VoicOfReason, 2017-10-21:
...ok, I went ahead and ordered the Omnibus F4 V3 Pro from Ready to Fly. Looks like a good choice. I was originally planning on using an Omnibus as I had one sitting around but it was faulty so I just decided on something more powerful (F7). Not a good choice apparently.

VoicOfReason, 2017-10-21:
Quick question on the Omnibus F4 boards. I ordered a V3 board, but
wondering if soft serial is supported on iNav on that board?

teckel12, 2017-10-21:
I believe the F7 v2 support was just added, so there's bound to be problems. It will probably work fine after a few revisions.  So your purchase isn't a waste, just a little too early. 

Yes, you can run soft serial on the Omnibus F4 Pro. You use two small surface pads CH5/CH6 that are between the F4 microcontroller and the motor out pad. I use it for telemetry. There's a port with UART6 and UART3 (which is used for I2C) so that's a single port connection for your GPS/magnatometer. Makes it nice as the GPS is always in some ackward mounting location with a long connection.

teckel12, 2017-10-21:
Thoughs on the F7...  While the F7 is slightly more powerful (as in clock speed) it's not much. Also, CPU speed isn't much of an issue anymore (I run at like 4% CPU load on my F4). The F7 does have more UARTS and built in serial inverters which is really nice. But, I've yet to find an F7 FC that's designed well and exposes all the UARTS. The larger size of the chip also results in compromises being made on the FC to fit everything. For these reasons and the fact that there's some sketchy support I think the F4 is a better FC microprocessor at the current time. I'm sure that will change in the future.

VoicOfReason, 2017-10-21:
Good info. Thanks

VoicOfReason, 2017-10-21:
Do all the omnibus F4 boards still require the inverter hack for using XSR
receivers?

W. Peter Freund
Adjunct Instructor - Philosophy
Northwest Arkansas Community College
----
“Hold no knowledge in contempt, blush to learn from no man, once you have
attained knowledge, look down on no-one.” (St. Victor)

teckel12, 2017-10-21:
@VoicOfReason No hack required. The SBUS port includes an inverter by default and soft serial can be inverted via software.

VoicOfReason, 2017-10-21:
Same for telemetry? That's my concern.

teckel12, 2017-10-21:
@VoicOfReason For telemetry you use the soft serial port. As the name implies, it's not really a serial port, but just two pins and the microcontroller software emulates a UART. Since it's software, it can invert the signal without hardware.

So yes, you can run SmartPort telemetry on the soft serial port on a Omnibus F4 Pro. I run GPS (UART6), magnatometer (I2C on UART3), SBUS (UART1), and SmartPort telemetry (soft serial CH5/CH6).

Please read the INAV Omnibus F4 target manual and the INAV telemetry manual (SmartPort section) for more details. I have 2 models running this hardware so just ask if you have any other questions.

VoicOfReason, 2017-10-21:
Excellent. I've used soft serial before but not for telemetry like that.
That's a great idea.

digitalentity, 2017-10-22:
@VoicOfReason I've tested the 1.8-RC1 firmware on original Omnibus F7 V2 from Airbot and it works fine. Where did you get yours?

VoicOfReason, 2017-10-22:
Drone Squad

digitalentity, 2017-10-22:
Actually, you probably flashed the V1 firmware. I just realized that I didn't upload the V2 firmware to the release, sorry.

I've just uploaded the 1.8-RC1 for OMNIBUSF7V2 target, can you please test?

VoicOfReason, 2017-10-22:
Yes! :)

VoicOfReason, 2017-10-22:
Also, is soft serial enabled on the f7 v2?

digitalentity, 2017-10-22:
No softserials on F7 yet

VoicOfReason, 2017-10-22:
But eventually?

digitalentity, 2017-10-22:
Eventually we'll probably do something about this.

VoicOfReason, 2017-10-22:
Ok. I’ve run out of ports. I need uarts for my runcam split and tbs smart
audio I don’t have.

VoicOfReason, 2017-10-22:
UPDATE: I flashed with the new F7 V2 1.8RC1 firmware. It flashes just fine, same as before, but it STILL won't let me connect to the board in iNav configurator on my PC after I've flashed it. So, it's still not working. Going to have to go back to Betaflight.

digitalentity, 2017-10-22:
Now this is odd. My F7 V2 works fine - just flashed it with firmware I uploaded to GitHub. Can you post a link to the board you purchased?

VoicOfReason, 2017-10-22:
OMNIBUS F7 V2
https://www.amazon.com/dp/B07534KR56/ref=cm_sw_r_cp_api_PSc7zb40DB388

digitalentity, 2017-10-22:
Try this build:
[inav_1.8.0_OMNIBUSF7V2.zip](https://github.com/iNavFlight/inav/files/1404628/inav_1.8.0_OMNIBUSF7V2.zip)

VoicOfReason, 2017-10-22:
Nope. That didn't work either. Failed to open serial port. Same as before.

digitalentity, 2017-10-22:
Ok, how do the LEDs on the board behave?

VoicOfReason, 2017-10-22:
They all seem to be fine. In fact, after I upgrade to iNav 1.8, when I plug in USB cable, I get 3 blue light flashes just as I do in betaflight. After that, I don't have any more blue light flashes, I don't think. But, I get a solid green and blinking blue light (3 times).

VoicOfReason, 2017-10-22:
...I'm flashing using DFU. It's the only way it will flash.

digitalentity, 2017-10-22:
Are you doing a full erase? Remainder of Betaflight's configuration is known to cause boot issues in INAV.

VoicOfReason, 2017-10-22:
Yes I’m doing a full erase each time

VoicOfReason, 2017-10-22:
Maybe you could walk me thru exactly the steps you are doing to flash your f7?

VoicOfReason, 2017-10-22:
.... also, is it possible that migrating from bf3.2.1 causes issues as opposed to migrating from 3.2?

teckel12, 2017-10-22:
If you do a full erase while flashing it doesn't matter what version you're coming from. The steps are just to select full erase and instead of selecting to download firmware, select local and select the local file that digitalentity linked to.

VoicOfReason, 2017-10-22:
Yeah, I did All that and it flashes but iMac configuration can’t connect to
the board after flashing. It can’t open that same serial port anymore.

artusmuc, 2017-10-22:
I have a similar problem with my F4 board (solved) - is there anything connected to your board - try to start without mainpower only USB and no additional devices after flashing.

VoicOfReason, 2017-10-22:
How did you solve your problem exactly?

VoicOfReason, 2017-10-22:
Problem solved! I got the F7 V2 to flash with iNav finally and it connects. Here's why it wasn't working: I had a faulty GPS (BN-880) connected through UART6 and UART3 (I2C). It burned out from a prior crash but I just left it hooked up and it seems to have prevented the FC from connecting. I just replaced it with a new BN-880 and re-flashed and now it all works. Developers, take note. A faulty GPS can cause problems with iNav flashing and connection to PC.

VoicOfReason, 2017-10-22:
Since you all are so helpful, I have a new problem. My motors won't spin up properly under iNav. Under Betaflight, it flew wonderfully, but with iNav, motors are twitching, some spin up, some don't, and then at some point, motors just cut off and it drops from sky. Obviously this sounds like a oneshot/multishot, etc. issue, but not sure what's going on. I Have the Lumenier 4-in-1 ESC 35a with Dshot600 and support for oneshot, multishot, etc. I tried the preset settings in iNav for 5" racer. I also tried with notch filters turned off. I changed 1kz/2kz/4kz settings all over the place. I even disabled ESC protocol completely and flew as "standard" with 50hz refresh. It was erratic in all modes. Again, same setup runs perfectly in betaflight 3.2.1. Any thoughts?

Motors: TBS Endurance
F7 V2, of course

teckel12, 2017-10-23:
@VoicOfReason I really doubt software broke your quad. It sounds like there's some kind of hardware issue (we already know there was a GPS hardware issue).

I've found that Multishot is more reliable than Oneshot (I've had a few crazy cases where Oneshot stuck a motor at 100% for no reason).  Also make sure you've upgraded your ESCs to the latest version.

To me, this sounds like an ESCs configuration issue. Calibrating the ESCs min/max, calibrating the transmitter range, etc. Also, what ESCs do you have and what are the settings?

VoicOfReason, 2017-10-23:
II’ve already done everything you’ve said. I even Reflashed esc’s. Perfect
signal strength.

VoicOfReason, 2017-10-23:
... Does it now under iNav and Betaflight when it didn't under Betaflight just beforehand. Everything works perfect if the props are off. I can spin motors up to full speed. As soon as I put on the props, it can barely takes off, then motors twitch, and then all turn off at once and then quad falls to the ground. Batteries are fine (checked 2 batteries, brand new).

VoicOfReason, 2017-10-23:
I really appreciate your help on this...

So, here's a summary of my issue. I'm still posting here because I'm not entirely certain it isn't an iNav bug since the problem began immediately after flashing iNav with absolutely no hardware changes or damage.

1. Flashed iNav 1.8
2. Configured for flight using Oneshot
3. Tried to fly but was twitching, crashing, cutting power, etc. (see video below)
4. Went back into Inav config and tried all the difference ESC profiles (multishot, etc.)
5. Same problem
6. Flashed Betaflight 3.2.1 back on the board
7. Now the problem exists in Betaflight
8. Reflashed the ESC's 3 times with different firmwares
9. Problem remains

Keep in mind, I flew this quad for 28 minutes straight on a battery pack yesterday and everything performed flawlessly even at a distance. I checked battery packs but I've got 2 new packs I tested. Same problem. None of this makes sense to me.

Clues:
1. I had trouble flashing the F7 board to begin with, so I'm starting to suspect the F7 board was bad to begin with
2. I can spin up the motors to full speed within Betaflight using motor controls
3. I can also spin up the motors to full speed in the field with the props off
4. As soon as I put the props on and the quad starts to lift off, it drops all power and drops to the ground

https://youtu.be/i2xw-VemBVI

teckel12, 2017-10-23:
What do you have connected to your FC's 5v power supply? It could also be the lack of a capacitor on your battery input (although that typically happens at full throttle).

I'd remove VTX, GPS, etc. Also, only the latest ESC firmware should be used. And using wrong settings for your ESC model could cause all kinds of problems.

VoicOfReason, 2017-10-23:
I already Have a capacitor attached. But like I said, the exact same
hardware setup produced perfect flying just before I flashed to iNav so I
don’t see how it could be my hardware setup. I also Use latest esc firmware

VoicOfReason, 2017-10-23:
Hooked up via 5v from FC is GPS, buzzer, and LED. Camera and vtx get power
from esc directly.
-- 
W. Peter Freund
Adjunct Instructor - Philosophy
Northwest Arkansas Community College
----
“Hold no knowledge in contempt, blush to learn from no man, once you have
attained knowledge, look down on no-one.” (St. Victor)

teckel12, 2017-10-23:
Explain "Camera and vtx get power from esc directly". You have ESCs with BECs? How old are they? I haven't seen an ESC with a BEC since I built my first multiwii 450mm drone "back in the day".

VoicOfReason, 2017-10-23:
Lumenier 35a 4-in-1 Blheli_S DShot600 ESC with 12v output (using this for VTX and camera) and 5v output (using this for LED light).

teckel12, 2017-10-23:
Ah a 4 in 1 ESC. With your quad size I assumed individual ESCs.

I'd really like to see what firmware you flash your ESCs and what ESC settings you use. It looks like the ESCs are shutting down (like a overload protection which should be set to off). Or the wrong ESC firmware. 

For sure I would start over. Flash,  calibrate, program ESCs, full erase FC and manually enter settings instead uploading a dump.

VoicOfReason, 2017-10-23:
[image]

VoicOfReason, 2017-10-23:
I reflashed the ESC's to exact latest software I had before and I erased and reflashed Betaflight. Then, I manually put in all the correct values from last time (no dump) and no change. It still flies a few inches up and then all motors cut and it falls. Same as video. :(

teckel12, 2017-10-23:
@VoicOfReason It looks like you haven't done a throttle calibration on your ESCs (see ESC 2). 

Remove battery, enable motors, move sliders to max for all motors, connect battery, reduce throttle sliders from max to minimum for each motor.  Your ESCs should be singing to you. The motors should never spin, but still a good idea to remove props first. 

Also, I would turn temp protection and low rpm power protection **Off**. I'd rather not have one of my ESCs decide to turn off while in flight, guaranteeing a spectacular crash. So what if an ESC overheats a bit. 

Finally, after you get past this problem, consider turning on brake on stop. You'll get better control and you have a cap on your battery input so you're ready for motor breaking.

VoicOfReason, 2017-10-23:
OK. I went ahead and did the throttle calibration and changed all of those settings. No change. It still won't fly more than an inch of the ground before all motors cut.

teckel12, 2017-10-23:
@VoicOfReason is it now showing the same values (except direction) for each ESC? And what's the min/max throttle now?

VoicOfReason, 2017-10-23:
Here are my new values after calibrating all 4 motors.

[image]

teckel12, 2017-10-23:
@VoicOfReason See the min/max throttle values? They should be the same for all ESCs. Do the calibration on each ESC individually and make sure it's working. You should read up on setting up your ESCs. With this configuration,  it's no wonder it's not working.

VoicOfReason, 2017-10-23:
Well, this is a bit of a disaster. I just calibrated motor 1 fine. Then, I went to motor 2 and it literally went up in smoke. Wish I had left the temperature protection turned on.

teckel12, 2017-10-23:
@VoicOfReason The motors don't even spin during calibration. And even if at max throttle, the motor or ESC shouldn't smoke unless you run it for a long time. Something is seriously wrong with your configuration.

VoicOfReason, 2017-10-23:
Exactly.

VoicOfReason, 2017-10-23:
Except that I flew A perfect flight for 28 minute straight and motors
weren’t even warm. When I seitched To iNav that’s when problems all started.

[16 more replies]

closed 2017-10-24 by digitalentity