iNavFlight/inav #2381: Help with rate mode -> it just drops

stefan73, 2017-10-21:
I need some help to get acro + airmode running. Quad flys nicely in angle + horizon mode. It flips in horizon as desired. It does PosHold + RTH nicely. So i set up a new switch configuration, that just turns on Airmode. My expectaion: This will make it fly in rate/acro mode and have airmode enabled.

I set up:
Angle Mode: CH8 = 1000
Acro (no mode) + Airmode: CH8 = 1600

Experiment 1: Switch from angle mode (CH8=1000) to Airmode (CH8=1600) flying about 3 meters high. Seems to flip over. Motors basically off. Drop.
Experiment 2: Switch to Airmode (CH8=1600) while disarmed. Arm. Quad flips over on ground.

What is wrong?

Thx
Stefan

 # dump

# version
# INAV/SPRACINGF3 1.7.3 Sep  9 2017 / 10:20:09 (974440df)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature GPS
feature PWM_OUTPUT_ENABLE
[... 433 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 60
set pitch_rate = 60
set yaw_rate = 60
set tpa_rate = 0
set tpa_breakpoint = 1500

stefan73, 2017-10-21:
Went home. Fixed the quad to the ground.
Disarmed -> Acro on -> OSD says ACRO -> Arm -> props beginn spinning --> NO Reaction to Throttle! Just keeps on minimum throttle.

Switch to Angle. All great.

stefan73, 2017-10-21:
And another addition: It is the RC. Seems to be a bug in the FS-i6 14 Channel FW. Remote freezes all channels.

giacomo892, 2017-10-21:
Can you take a look into receiver tab from configurator and see what is going on while on the bench?

stefan73, 2017-10-21:
Solved. My fault. It was the transmitter config. I had one of the switches defined as Trainer mode toggle on my FS-i6. I forgot about that and defined this switch as Acro switch. 
So Acro is switched on and due to trainer mode, throttle goes to zero, other channels to middle and channels are locked.
Removed trainer setting and went out for a flight. Acro mode with Airmode perfectly working.

giacomo892, 2017-10-21:
@stefan73 close then issue then. Thanks!

closed 2017-10-22 by stefan73