iNavFlight/inav #2382: Sp F3 deluxe OSD and PPM+Softserial

michaelvanderhoek, 2017-10-21:
So, after a big disappointment on the banggood omnibus F4 v2 with the leds not working i bought an sp F3 deluxe with integrated osd.

It has anything on it, baro, compass, multiple uart's so it looked like the best choice for my hexa.

Well..... i guess not...

With inav 1.7.3 and 1.7.2 i can not get ppm receiver ans softserials to work together. When i switch the one, the other goes off. When i setup the softserial for my bluetooth i wont work either. Tried every baudrate but no luck. 

I checked the wiring over and over again and finally flashed betaflight to see if it was firmware related. 
Everything works like a charm when betaflight is flashed. 

Back to inav, nothing works. No ppm and softserial at the same time, no softserial working with bluetooth module, no bluetooth working at uart 1. 

Oh, and the known bug with 1.7.2, cant configure hexX and save. It pops back to quad X. 

Anybody had this problems earlier? 

Cheers, 
Michael

michaelvdhoek, 2017-10-22:
I checked everything again today, it works on betaflight and on cleanflight. 
But, INAV keeps disappointing me. 

Here's a dump from Cleanflight.

2017-10-22 @ 14:10:47 -- Running - OS: Windows, Chrome: 61.0.3163.100, Configurator: 2.1.0
2017-10-22 @ 14:10:56 -- Loaded release information from GitHub.
2017-10-22 @ 14:13:06 -- Serial port successfully opened with ID: 5
2017-10-22 @ 14:13:14 -- Serial port successfully closed
2017-10-22 @ 14:13:16 -- Serial port successfully opened with ID: 6
2017-10-22 @ 14:13:26 -- No configuration received within 10 seconds, communication failed
2017-10-22 @ 14:13:26 -- Serial port successfully closed
2017-10-22 @ 14:13:39 -- Serial port successfully opened with ID: 7
2017-10-22 @ 14:13:39 -- MultiWii API version: 1.36.0
2017-10-22 @ 14:13:39 -- Flight controller info, identifier: CLFL, version: 2.1.0
2017-10-22 @ 14:13:39 -- Running firmware released on: Jul 30 2017 13:13:34
2017-10-22 @ 14:13:39 -- Board: SRF3, version: 0
2017-10-22 @ 14:13:39 -- Unique device ID: 0x3500374134570820333634
2017-10-22 @ 14:14:16 -- EEPROM saved
2017-10-22 @ 14:14:16 -- Device - Rebooting
2017-10-22 @ 14:14:19 -- Device - Ready
2017-10-22 @ 14:14:26 -- EEPROM saved
2017-10-22 @ 14:14:27 -- Device - Rebooting
2017-10-22 @ 14:14:29 -- Device - Ready
2017-10-22 @ 14:14:47 -- EEPROM saved
2017-10-22 @ 14:14:47 -- Device - Rebooting
2017-10-22 @ 14:14:49 -- Device - Ready
2017-10-22 @ 14:15:07 -- EEPROM saved
2017-10-22 @ 14:15:07 -- Device - Rebooting
2017-10-22 @ 14:15:10 -- Device - Ready
2017-10-22 @ 14:15:59 -- EEPROM saved
2017-10-22 @ 14:16:41 -- Serial port successfully closed
2017-10-22 @ 14:23:04 -- Serial port successfully opened with ID: 8
2017-10-22 @ 14:23:04 -- MultiWii API version: 1.36.0
2017-10-22 @ 14:23:04 -- Flight controller info, identifier: CLFL, version: 2.1.0
2017-10-22 @ 14:23:04 -- Running firmware released on: Jul 30 2017 13:13:34
2017-10-22 @ 14:23:04 -- Board: SRF3, version: 0
2017-10-22 @ 14:23:04 -- Unique device ID: 0x3500374134570820333634
2017-10-22 @ 14:23:29 -- EEPROM saved
2017-10-22 @ 14:38:41 -- OSD settings saved
2017-10-22 @ 14:38:52 -- Serial port successfully closed
2017-10-22 @ 14:42:13 -- Serial port successfully opened with ID: 9
2017-10-22 @ 14:42:13 -- MultiWii API version: 1.36.0
2017-10-22 @ 14:42:13 -- Flight controller info, identifier: CLFL, version: 2.1.0
2017-10-22 @ 14:42:13 -- Running firmware released on: Jul 30 2017 13:13:34
2017-10-22 @ 14:42:13 -- Board: SRF3, version: 0
2017-10-22 @ 14:42:13 -- Unique device ID: 0x3500374134570820333634
2017-10-22 @ 14:42:20 -- CLI mode detected
Setup
Ports
Configuration
Power & Battery
Failsafe
PID Tuning
Receiver
Modes
Adjustments
Servos
GPS
Motors
OSD
Race Transponder
LED Strip
Sensors
Tethered Logging
Blackbox
CLI
Note: Leaving CLI tab or pressing Disconnect will automatically send "exit" to the board. With the latest firmware this will make the controller restart and unsaved changes will be lost.

CLI
[... 538 lines cut ...]
set p_level = 50
set i_level = 50
set d_level = 75
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
set horizon_tilt_effect = 75
set horizon_tilt_expert_mode = OFF
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_expo_yaw = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

#

michaelvdhoek, 2017-10-22:
I think i found it, 

When using the sliders in the GUI i cannot get it on ppm and softserial. 
I tried via CLI and that is working!

maybe its a GUI thing?

mattyra, 2017-10-22:
@michaelvanderhoek  can you share what command you used in the CLI pls i have the same problem on another board

michaelvanderhoek, 2017-10-22:
Hey mattyra,

I only set the features rx_ppm and softserial in CLI.

Thats all. Then i configured my ports like i would like them. Uart 2 for GPS and softserial 1 for bluetooth at 9600 baud. 

What problem do you have?

DzikuVx, 2018-04-04:
Issue appear to be dead, closing. Thanks for participating

closed 2018-04-04 by DzikuVx