iNavFlight/inav #2384: Fixed wing: Getting rid of "Board aligment pitch"-tuning for flying levelled

Pairan, 2017-10-22:
I've made my mind about why we need to do this in the way we do this and came to no clue.

To define it to be "levelled" is ok and already detected (runtime calibration). But if I think back to lets say Cyclops "StormOSD" or other systems, they can detect the FW flying level. This is, what takes the most of my time by now! (Ok, lets say, it's just me who struggles with this in the beginning)

Is that something that can be  integrated into iNav? 

By now it already detects alt changes (numeric vario). It the value is about zero then I would assume the FW to fly levelled! 

Maybe its a strange point of view but I would assume iNav to improve.

Please share your thoughts and ideas to that

Redshifft, 2017-10-22:
An Aeroplane is not like a multirotor, it will fly perfectly well without a flight controller and without exception should be fully set up manually before you even think of using Inav's functions. If it is not done this way the plane will never fly well in ALL situations.

Adding more "auto this and that" just gets inexperienced people in more trouble.
It could not be much simpler, mount the leveled FC in the aeroplane resting at it's general flying incidence and you won't have any trouble. If you can not do that, just redefine level again with the plane at it's estimated incidence as before.

If you end up "chasing" level you have a badly set up plane, not an Inav issue or indeed anything that would be helped by some auto feature.

digitalentity, 2017-10-22:
I totally agree with @Redshifft. Properly set up plane has an inherent ability to fly level without any control or stabilization at all. This flying incidence should be figured out by trimming the plane when flying manually (or guessed). Firmware should be configured to consider this very incidence as "level".

If you are not able to make your plane fly level in manual mode - something is wrong with your machine - most likely CG.

digitalentity, 2017-10-22:
I'll keep this open for a while to allow people to exchange thoughts on the topic, but my personal view is that another "auto" for this is not a good idea.

Dronek, 2017-10-22:
I support the statement by @Redshifft but at the same time I guess @Pairan is not talking about a badly setup plane to fly manually. A plane can be adjusted to fly perfectly in pass-through mode by trimming and setting COG right. However, this is more related to servo autotrim and will result in a straight flying plane more or less regardless of pitch angle etc. It has nothing to do with leveling.

A manually controlled plane normally doesn't/shouldn't exhibit leveling by itself, as far as I know.
In order to have iNav working properly, board alignment is crucial but not every fuselage will have a perfect mounting position for an FC at the right incidence. In my s800 I mounted the FC more or less parallel to the bottom of the bay but had to correct the pitch angle by 5 degrees in iNav, in order to let the wing keep it's altitude roughly when flying in angle mode. Maybe I had some offset in the mounting or anything else...
I guess it's this type of correction that @Pairan is talking about, no? 
My wing was well adjusted and flying well in pass-through and needed this correction nevertheless, which is perfectly reasonable for me.

@Pairan: I don't agree that adjusting board alignment is a difficult and/or time consuming task though. I also don't think that iNav needs any automation for this.

Pairan, 2017-10-22:
Now ... let me get clearer in what I wanted. Im flying a complete refurbished wing these days and it does indeed fly straight in passthrough which is a desired state and mandatory.

As @Dronek assumed right, its about keeping it in that way! 

My "problem" starts with hitting Horizon mode and blends over to posHold! 

1) If I hit Horizon mode now, it looses altitude which needs to be fixed by adjusting the board alignment (pitch). Ok, done! works!
2) If i make it work for Horizon mode then i will have to recheck this in PosHold too (tends to loose altitude making it unsafe). Having a look back in the wiki tells me: Dude, you'll better take ALT-HOLD in addition to poshold for this to happen! And maybe i did wrong ...

But hey ? It does fly level without **any** iNav maths :) iNav does so many cool things and offers even more. Yet again we need to figure this board alignment to make it hold altitude/fly level (although it does by default). 

Basically Im ok with that! If I loose alt, then I'll correct the pitch alignment UP, otherwise DOWN until it makes me happy. It takes some attempts. 

And I'm ok with the words of @Redshifft and @digitalentity BUT I don't wanted to invoke a "IT DOES ANYTHING BY ITSELF WHILE I DONT CARE WHAT IM DOING!"-thing. So, in case you thought so ... no !

I simply had the idea of skipping THIS alignment part as from the point where I start, my plane flies LEVELLED BEFORE iNav gets the job to do the maths! Further I don't stand here and point fingers at "THIS IS NEEDED OTHERWISE WORTHLESS!" :D ... no, my friends- I simply wanted to hear you about the idea, which seams to be only a thing I would wish to have, not the majority.

digitalentity, 2017-10-22:
Ok, now after you explained your suggestion it makes more sense than before.

Pairan, 2017-10-22:
I tend to be confusing by time ... sorry!

Dronek, 2017-10-22:
> If i make it work for Horizon mode then i will have to recheck this in PosHold too (tends to loose altitude making it unsafe). Having a look back in the wiki tells me: Dude, you'll better take ALT-HOLD in addition to poshold for this to happen! And maybe i did wrong ...

I guess, this is the crucial point. If you want your aircraft to hold altitude reliably you should really enable ALTHOLD in the first place. Expecting POSHOLD to hold altitude during turns is a different story and IMHO much harder to achieve than in combination with ALTHOLD.

Pairan, 2017-10-22:
So far Horizon mode did a good job for me

skaman82, 2017-11-10:
@Dronek I think its not about holding altitude. Its about flying level (not climbing/sinking when the stick is centered) in Horozon/Angle modes.

Feldsalat, 2017-11-10:
What about current roll and pitch degrees as OSD elements? I would find it useful to optimize settings more easily.

MW2Stambo, 2017-11-12:
Would it be possible to enable board alignment with inflight adjustments.
Especially on the pitch axis.
Then you could use a 3 pos switch to adjust up or down in one single flight.
Once it's flying level, land and save with stick commands, job done.
I know you can already do this with stick commands but you have to land each time.

Just a random thought I had while reading this thread.

ShawnzAZ, 2017-11-18:
I've been fighting this too.   No problem in angle / horizon but pos hold loses altitude unless I make the loiter circle HUGE.  Still dialing board alignment to make it work.

Part of me wonders if the PIFF tuning isn't a factor too because the plan is VERY numb in any stabilized mode.

shellixyz, 2018-02-07:
@ShawnzAZ If the plane is numb is stabilized flight mode there is clearly an issue with your PIFF tuning or your expo is too high.

shellixyz, 2018-02-07:
Would anybody be against RC adjustments for pitch and roll board alignment like @MW2Stambo described ?

digitalentity, 2018-02-07:
Changing configuration settings on the fly might be unsafe. These settings change transformation matrix that is applied to sensors and may cause instability.

Dronek, 2018-02-08:
> I've been fighting this too. No problem in angle / horizon but pos hold loses altitude unless I make the loiter circle HUGE. Still dialing board alignment to make it work.

gregorg, 2018-02-13:
this feature is the most important that causes users moving to ET Vector which has a "record level" feature, on the ground and inflight.
Actually I have to flight level with artificial horizon bar, guess how many degrees it is too up or down, land, connect to computer, adjust pitch angle in configuration tab by 1°, and redo, then adjust by 0.2° and redo many times... Anoying.

Pairan, 2018-02-13:
Thats exactly, what I want to get rid of if it is possible

Dronek, 2018-02-13:
> Actually I have to flight level with artificial horizon bar, guess how many degrees it is too up or down, land, connect to computer, adjust pitch angle in configuration tab by 1°, and redo, then adjust by 0.2° and redo many times... Anoying.

Sorry for keeping up a general discussion but why is it actually so crucial to get the aircraft leveled to a precision of 1/10 of a degree?
In other words: what is a perfectly leveled aircraft expected to do better than a roughly leveled one?

If it's about holding altitude: even a perfectly leveled aircraft will change altitude depending on thrust so you have to decide at which particular thrust you want to specify an altitude holding level...

When doing turns, the aircraft will lose altitude even if perfectly leveled if no additional thrust is applied, no? 

That's why inav and other autopilots provide altitude hold modes based on baro / gps. This modes manage pitch and throttle together.

Pairan, 2018-02-13:
As described before: 
When you DO fly level (passThr) and activate HORizon mode, the wing will commonly rise or nose down UNTIL you adjust the level of the Board! That's what i wanted to adress with this change / get rid of!

if find it hart to adjust as I need to fly it in stabilised mode to confirm the correct setting. there is no indication like "v Horizon needs to be here/ thats my level" or similar ... further, the adjustment of the board alignment will move the virtual horizon in the osd (or am I wrong on this one?)

_(btw PosHold on my latest flight was configured with altHold but it did loose alt on its 70m radius. )_

Dronek, 2018-02-13:
> When you DO fly level (passThr) and activate HORizon mode, the wing will commonly rise or nose down UNTIL you adjust the level of the Board!

In my opinion it would be more useful to address this issue with a kind of "attitude hold" (not to confuse with altitude hold) or "pitch hold" mode. It would be more versatile because it would allow to hold any pitch angle, e.g. for constant climbing.

To adjust your board alignment you could then apply "pitch hold" when the plane is flying leveled in acro/passthrough, record a log and afterwards you could see the difference in degrees between level and actual pitch. This would be your desired board alignment correction angle if you still need angle mode.

Pairan, 2018-02-13:
Well, if the board alignment should remain a manual task then please lets do it in the fassion of "autotrim" ...
activating a switch while i flies level in PASSTHR sets the alignment of the Board (Pitch) to the config and we're done! And it this is to much, then maybe displaying the current pitch (degrees) in the OSD to be able to reconfigre the board alignment!? This would at least help me pretty much ...

Recording a log, evaluating it and then figure what to do will sure make this task inconvenient for (normal) people who simply want to fly. Maybe I'm overshooting here but thats how it feels to me.

brat002, 2018-02-13:
So, the main reason of this hassle is an incorrect understanding of levelled flight of an airplane for iNav.
Plane flies levelled when can keep altitude and nothing more. But there is another problem of iNav: impossibility to trim plane by trims of RC in the flight. Only manually trimmed plane can correctly and effectively handle horizon.

Pairan, 2018-02-13:
Well ... I do trim out my plane manually as its the foundation of a well flying plane. It's what you do on every plane. Fly, trim with the rc and later (because of iNav) set that trimming to the rods! 

All the discussion is getting to no end. new Ideas and problems are blended in. Lets finish this main thing first, otherwise this will never end-

I personally would be thankful if we simply could display the board alignment in the osd, so I could be able to easy take over the values to the board alignment config. That should be easy! 

Something like this :
[image]

In the upper RIGHT you'll see **P** (for Pitch) and **R** (for Roll). These indicate how the board/Plane is currently aligned.

The task of setting a correct board alignment would be:

1) Fly PASSTHR
2) see the P value while flying levelled
3) land
4) set it in the config and be done with it!

After that, everything should be fine for THIS issue/thread. No additional hocus pocus fancy coding, no new complex routines. simply pass the current alignment to the osd.

Thoughts on that?

digitalentity, 2018-02-13:
Definition of flying "level" for airplanes is keeping altitude at certain angle of attack, certain throttle, certain air pressure and certain all-up weight. Change in any of the conditions will cause the plane to climb or sink.

I see no reason of aligning it with sub-degree accuracy. Next day you'll fly with a lighter battery and plane it won't be "level" anymore.

For this reason I don't think we'll have any automatic leveling for planes. However adding a numeric display for pitch angle in OSD seems a reasonable improvement.

brat002, 2018-02-13:
Maybe it is possible to link rc trims with board alignment values?

Pairan, 2018-02-13:
Brat, please make it another Issue!

digitalentity, 2018-02-13:
@brat002 if trims would be sent as separate channels - it would be possible. Unfortunately trims are offsetting the sticks and FC has no way of knowing if it's trims or pilot moving the sticks.

Pairan, 2018-02-13:
If this is available, I will happily make a video on it

brat002, 2018-02-13:
> if trims would be sent as separate channels - it would be possible.
Yes, I mentioned exactly that. Is it already possible?

digitalentity, 2018-02-13:
@brat002 no, it's not possible (hence the "would be") 😄 
Can you even send trims as separate channels?

shellixyz, 2018-02-13:
With OpenTX I think it is

digitalentity, 2018-02-13:
Or we could add inflight adjustments for board roll/pitch alignment. Will require some work to ensure safety since board alignment affects sensor readings but it should be possible.

Pairan, 2018-02-13:
now we are back at the statement from last week:

> Changing configuration settings on the fly might be unsafe. These settings change transformation matrix that is applied to sensors and may cause instability.

digitalentity, 2018-02-13:
@Pairan exactly. It's possible, but I have doubts it's safe to have as in-flight adjustment.

Pairan, 2018-02-13:
It's your call @digitalentity - I'm just looking to get it done in a way we all can head on with :)

@shellixyz you are right! I found this (perhaps outdated):
http://rc-soar.com/opentx/setups/overloading_trims/index.htm

It descrives how to free the trims (and then reassign them I guess) ... but this again is more a "how to program your OpenTX" instead of an iNav issue, right?

giacomo892, 2018-02-13:
Currently fastest way to trim "level" is via rc adjustments. Just rise the throttle when disarmed and trim using pitch/roll. It will automatically apply yaw rotations and will trim the correct axis (at least it seems). Doing it during flight will let to poor calibration since the accelerometer may be also off due to slow turns being done before the level calibration and due to other sources of noise.

gregorg, 2018-02-13:
@digitalentity , would you be OK to add in-flight adjustment for board roll/pitch alignment, @Pairan and me will test it on our own wings ?

Pairan, 2018-02-13:
I'm OK with that, but the OSD indicator should be present too ... a bit more control is never bad ...

shellixyz, 2018-02-13:
I can make the mod and provide custom hex files if you'd like to test.

Pairan, 2018-02-13:
at that time I'll need a hex for `MATEK405-OSD` ... and hopefully nothing weird happens while changing the values mid air ...

shellixyz, 2018-02-13:
There is another solution to correct the pitch for flying level that do not involve changing the rotation matrix in flight: a flight mode similar to `SERVO AUTOTRIM` that would use the baro to calculate the mean pitch correction needed during a short interval (5 seconds for example) then apply it to the rc commands as if it was the pilot. Then on the ground the pilot would could correct the board pitch alignment simply with the save stick combo. It would not allow to automatically correct the roll alignment though.

Pairan, 2018-02-13:
As my MATEK has no BARO I cant help here

giacomo892, 2018-02-13:
@shellixyz you can use navPos[2] as reference for the height. It works even without the BARO.

shellixyz, 2018-02-13:
@giacomo892 Hmm good idea. Would it be accurate enough without the correct board alignment as I guess it is using the accelerometer data ?

shellixyz, 2018-02-13:
Here is a custom version of 1.8.0 with RC adjustments for the board pitch and roll alignments: 
[inav_1.8.0_BOARD_ALIGNMENT_RC_ADJUST.zip](https://github.com/iNavFlight/inav/files/1720966/inav_1.8.0_BOARD_ALIGNMENT_RC_ADJUST.zip)

The adjustment function numbers are 21 for roll and 22 for pitch. You will need to set them through the CLI. Each adjustment "clic" will modify the alignment by 0.5°.

For ease of use you should configure an adjustment for another setting in the GUI then go to the CLI and display the adjustments config with the `adjrange` command. Then change the function number of the adjustment you created with the GUI. It is the number before last.

Example:
I created an adjustment for ROLL D when CH6 is between 1600 and 2000 using slot 1 via CH7:
`
adjrange 0 0 1 900 1600 0 2
adjrange 1 0 1 1600 2100 20 2
`

then you only need to replace 20 in the second line by 21 for adjusting the roll alignment or 22 for the pitch alignment:

roll alignment:
`
adjrange 0 0 1 900 1600 0 2
adjrange 1 0 1 1600 2100 21 2
`

pitch alignment:
`
adjrange 0 0 1 900 1600 0 2
adjrange 1 0 1 1600 2100 22 2
`

Test on the ground if the changes you are making with the adjustment channel are the ones you are expecting. Only adjust the alignment "clic by clic" (0.5° at a time). Use at your risk. I cannot guarantee this won't crash your plane/wing or make you bald.

brat002, 2018-02-13:
Could you build firmware for Omnibus f4 v5?

shellixyz, 2018-02-13:
@brat002 You mean Omnibus F4 v3 ? If yes then it is already included in the ZIP file.

shellixyz, 2018-02-13:
@Pairan 

> As my MATEK has no BARO I cant help here

I think it could also work with the GPS as it would use the ALTHOLD flight mode.

Redshifft, 2018-02-13:
I wonder if we have other issues here?
I have a plane with a high motor thrust line that really is a pain in passthrough due to massive throttle/ pitch changes, OK I only ever use Angle or Acro but in Angle that thing holds level so well you can not even land it (unless you point it at the ground) It would have to stall before it gives up level :)

If the problems are after switching between modes then I think it may be similar to the old issue some Quads had when after throwing them around in Acro Level goes all to cock ! 
Vibration and thermal drift on the IMU could be a factor.
I would mention although I have a Baro I have never used any Alt hold or anything else, why would I as when I let go of the sticks it just flies straight and level (in that angle mode)

shellixyz, 2018-02-13:
@Redshifft 
The horizon drift with accelerations not parallel to gravity acceleration will affect the plane when you switch in angle in any case. Just like when you stay in angle and do tight turns and then the plane doesn't fly level for some time. But I don't think it is the issue we are talking about here.

Concerning the ALTHOLD mode it is different from ANGLE mode even if your plane flies straight. It makes the throttle control automatic and changes the pitch stick to a climb/descent rate control. Totally different.

Feldsalat, 2018-02-13:
I do not like the current way of inflight adjustment. It is in my opinion simply not usable in a user-friendly way. I can not handle it properly, especially since I have no more switches available and the pots do not work that well. Therefore, I never use this possibility, although I would find it useful.
It would be easier to use the pots to adjust values in a predefined range.
Displaying pitch and roll angles in the OSD would be a very good thing and so far my favorite. 
Could also be helpful for other settings. For example: nav_fw_bank_angle, nav_fw_climb_angle ...

Redshifft, 2018-02-13:
@shellixyz  Strange, none of my planes loose level if I throw them around.

I referred to  Alt Hold just because I have never needed it and yes it's a "nav function" which applies Nav parameters/controls.

I mention all this  just to help understand why others have different results. I personally have not used Horizon in a Plane for years as I hate it, but would say It did not have a different P/R center to Angle (ok I may have not tested this in INAV)

If we look at this pitch level logically and we have a Plane with a large speed range it follows where you define the FC level is always going to hold "straight and Level" at the "balanced" wing incidence/speed.
It follows if you fly faster the Plane will generate more lift and climb and obviously the opposite.
Possibly I don't vary my speed as much as most but I never consciously think the plane is climbing now I am going faster ! I do however often think this thing won't land!!! Oh, I'm in level mode :) Used to get the same with APM FC's
Maybe we need to start asking what Planes/setups do these guys have? 
Whenever I have had to alter level it's never more than a degree or 2 as I set level at a "guesstimated" incidence on the bench, It's a much bigger job to get CofG and reflex correct on a Flying Wing for example. Also don't forget if you start getting your CofP (center of pressure) lift balance point a long way from your CofG you will always have a very speed/pitch sensitive pig of a plane.

digitalentity, 2018-02-14:
@shellixyz cool, lets test your mods for in-flight adjustments. Can you also submit a PR for review?

giacomo892, 2018-02-14:
@shellixyz I suppose for few seconds it will be accurate enough to tell you that you are flying straight.

Redshifft, 2018-02-15:
Anyone notice the similarity with this https://github.com/iNavFlight/inav/issues/2506#issuecomment-365784325
I wonder, also mentioned Inav after 1.7 had issues flying level for some, seems to me we have too many " Plane" issues and moves to change stuff ! Maybe there are underlying issues with these newer releases? just saying :)

Pairan, 2018-02-16:
@Redshifft  the other one is different and (I) assume it to be solved this weekend. But its about the altitude controller and settings handling PosHold and RTH

This is about ANGLE/HORIZON mode and keeping the fixed wing flying leveled while it does so in PASSTHR/MANUAL

davyorrfpv, 2018-02-22:
Any flying aircraft has a natural angle of attack when in flight. Correct pitch alignment is only to maintain that angle of attack at zero degrees which horizon and angle modes then seek to achieve. If you lower your speed you will lose height. If you increase your speed you will climb. This is not about maintaining 'level' flight but about zero pitch at the natural angle of attack for the aircraft in question. Having an osd readout of that angle will only give a guide for a given speed.

PR #3169 (2018-05-08): Add pitch and roll attitude angles in OSD
Closes #2384 

~Do not merge.~
closed 2018-08-04 by stale[bot]