iNavFlight/inav #2385: Fixed wing auto arm desn´t work properly on 1.7.3 CC3D/NAZE

OviV, 2017-10-22:
Setting fixed_wing_auto_arm by CLI manual description doesn't work on my CC3D and NAZE board (other arm options are working as expected - by stick and by switch). By my opinion the auto arm triggers on throttle min_check value instead of min_throttle. At this moment is the throttle value to high and the arm is rejected (the error LED blinks slow). Is it possible to correct this bug ? I don't try iNAV 1.8.0 and other boards.

oleost, 2017-10-22:
It's mincheck it will arm on. If it says min_throttle in documentation that is wrong.

digitalentity, 2017-10-22:
Then we need to fix the docs.

OviV, 2017-10-25:
By my opinion it is software bug, not a documentation issue, thanks for Your reply. I'll try it with SPRF3 board and iNav 1.8.x.

closed 2017-10-25 by OviV