iNavFlight/inav #2386: omnibus F4PRO INAV1.8RC1 broken LED

niko919, 2017-10-22:
upgraded fw from 1.7.3 to 1.8.rc1  and found that LED stripe on ws2811 is not working anymore on pin PA1. get back to 1.7.3 and led stripe is working

teckel12, 2017-10-22:
@niko919 Correct, you need to use the J1 port for LED strip with INAV 1.8+ (instead of PWM5). Basically, the pin moved to where it should have been all along (J1).

niko919, 2017-10-22:
@teckel12 then need to mention it somewhere. hope this information will be in doc when release will be ready. i`ve  just checked J1,  it didn't lit my led too. :-(. need someone else f4 pro user to check it.

teckel12, 2017-10-23:
I have two quads with the F4 Pro and my LED strips are connected to J1 and they works great with 1.8-RC1.  Just to be sure you have the right connection, J1 is labeled as the LED strip connection in the Omnibus F4 manual, it's a 3 pin micro mini JST connection. 

I'm in the process of making changes to the Omnibus F4 docs which will note this change.

teckel12, 2017-10-23:
@niko919 I believe I read that some OBF4Pro Chinese clones wired the LED strip pin incorrectly. If your board is from Banggood or the like that could be the issue. Mine are from http://www.readytoflyquads.com and the LEDs work fine.

niko919, 2017-10-24:
@teckel12 yep. my f4v2pro is from ali, and шёь shure it is not original ^-^.
i found the solution (http://forum.rcdesign.ru/f123/thread495417-18.html#post7154262)
but it`s a bit hard to do without magnifying glass.

is it possible to remap pins as on BF via CLI? or will it ever be added to inav?

thx for help.

fiam, 2017-10-25:
@niko919 next release (might be another RC or the final 1:8) will have a target with the ledstrip at the motor5 pin. We’ll eventually support resource remapping, but it’s not available at the moment.

niko919, 2017-10-25:
@fiam that`s great. thx for help.

commit 9d6363cb (2017-10-24): Add OMNIBUSF4PRO_LEDSTRIPM5

Same as OMNIBUSF4PRO target, but uses Motor 5 pin for LEDSTRIP.
Supports defective boards which didn't have the actual LEDSTRIP pin
correctly routed in the PCB.

Fixes #2386
PR #2397 (2017-10-25): Add OMNIBUSF4PRO_LEDSTRIPM5
Same as OMNIBUSF4PRO target, but uses Motor 5 pin for LEDSTRIP.
Supports defective boards which didn't have the actual LEDSTRIP pin
correctly routed in the PCB.

Fixes #2386
closed 2017-10-25 by niko919