iNavFlight/inav #2387: Omnibus f4 Pro Corner

wallbanger123, 2017-10-22:
What is the correct target to flash to this board?

Also, does anyone have correct instructions to connect a M8N GPS module including the mag?

teckel12, 2017-10-23:
@wallbanger123 The only difference between the v3 and pro target is that the Pro has an SD card (like the Pro Corners). It's also called the Pro Corners, so the target is the OmnibusF4Pro.

wallbanger123, 2017-10-23:
Thanks that is what I thought would be the case. The betaflight target is appended with SD so I wasnt entirely sure. 

Any thoughts on GPS and I2C compass? I haven't had any luck on UART6 for the GPS. I can't seem to find solid info on I2C connections to this board.

teckel12, 2017-10-23:
@wallbanger123 That's why I opted for the Omnibus F4 Pro. It's more mature and has known configuration. Going with the latest FC is always a pain. Many times it's not fully supported, buggy, or come to find out something is totally missing, like soft serial or I2C.

Sorry, can't help. This post is to help others considering the Pro Corners or any new FC.

teckel12, 2017-10-26:
@digitalentity I believe this can be closed.

closed 2017-10-26 by digitalentity