iNavFlight/inav #2388: INAV/OMNIBUSF4PRO 1.7.3 presets don't change smix

MW2Stambo, 2017-10-23:
As the title said, when I select and save any preset in the configuration or presets tab, then check the smix in the cli nothing has changed.
I assume this is due to the different smix format that v1.7 seems to use.
Old
smix 0 2 0 -100 0 0 0
New
smix 0 2 0 -100 0
Been messing with this all weekend trying to get a custom mix to save and it kept saying parse error.
Found the answer in a reply to a Painless 360 video.
Thank you Spencer Hayes :+1: 
(I am not smart enough to figure it out on my own)

DzikuVx, 2017-10-23:
Presets are not change `smix`. If you want to reverse any servo, you have to do it manually

closed 2017-10-23 by DzikuVx

MW2Stambo, 2017-10-23:
I don't think you understand what I meant.
I am not asking about reversing servos.
That can be done in the GUI.

Presets MUST change smix otherwise change from flying wing to airplane could not work.
The format for smix has changed however the presets may have not changed to match the new format, therefore if you change from default quad to airplane or flying wing there is presently no smix.
I had to do a custom one using the correct format.