iNavFlight/inav #2393: OSD Head Free and Altitude Hold are not flight modes

teckel12, 2017-10-23:
### Discussion

Currently with the OSD, Head Free and Altitude Hold are displayed in the flight mode field.  However, one could be in Acro, Angle, or Horizon mode and in Head Free and/or Altitude Hold.  Position Hold makes sense to be a flight mode, as it forces Angle mode.  But Head Free and Altitude Hold are not the primary flight mode but instead a secondary flight mode.

To me, it would seem more logical for Head Free to be a warning (especially with OSD) with the current flight mode still being displayed (Acro, Angle, Horizon).  Not sure how Altitude Hold should be indicated, not a warning and not a flight mode.  Maybe an indication along with the altitude display? Suggestions?

I have a few other issues with the OSD flight modes.  Like `RTL` instead of `RTH` as it's labeled everywhere in the code.  Also, `STAB` instead of `ANGL` as the mode is called and a few others.  I know some of these date back a long time, but the terms seem to be consistent everywhere else so why not with the OSD?

I'm just raising these for discussion.  The actual changes are not that difficult and I already have a development branch, but how responsive would others be with these changes?

fiam, 2017-10-24:
@teckel12 Altitude Hold is already displayed in the OSD_MESSAGES indicator when it can be combined with other modes. In FW, AH implies ANGLE so it's displayed in the OSD_FLYMODE indicator. See `osd.c` around lines 972 and 1326.

Now, for HF... I totally forgot about that one. I'll submit a PR for it before 1.8 final is released.

As for the other suggestions, I think keeping `STAB` is better than changing it to `ANGL` since users are already familiar with that term and CF/BF use `STAB` too. However, I'd be in favour of changing `RTL` to `RTH`

fiam, 2017-10-24:
@teckel12 I got permission from the BDFL to change `RTL` to `RTH`, so I'll submit another PR for that and it should fix this issue. Can we agree to keep `STAB` as it is?

teckel12, 2017-10-24:
@fiam I got a thumbs up from @digitalentity to change `STAB` to `ANGL`, so I've submitted a PR.

teckel12, 2017-10-24:
I'm closing this as it seems the changes address the issue.  I'll test the next build and see if there's anything missing or still an issue.

closed 2017-10-24 by teckel12