iNavFlight/inav #2400: Waypoints doesn't work on Omnibus F4 Pro

compuatic, 2017-10-24:
My experience trying to get waypoints to work with iNAV 1.7.2, 1.7.3, 1.8RC1

1. iNAV configurator doesn't support any waypoint entry, even rudimentary via command line or other.
2. I went to mobile flight in hopes of making it work and discovered that you need a BLE module to make it work on iOS which I got. The App itself works great and uploads the waypoints but something is broken between it and iNAV since the controller wouldn't recognize and/or initiate the mission.
3. As a result of going to a BLE module, iNAV config no longer connects wirelessly since it needs a virtual com port which BLE doesn't provide.
4. I then tried EZ-GUI which only works on an Android phone. Same result as MobileFlight. Everything seems to work but mission doesn't work in iNAV.
5. All my settings were shared here. No issue found.
6. Blackbox shared, nothing found on 1.7.3
7. Recommendation to try 1.8RC1 and blackbox shows that iNAV doesn't recognize the mission which is definitely present in the flash and can be ready back by both Apps. otherwise 1.8RC1 seems to work well btw so well done there.

The last option now is to compile MWP and run it on a linux box to try yet one other thing. I have no other reason to setup a linux machine and this seems like way too much hassle so I'm stopping here. Not going to try yet another piece of software on yet a fourth operating system.

This is not a complaint to you or any particular person. It is just sharing my personal frustration with the sitation. I realize that this is Open Source and that users have to pretty much live with what they get, and for that I thank the developers for their hard work and all the other great features that work well. I have however made the conclusion that for the time being, Waypoints Navigation has had minimal support and is currently not working, or barely working under very specific usage conditions for iNAV. At minimum, it certainly it user friendly and hasn't had the same level of focus or development. Even a simple way to setup via CLI would have been nice. This to me is the only thing keeping it behind the big boys (APM, Vector, etc). Keep up the good work but iNAV doesn't do waypoints for me for now.

teckel12, 2017-10-24:
It would seem to me that waypoints should be added via a transmitter switch.  But maybe I'm over-
simplifying it.

fiam, 2017-10-24:
I’m working on a desktop and eventually mobile app for waypoints. Once I have it working I plan to to make sure WP mode works fine and implement the missing WO types (there are like 5-6 types defined,  but only 2 are actually working).

stronnag, 2017-10-24:
WP works just fine; I fly them multiple times per week and have done since 2015-12-12 on a number of FCs.

stronnag, 2017-10-24:
Looks to me much more like an ezgui and / or MobileFlight issue to me.

fiam, 2017-10-24:
Just to be clear, I’m not saying that WP doesn’t work right now, just that I don’t have the right setup to test it at the moment. @stronnag uses WP regularly with his own software and it works fine for him,  so it’s very likely that the issue is not in inav but in the software used for setting the waypoints.

stronnag, 2017-10-24:
@fiam, looking forward to you implementing the missing WP types; mwp is ready.

Any for the records, ezgui works just fine too. If I upload a set of waypoints from ezgui, then disconnect ezgui and connect mwp (both BT/3DR), then in mwp (which has no mission loaded), I get:

[image]

i.e the result of MSP.WP_GETINFO, the FC tells me that there are 15 waypoints and a confirmed, valid mission in FC memory (@compuatic, these are the valids that are reported in the blackbox log in 1.8); 
mwp then automagically downloads, for visual confirmation (and I know why I can't arm), which is not surprising as I'm at home c. 600 from the field.

[image]

**So, mwp works, ezgui works**, the MobileFlight author has admitted that MobileFlight doesn't work with iNav at the moment; **there is no iNav issue here**.

compuatic, 2017-10-24:
@stronnag, EZgui upload and reads back just fine for me as well. The problem is switching into waypoints navigation during flight does absolutely nothing.

What mode are you in prior to arming? what controller are you using? could this be an omnibus F4PRO issue only?

compuatic, 2017-10-24:
And by the way mobile flight connected to inav just fine, it now won't connect to 1.8 only. This is separate from the waypoints issue.

stronnag, 2017-10-24:
You cannot prove that you have waypoints the the FC, in my case the result of MSP_WP_GETINFO demonstrates there are WP in the FC; your log showed there are 0 waypoints in the FC.

Mode prior to arm would be either acro or horizon. As WP demonstrably works for me on NAZE, SPRacingF3, SPRacingF3EVO, RMRC Dodo, OmnibusF3, Matek F405 AIO and Matek F405 OSD, I would be extremely suprised is it didn't work on another F4.

But that's irrelevant, your log showed there are no waypoints in the FC.

stronnag, 2017-10-24:
Apropros Mobile Flight, see https://www.rcgroups.com/forums/showpost.php?p=38492811&postcount=326

compuatic, 2017-10-24:
I will take a video showing uzgui downloading the mission file from the controller upon powerup with. O prior mission on ezgui. Hopefully that qualifies as proof of what I'm doing.

stronnag, 2017-10-24:
Do what you like, until your blackbox shows a valid mission you're wasting our time.

digitalentity, 2017-10-25:
Waypoint flight works for me. Please provide a blackbox log of a flight done in 1.8-RC2 where waypoint mission doesn't activate.

compuatic, 2017-10-25:
Here is a log from 1.8RC1
https://www.rcgroups.com/forums/showatt.php?attachmentid=10452190&d=1508733124

Stronnag says datalog confirms that iNAV sees no mission file even though ezgui confirms a valid mission is present via download off the controller. This is a plane not multirotor.

digitalentity, 2017-10-25:
Please note that disconnecting the power will erase mission from RAM. Flash may retain the previously saved mission, but it is not loaded back to RAM automatically.

compuatic, 2017-10-25:
Thank you for that info. I have noticed that the board will not arm if I upload a new mission at the field. The safety distance is set to the max of 6500 meters and the first waypoint is only a 100 meters or so away from home.

compuatic, 2017-10-25:
Konstantin,
   I got it working with ezgui. Your hints above was the ticket. The trick was to first load and save into non volatile memory, then load into ram without power cycling. Waypoints now work. Powercycling disables it again. Non intuitive but I am relieved.

Any reason why iNAV can not auto load from flash? Ability to do manually or via a mode would be much more convenient.

digitalentity, 2017-10-25:
You may have an old mission in non-volatile storage. Auto-loading mission may prevent you from flying. 
Actually you don't have to save mission to non-volatile memory at all. EzGUI will upload it to RAM and you can execute mission normally.

Alternatively, you can plan mission at home, upload, save to non-volatile memory, power off. Then you go to the field, power up the machine, load the mission using stick combo and fly it without having to use EzGUI:
[image]

DzikuVx, 2017-11-17:
Since ticket got into stuck phase, I'm closing it. If any new data appears, fee free to reopen

closed 2017-11-17 by DzikuVx