iNavFlight/inav #2406: fixed_wing_auto_arm broken since version 1.7.3

Arvycka, 2017-10-25:
fixed_wing_auto_arm no longer works since version 1.7.3 (neither does in 1.8-RC1, RC2)

Omnibus F3 pro
- Fixed wing
- fixed_wing_auto_arm=on
- Feature Motor_stop

Flash Version 1.7.2: Push throttle stick up - motor auto-arms and starts spinning
Flash Version: 1.7.3: Push throttle stick up - motor does not arm. (stick arming still working with fixed_wing_auto_arm=off)
Flash Version 1.8 RC1/RC2: Push throttle stick up - motor does not arm. (stick arming still working fixed_wing_auto_arm=off)
Flash Version 1.7.2: Push throttle stick up - motor auto-arms and starts spinning.

Has this option been removed?

digitalentity, 2017-10-27:
Fix ready. Closing

closed 2017-10-27 by digitalentity

digitalentity, 2017-10-27:
@Arvycka found a bug, thanks for reporting. Will be fixed in next build.

Arvycka, 2017-10-27:
Thanks. Sent you a beer :)

PR #2421 (2017-10-27): Fix non-working fixed_wing_auto_arm
Bug caused by pre-arming check that disabled arming if throttle is above zero.
Fixes #2406