iNavFlight/inav #2409: 1.8-RC2 - tz_offset yields odd results

teckel12, 2017-10-25:
I set the `tz_offset` value to -240 (4 hours).  The OSD was off by 16 minutes?  So, I set `tz_offest` to -256 and now the time is correct.  Something isn't right here.

Also, the file timestamp on my blackbox logs were still UTC time.

digitalentity, 2017-10-26:
Minutes offset here is unsigned which causes issues. @teckel12 in your case time is probably off by 1088 hours and 16 minutes. @fiam can you look into this?
```
static void dateTimeWithOffset(dateTime_t *dateTimeOffset, dateTime_t *dateTimeInitial, uint16_t minutes)
{
    rtcTime_t initialTime = dateTimeToRtcTime(dateTimeInitial);
    rtcTime_t offsetTime = rtcTimeMake(rtcTimeGetSeconds(&initialTime) + minutes * 60, rtcTimeGetMillis(&initialTime));
    rtcTimeToDateTime(dateTimeOffset, offsetTime);
}
```

teckel12, 2017-10-26:
Sorry that I live "left" of UTC. On this side of the planet, we refer to our time as -4 or -5 hours from UTC (depending on DST).

digitalentity, 2017-10-26:
Fixed by #2413

closed 2017-10-26 by digitalentity

teckel12, 2017-10-26:
Those of us this side of the world thank you.

PR #2413 (2017-10-26): Fix incorrect type for the minutes argument in dateTimeWithOffset()
minutes was unsigned, making it impossible to use negative offsets.
Fix actually found by @digitalentity, I just did the testing.

Fixes #2409
commit 0dd791a3 (2017-10-26): Fix incorrect type for the minutes argument in dateTimeWithOffset()

minutes was unsigned, making it impossible to use negative offsets.
Fix actually found by @digitalentity, I just did the testing.

Fixes #2409