iNavFlight/inav #2410: 1.8-RC2 - Home reset sets OSD distance to 0 but not altitude

teckel12, 2017-10-25:
With distance and altitude shown in OSD I launch and fly 40 feet and land.  Distance shows 40 feet and altitude shows -5 feet.  I do a home reset from a transmitter switch and the distance correctly changes to 0 feet.  But, the altitude still shows -5 feet.

digitalentity, 2017-10-26:
@teckel12 this is intended behavior. HOME RESET doesn't reset altitude.
I assume you don't want your expected landing altitude to be 5 feet above the ground.

teckel12, 2017-10-26:
If the home position is reset, shouldn't it be x, y & z = 0 to the new home position? I get that it would be -5 feet below the old home position, but the new home position should be 0 distance, 0 altitude because that's where the model would return. 

Doesn't the altitude reset when the home position is reset? In other words x, y, z should reset when home is reset. Are you just saying x & y is reset?

This makes me want the the feature that knows if the model is in the air. If you're in the air, sure, just reset x & y. But if you're on the ground, x, y & z should be reset.

digitalentity, 2017-10-26:
Altitude is not calculated relative to home. It's controlled by `reset_altitude` parameter. It had absolutely nothing to do with HOME position. Local coordinate reference is acquired once per battery (altiude may be adjusted if `reset_altitude` is set accordingly).

In case when you move the disarmed machine by hand after reference is acquired, 3D home position is moved, but altitude is still calculated relative to position reference. RTH will land machine according to recorded home altitude, regardless of it's absolute value.

HOME RESET mode does precisely what it should do IMO - it changes coordinates of home. Since we have no idea where the ground level is we can't reset altitude.

digitalentity, 2017-10-26:
Imagine resetting home position and altitude 100m above the ground. RTH will return to home position and descend fast to 100m above the ground. After that RTH will enter slow descend phase (since it assumes that 3D HOME is at ground level. You don't want the machine to descend from 100m at 25cm/s rate as it will take 7 minutes.

digitalentity, 2017-10-26:
Regarding OSD - it's a matter of representation. We can work in global coordinates all the time with altitude counted from MSL (which would make the code much much simpler by the way).

IMO current behavior is most logical and most safe. Ground level doesn't change much in most cases.

teckel12, 2017-10-26:
@digitalentity Makes total sense.  I would never purposely set home position while flying. But if I did, I can understand why it would only reset X&Y and not Z. 

With that said, once we know if the craft is in the air, if the craft is not flying (on the ground) home reset should zero 3D. If in the air, just an X & Y reset.

teckel12, 2017-10-26:
I'll defer this till when know the craft is in the air or not.

digitalentity, 2017-10-26:
Let's keep it open and notify that it's blocked by #2331

teckel12, 2017-11-10:
Was fly/no fly implemented? I got a notification on this issue.

closed 2021-11-07 by DzikuVx