iNavFlight/inav #2412: Adding position angle on OSD

IcemanDZ, 2017-10-26:
I think position angle is more important than heading angle, with position angle, I know exactly where my drone is,  and if I have high gain antenna, I can turn the antenna towards to the drone, so I hope next update someone can add ths feature.

digitalentity, 2017-10-26:
Whta's position angle and what's heading angle? What OSD element are you referring to?

IcemanDZ, 2017-10-26:
[image]
Pls check the photo, position angle is the angle that  the take off position as the original point, the north of the take off point is  0 degree, everytime you fly the drone, you just need know the north, then you can know exactly of the direction of the drone, and the heading angle is the angle where the drone heading, that is for fixwing, but for me , I think that is useless, in my opinion the position angle is more important than heading angle.

digitalentity, 2017-10-26:
I wonder how many users would want this feature. We already have 36 indicators in OSD which is quite a lot. I hope more flyers will drop by and share their oppinions.

IcemanDZ, 2017-10-26:
I think this feature is one of the most important feature in the OSD, especially for fixwing user, when the drone in a unfamiliar place, I can know where is my drone and I can puts me antenna toward to drone to get more stable video.

teckel12, 2017-10-26:
I do this with transmitter telemetry. To me OSD is while flying, not to locate a downed model. @IcemanDZ Are you running SmartPort telemetry? If so, my telemetry script does exactly what you want https://github.com/iNavFlight/LuaTelemetry

Dronek, 2017-10-26:
@IcemanDZ, have a look at "map mode" in MWOSD. I use it many times during each flight. It can be configured to show aircraft position relative to home as well as aircraft orientation relative to home. This way you can estimate where to point your directional antenne and whether the aircraft is out or towards home point.

Pairan, 2017-10-26:
commonly I use the home-direction-arrow to figure, where the Wing is and relativly where I am ... personally that helps me more than a indicator showing degrees (and I'm flying without telemetry).
But maybe that's just me

fiam, 2017-10-26:
@Dronek could you upload some captures from "map mode" in MWOSD? My searches have turned up nothing of interest.

ShikOfTheRa, 2017-10-26:
@fiam – great job on inav OSD…. Look forward to using it myself at some point. Send me pm for email if you need any Q about MWOSD functionality. 
I don’t have a video of it in operation, but just did a quick one with simulator so can get idea. Vid is crap really.
Those transitioning from LOS particularly seem to like it… feel the need to know where aircraft is. Never use myself though!

There are a few different map mode options, but this one shows main one:

1. Home is center of screen
2. icon shows position of aircraft in relation to pilot and direction of launch / arming (consider if no mag with fixedwing that have to get direction shortly after launch) 
3. icon shows direction of travel. As aircraft turns, font changes to indicate aircraft heading.

Easy to implement – but more memory / fonts/… 😊
[https://youtu.be/cduyvFrygew](url)

Dronek, 2017-10-26:
> @Dronek could you upload some captures from "map mode" in MWOSD? My searches have turned up nothing of interest.

The description of @ShikOfTheRa pretty much sums it up @fiam . Could not demonstrate it better than the creator of the OSD :)

It is actually one of the most useful features of MWOSD for me. The way I use it:
- decide in which direction you want to fly mainly (in most cases against wind, that is)
- point aircraft in this direction and arm (in map mode 4 this direction is set as "UP" on the map)
- during flight you not only can use map mode to see where the aircraft is but also if you stayed "on course" or if you are drifting sideways etc.
- I have it on a switch so I can toggle between "main" OSD mode and map mode. Lately I coupled the same channel with a cam switch so it switches to a downwards pointing cam in map mode.

ShikOfTheRa, 2017-10-26:
@fiam 

take a look here for code: 
https://github.com/ShikOfTheRa/scarab-osd/blob/master/MW_OSD/Screen.ino#L2155

mapmode case = 3 as described above.

Pairan, 2017-10-27:
>  wonder how many users would want this feature. We already have 36 indicators in OSD which is quite a > lot. I hope more flyers will drop by and share their oppinions.

**May I do a suggestion here:**
How about creating 3 screens for OSD (switchable through the OSD switch defined in the MODES)?

Lets say, Page 1 + 2 can be user defined while page 3 remains BLANK? Or we could have all three screens user defined as an alternative.

By hitting the OSD switch the screens could cycle through.

Maybe this could reduce the amount of indicators on the osd screens and create a cleaner screen which still holds all needed ( or maybe desired ) information/indication.

Dronek, 2017-10-27:
I suggest, you really check out MWOSD's features and code. Multiple OSD screens, map mode, etc. have been around for quite a long time and are working really reliably. It makes more sense to collaborate based on this instead of inventing all existing features all over again. Maybe MWOSD can benefit from this as well for those who keep using separate OSD boards (e.g. minimosd) instead of AIO FCs.

digitalentity, 2017-10-27:
@Dronek I agree. Porting some feature from scarab-osd is definitely a good idea. Not sure if code would be directly reusable though.

ShikOfTheRa, 2017-10-27:
Happy to help out with any questions or how things work. Love to see some of this implemented. Its an exciting time for iNav development. Hope you have plenty of free resources!
Unfortunately due to illness in family I have for some time now been unable to do much in way of coding. Maybe another couple of months to go

Some features. Not all may be applicable to iNAV as it supports many other FC types like DJI etc…
https://github.com/ShikOfTheRa/scarab-osd/wiki/Features

alexeystn, 2017-12-12:
I have one more OSD feature to suggest:
direction of ground speed relative to aircraft's heading, an arrow indicator.
Let me explain how it may work. My quad is heading north in AltHold mode, I hold pitch stick forward. But there's a strong wind blowing from east, so my drone is drifting north-west instead of flying straight to the north. 
In such case, a small "left-up" arrow may indicate the real direction of my movement.
It would be very useful for multirotors.
Sometimes, at high altitude it is hard to estimate, are you going forward or are you drifting away with the wind.

What do you think about it? Have anybody ever needed this kind of indicator?

ShikOfTheRa, 2017-12-12:
I have just a variation of this into dev for MWOSD. A second arrow to indicate wind direction. It comes up occasionally but to be honest only a couple of times and mainly for the APM guys who have been used to it..

I guess because GPS heading is normally what is reported on OSD and pilots compensates so it heads where they want. Many without realising. It does get interesting when windspeed > airspeed :) that sorts the men out from the boys

Dronek, 2017-12-13:
> direction of ground speed relative to aircraft's heading, an arrow indicator.

I'll throw in some discussion regarding compass heading vs. gps heading and how to calculate wind speed and direction. It is related to your request: [https://github.com/ShikOfTheRa/scarab-osd/issues/381](url)

As far as I know, inav combines compass and gps heading into one heading value, at least in fixed wings. This makes things more complicated.

ShikOfTheRa, 2017-12-13:
The changes so far have been for PX4, but I want to make them for iNav and other fixedwing too. So I guess for me, key question is does iNav MSP report the actual values or a calculated value via MSP. Have to take a look.

alexeystn, 2017-12-14:
Yes, the idea of wind indicator looks interesting!

I looked to iNav code to figure out how ground speed is calculated. I was sure, it was done by software as difference of current and previous positions. It was a big surprise for me to find that GPS module reports to FC already calculated ground speed. Moreover, ground course is also reported by GPS via ublox/nmea protocols! Besides, this value is also transfered in MSP_RAW_GPS packets, so it can be easily implemented both in iNav OSD and MW_OSD!

The only thing to do is to put an arrow on the screen, according to GPS ground course value.
Well, I'm not gonna wait for the next release, want to try to code that by myself now :)

MorrisR2, 2018-02-06:
Heading to the aircraft would sometimes be useful to me, an arrow or number showing which way the aircraft is.  Yes, there are already a lot of *options* you *can* choose to put on your OSD.  I wouldn't use most of them.  I'd use heading, I wouldn't use amp draw, or most others.  Each user can display the items they want, so screen clutter isn't an issue for the software (creating clutter is a user error).   One could argue that heading shouldn't be displayed *by default* to avoid creating clutter.

villanyie, 2019-12-06:
Hi, firts of all the mwosd its a great and usefull softwear in my case for stand alone gps osd. Its any chance for adding azimuth?

villanyie, 2019-12-09:
Ok, I figured out, the Angle to home is the azimuth just the reference is South, not the North. Now the mwosd is the perfect solution for standalone gps for fpv. Thank you.

danarrib, 2020-06-29:
Ok, I'm here just to ask you guys to not forget this.

[image]

It would be super nice to have another OSD element to show the AIRCRAFT DIRECTION from home point, so we can keep the aircraft flying on the correct path.

Imagine a situation when I point a super-high gain antenna on a single direction. It's not a tracker, it's a fixed antenna.
I must keep the airplane on the exact direction in order to keep the antenna pointed to it.
This OSD element would help a lot to do it.

danarrib, 2020-07-03:
There's a Pull Request for this feature. 
#5906 
Testers are welcome.

DzikuVx, 2020-07-07:
Will be included in INAV 2.6

closed 2020-07-07 by DzikuVx