iNavFlight/inav #2414: 1.8RCx - Mag unstability when SBUS activity begins after Tx power up

straferr, 2017-10-26:
If my radio transmitter is OFF and the quad is powered up, the compass works fine (as reported by the 3D model in iNav configurator and OSD heading graph).  The moment I turn on my radio ON, the compass goes berzerk.  The 3D model in the configurator spins/jerks around randomly, and the OSD heading graph shows similarly erratic changes.  When I turn OFF my radio, erratic compass behaviour persists.  Moving the mag away from the quad and changing i2c speeds have no effect. This problem does not show up in iNav 1.7.x.

Apart from this glitchy compass, all other sensors work just fine, and the quad can be flown in acro/autolevel modes with no problems.

It appears that when my Tx is still OFF and no SBUS conversation has been initiated between Rx and Tx, all is OK.  As soon as the Tx and Rx see each other and begin talking, the compass goes nuts.  I strongly think its a software issue since earlier versions (1.7.x) don't exhibit this behaviour. 

My hardware:
- Original Airbot F4 V2 (not a clone)
- Tx: FrSky XJT module
- Rx: FrSky XM+ on UART1 for SBUS
- Mag: GY273 module (HMC5883L) on i2c/UART3 (no i2c errors reported @ 400khz default)

iNav versions I tried: 1.8RC1 and RC2 (Omnibus F4 V3 hex) 

The problem is easily reproduced, and I've confirmed this several times, flashing between earlier/newer versions of iNav.

(I also posted abou this on rcgroups, but I think it got drowned out in the chatter.)

teckel12, 2017-10-26:
Is this the model with the current sensor? If so, you should use the Pro target

straferr, 2017-10-27:
Nope, its not the Pro.  This is the Airbot Flip32 F4 AIO V2.  Its got the SD card reader and onboard OSD, but the onboard BEC is linear, not the SBEC of the Pro, and its got no current shunt resistor ie. current sensor.

[image]

[image]

(I actually even bought and sent one of these personally to digitalentity some time ago for him to work with.)

digitalentity, 2017-10-27:
Can't reproduce on any of my Omnibus boards

straferr, 2017-10-27:
Lemme see if I can get a video up showing this issue.  I can repeatably and reliably get this to happen, just switching between 1.7.x and 1.8.x.  Will upload video here, or link to Youtube.

digitalentity, 2017-10-27:
Please also provide a dump of your configuration of 1.8.x

teckel12, 2017-10-27:
While it may have been called the v2, I believe if it has a linear regulator it's really the v1 and you should use the appropriate target. The v2 is what they called the Pro model originally, but that's Uber-confusing because there now v3-v5 versions of pro.

straferr, 2017-10-27:
@digitalentity:  Here is the dump of 1.8RC2:

`# dump

# version
# INAV/OMNIBUSF4V3 1.8.0 Oct 24 2017 / 07:06:11 (0f1975ce)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_UIB
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature GPS
feature RSSI_ADC
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
[... 549 lines cut ...]
set gps_sbas_mode = AUTO
set deadband = 0
set yaw_deadband = 0
set inav_use_gps_velned = OFF
set inav_w_z_gps_p =  0.000
set inav_w_z_gps_v =  0.000
set nav_extra_arming_safety = OFF
set nav_rth_tail_first = ON
set nav_rth_altitude = 500
set osd_video_system = 1
set mc_p_pitch = 110
set mc_i_pitch = 50
set mc_p_roll = 65
set mc_i_roll = 50
set mc_d_roll = 28
set mc_p_yaw = 90
set mc_i_yaw = 60
set mc_p_level = 30
set dterm_lpf_hz = 110
set rc_expo = 50
set roll_rate = 66
set pitch_rate = 66
set yaw_rate = 55
set tpa_rate = 10
set tpa_breakpoint = 1650`

straferr, 2017-10-27:
@teckel12 :  Which target should I use then?   I tried flashing the plain "Omnibus" hex, which resulted in a hung flash and I needed to use the FC's physical boot button to recover from.  The Omnibus F4 Pro hex results in the same mag instability.  There is no specific hex for the Omnibus F4 V2, so I had settled on the Omnibus F4 V3, whose namesake and physical lineage most closely matches the board I have.

straferr, 2017-10-27:
Here is the video I of the bug, as promised: 
[https://www.youtube.com/watch?v=32yMbrTKEjA](url)

digitalentity, 2017-10-27:
@straferr ok, this indeed looks odd. Will have a look

digitalentity, 2017-10-27:
@straferr can you check if it will also appear on slower looptimes?

digitalentity, 2017-10-27:
@straferr I was able to reproduce the issue, will now investigate

digitalentity, 2017-10-27:
Fixed by #2420. Connection to S.Bus activity is coincidental, issue is always present, but sometimes doesn't manifest itself

straferr, 2017-10-27:
@digitalentity : Awesome work!

digitalentity, 2017-10-27:
Fix merged. Closing

closed 2017-10-27 by digitalentity

teckel12, 2017-10-27:
@straferr It would seem that you should use the OMNIBUSF4 target.

straferr, 2017-10-27:
@teckel12 
Thanks, I'll give it a try when RC3 rolls out.  For now, 1.7.2 OMNIBUSF4V3 hex works beautifully for me.  Will make a note here if the OMNIBUSF4 hex does the job well on RC3 :)

digitalentity, 2017-10-27:
Most likely there won't be any RC3. If no critical bugs will arise we'll unroll final 1.8 release next week.

straferr, 2017-11-01:
@digitalentity 

I just tried 1.8.0 RELEASE on the same hardware that showed up this issue, and its all OK now!  Thanks DE!

@teckel12 

I tried the OMNIBUSF4 hex you suggested for this board, and I can now confirm its probably not the right hex (the onboard baro is not detected).   It seems the correct hex for this FC is indeed the OMNIBUSF4V3.  This is what I have just flashed on 1.8 and earlier versions.  Thanks for the input! :)