iNavFlight/inav #2415: Analog RSSI not reporting ANYFCF7 INAV 1.8.1Fie

markandkymward, 2017-10-26:
Firmware ANYFCF7_EXTERNAL_BARO, stable, RC1 and RC2.  Confirmed setup with BetaFlight

digitalentity, 2017-10-26:
Do a full erase and re-configure.

markandkymward, 2017-10-26:
Attempted, no joy.

teckel12, 2017-10-26:
I needed to manually reset my RSSI channel after upgrading to 1.8-RC1. All is good now.

digitalentity, 2017-10-27:
@teckel12 what did you have to adjust?
@markandkymward please provide a "status" and "dump" of your config.

teckel12, 2017-10-27:
@digitalentity I don't use the RSSI input port on my FC (could never get it working). Instead, I just sent RSSI from my transmitter back to my RX on (I believe) channel 13. I know, that's a lot of back and forth, but it works without a hitch.

When I upgraded to RC1 I wasn't seeing RSSI. I had to set it back to channel 13. At this point I don't remember if it was when I did a full erase and imported my dump or if it was just an upgrade (I've done so many in the last couple weeks).

markandkymward, 2017-10-27:
@teckel12 Probably what I’ll do too if I can’t get it working.

 @digitalentity, will do

markandkymward, 2017-10-28:
# status
System Uptime: 58 seconds
Current Time: 0000-01-01T00:00:00.000+00:00
Voltage: 15 * 0.1V (1S battery - CRITICAL)
CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280, MAG=HMC5883
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: Manufacturer 0x3, 7761920kB, 11/2014, v8.0, 'SL08G'
Filesystem: Ready
Stack size: 6144, Stack address: 0x20010000
I2C Errors: 1, config size: 3356, max available config: 32768
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = ADC 3
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 1, cycle time: 2004, PID rate: 499, RX rate: 111, System rate: 9

markandkymward, 2017-10-28:
I've noticed that when I turn off the transmitter, the VBAT reading goes to zero.

markandkymward, 2017-10-28:
dump

# version
# INAV/ANYFCF7_EXTERNAL_BARO 1.8.0 Oct 24 2017 / 07:18:06 (0f1975ce)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_UIB
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature RSSI_ADC
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
[... 433 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500