iNavFlight/inav #2416: VBAT not reporting correctly ANYFCF7, iNav 1.8.1

markandkymward, 2017-10-26:
Firmware ANYFCF7_EXTERNAL_BARO, stable, RC1 and RC2.  Confirmed setup with betaflight.

digitalentity, 2017-10-26:
Do a full erase and re-configure.

markandkymward, 2017-10-26:
Attempted, no joy.

digitalentity, 2017-10-26:
@markandkymward can you show output of CLI `status` and `set adc` ?

markandkymward, 2017-10-26:
@digitalentity thank you sir.  I am on the road until Friday night.  I'll do so as soon as I get back to the bench.  I did check status during testing and ADC0(1) was VBAT, ADC1(2) was current and ADC2(3) was RSSI.  I checked the wiring using a Betaflight flash and it appeared to be operating correctly after adjusting the scaling factors available in BF - although the voltage did appear to fluctuate wider than expected.  In iNAV, VBAT reports a value but the scaling  makes it appear as though it is a single cell (~3.9 v).  ADC0 on the ANYFCF7 has a 1/100 voltage divider.  Status reports the battery as 1S (using 4S).

teckel12, 2017-10-26:
@markandkymward Is this in the configurator?

markandkymward, 2017-10-27:
@teckel12 configurator 1.8.  The status I mean is CLI

markandkymward, 2017-10-27:
Betaflight configurator had entries for voltage multiplier and divider values. When I adjusted those in BF I was able to able to approximate the actual voltage.  The resultant voltage values would vary widely (2 or three volts). Makes me wonder about the ANYFC F7 ADC

markandkymward, 2017-10-28:
# status
System Uptime: 249 seconds
Current Time: 0000-01-01T00:00:00.000+00:00
Voltage: 16 * 0.1V (1S battery - CRITICAL)
CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280, MAG=HMC5883
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: Manufacturer 0x3, 7761920kB, 11/2014, v8.0, 'SL08G'
Filesystem: Ready
Stack size: 6144, Stack address: 0x20010000
I2C Errors: 2, config size: 3356, max available config: 32768
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = ADC 3
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 1, cycle time: 2001, PID rate: 499, RX rate: 111, System rate: 9

markandkymward, 2017-10-28:
# set adc
vbat_adc_channel = 1
Allowed range: 0 - 4

rssi_adc_channel = 3
Allowed range: 0 - 4

current_adc_channel = 2
Allowed range: 0 - 4

airspeed_adc_channel = 0
Allowed range: 0 - 4

#

markandkymward, 2017-10-28:
I notice that when I turn the receiver off, VBAT goes to zero.

markandkymward, 2017-10-28:
Closing.  Misread the hand-drawn pinout.  Had ADC order backwards..

closed 2017-10-28 by markandkymward