iNavFlight/inav #2418: release 2 servo pin for servo_gimbal

chan2328, 2017-10-27:
could you please help to release/remap pins for servo_gimbal.

# version
# INAV/BEEROTORF4 1.7.3 Sep  9 2017 / 10:49:35 (974440df)
# resource
IO:
----------------------
A00: MOTOR4 OUT
A01: MOTOR3 OUT
A02: FREE 
A03: SERIAL2 UART RX
A04: MPU CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: MPU EXTI
A09: SERIAL1 UART TX
A10: SERIAL1 UART RX
A11: USB IN
A12: USB OUT
A13: FREE 
A14: FREE 
A15: OSD CS
B00: MOTOR1 OUT
B01: MOTOR2 OUT
B02: FREE 
B03: BEEPER OUT
B04: LED1 OUT
B05: MOTOR7 OUT
B06: I2C1 SCL
B07: I2C1 SDA
B08: FREE 
B09: MOTOR8 OUT
B10: SERIAL3 UART TX
B11: SERIAL3 UART RX
B12: SDCARD CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C00: ADC CH1
C01: ADC CH2
C02: FREE 
C03: SDCARD IN
C04: FREE 
C05: FREE 
C06: MOTOR5 OUT
C07: MOTOR6 OUT
C08: FREE 
C09: FREE 
C10: SPI3 SCK
C11: SPI3 MISO
C12: SPI3 MOSI
C13: FREE 
C14: INVERTER OUT
C15: INVERTER OUT
D02: FREE

chan2328, 2017-10-27:
i can use variable switch (CH9 to control servos in gui as below, but there is no reaction on servo which connect to FC) .  what is the problem? 

[image]

chan2328, 2017-10-27:
I just tried BF, turn on servo_tilt, and use CLI command "resource" to remap the pin from motor output to servo ouput. like as;

resource motor 7 none
resource servo 0 b05
save

then the servo works very well.

so why dont inav just add that command "resource" to solve a lot of problem from motor and servo pin. 
or add a selection of output pin for servo output only if servo_gimbal is turned on. otherwise servo_gimbal become useless. if it is technical problem that cannot be solved i will say sorry.

digitalentity, 2018-02-12:
Duplicate of https://github.com/iNavFlight/inav/issues/1145

closed 2018-02-12 by digitalentity