iNavFlight/inav #2419: Distance to home disappears from OSD when distance exceeds 1000m (.62mi) 1.8 RC2

snowblind64, 2017-10-27:
After updating to iNav 1.8 RC2 when I fly more than 1000m, the distance to home in the OSD disappears. It reappears when I return to less than 1000m from launch. I have updated the fonts and did not have this issue with RC1.

I am running an Omnibus F4 Pro. Flying wing configuration.

Is anyone else having this issue?

digitalentity, 2017-10-27:
@fiam can you have a look?

fiam, 2017-10-27:
@snowblind64 It seems to work fine both in metric and imperial units for me. Does it disappear completely or does it just blink? If it's blinking, note that 1000m is the default value for the distance alarm, so you might be triggering it. Note that it can be configured or completely disabled from the configurator if it bother you.

Also, if you can post the video from the OSD output it might be helpful.

snowblind64, 2017-10-27:
@fiam The distance value disappears completely and does not return until I come back to <1000m from home. I have my distance alarm set to 2 miles and am using imperial units. I'm going to do some flying this afternoon and will do a bit more troubleshooting. If the problem persists I will post the DVR footage.

teckel12, 2017-10-27:
I just flew today with the latest development branch and when I exceeded 999m it showed the distance in miles. I never had anything flash or disappear. I have my max distance set to 1/2 mile so I wouldn't expect it to flash either.

fiam, 2017-10-27:
@snowblind64 Thanks for the clarification. I ran several bench tests, mocking the distance and making it go from zero to 3km and back and it worked fine. Hopefully the DVR footage will help us figure this one out.

digitalentity, 2017-10-28:
@snowblind64 I believe we need your full `dump` to see if this is a configuration error.

teckel12, 2017-10-28:
I'll do a field test of flying further than 1000m tomorrow and see what happens. I always have the DVR running so I'll be able to review it.

snowblind64, 2017-10-28:
Seems the issue is related to the alarm blinking functionality. Sadly it seems to be intermittent. I am not seeing the distance or altitude blink when I exceed their alarm value. Most of the time it just remains solid.

On the last flight of the day I exceeded the 400ft alarm altitude I had set and altitude disappeared from the OSD, then returned when I dropped down again. I didn't have any issues with the distance today.

Could the be related to the blinking functionality? Perhaps it is caused by the timing at which point the alarm value is triggered. Unfortunately my DVR wasn't recording at the time. I'll see if I can replicate the issue tomorrow.

snowblind64, 2017-10-28:
Here's a full dump of my config:

# dump

# version
# INAV/OMNIBUSF4PRO 1.8.0 Oct 24 2017 / 07:05:30 (0f1975ce)

# resources

# mixer
mixer FLYING_WING

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1100 2000 1500 -100 -1 
servo 4 1000 2000 1620 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_UIB
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature MOTOR_STOP
feature GPS
feature TELEMETRY
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
[... 479 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 45
set pitch_rate = 18
set yaw_rate = 2
set tpa_rate = 35
set tpa_breakpoint = 1650

teckel12, 2017-10-28:
If the Omnibus is a clone, it could be the clone OSD chip?  I've heard people with clone Omnibus FC having OSD issues.  If so, either turn off the max values or when they disappear you know you're over the limit.

snowblind64, 2017-10-28:
I bought this FC from RTFQ. Do they sell clones? I'll do a bit more testing today and try and get some DVR recording. It could be a hardware issue.

teckel12, 2017-10-28:
@snowblind64 That's not a clone then. Just a thought.

snowblind64, 2017-10-29:
Here is a link to DVR video of the issue: https://youtu.be/02xXpehBeQ4

See the video description for relevant times when issue occurs. Seems to affect anything that is meant to blink. I'm running the same board in a quad with Betaflight. Blinking of time and mAh consumed seem to work fine on the quad. Both are Omnibus F4 Pros from RTFQ. The one in the wing is an earlier version without the current sensor but with an SD card reader. It could be a hardware issue with my FC. I'll just turn off the alarms for now.

digitalentity, 2017-10-29:
@fiam FIY I've noticed that OSD is semi-transparent with latest code as if 7456 is only drawing in even or odd fields. In Eachine EV100 goggles this results in blinky OSD while in Headplay goggles it's perfectly fine.

teckel12, 2017-10-29:
Could it be the number of sensors on the same line? I remember some strange display issues when I had too many sensors turned on while testing. snowblind64 is displaying a lot in my opinion.

snowblind64, 2017-10-29:
I tried with a single item on screen. Same result. Looks like it could be something to do with the timing of the blinking and video frame rate, as it is not consistent. Sometimes the value being displayed stays solid, other times it disappears. I'm running NTSC video.

fiam, 2017-10-30:
@snowblind We can do some changes to the blink duty cycle and see what happens. I’ll prepare a custom build for you.

fiam, 2017-10-30:
@snowblind64 can you give this a try? Set some alarm to a very small value so you can easily bench test it and let me know what happens with the indicator.

[inav_1.8.0_OMNIBUSF4PRO.hex.zip](https://github.com/iNavFlight/inav/files/1426837/inav_1.8.0_OMNIBUSF4PRO.hex.zip)

snowblind64, 2017-10-30:
@fiam I'll give it a try tonight when I get home and report back.

snowblind64, 2017-10-31:
I flashed the hex. The entire OSD overlay turns grey obscuring nearly the entire view. Text is still visible over the greyed out screen but alarms are not blinking (I set the RSSI alarm to a high value so it would be triggered).

digitalentity, 2017-11-01:
Moving to 1.9. Will be a "known bug" in 1.8

fiam, 2017-11-01:
@snowblind64 I've just added support for software based blinking at the display level, so it can be used to perform blinking when the hardware support for it is not present or it's not working correctly. I've pushed the code to the `agh_sw_blink` branch, but I'm also attaching an hex to test to this post.

After flashing with it, you need to go into the CLI and type:

```
set display_force_sw_blink=ON
save
```

I've made the SW based implementation use a 1sec on/sec off cycle for now, to make it more evident when the SW implementation is being used. Please, give it a try and let us know how it works for you.

[inav_1.8.1_OMNIBUSF4PRO.hex.zip](https://github.com/iNavFlight/inav/files/1434211/inav_1.8.1_OMNIBUSF4PRO.hex.zip)

snowblind64, 2017-11-02:
Looks like that did it. Software blink works correctly. Thank you!

fiam, 2017-11-02:
Reopening since we haven’t pushed a fix to development yet

teckel12, 2017-11-02:
@fiam Is there a performance reason we can't make the software blink the default?  I'm sure it's *slightly* less performant, but it sounds like we could have problems with different hardware using the hardware blink.  It seems like this could avoid a lot of frustration and issues, unless the software blink is a large performance hit.

fiam, 2017-11-02:
@teckel12 Performance is a bit worse, specially because we can't use DMA to write to the MAX7456 in most boards and SW blink increases the number of writes significantly when an indicator is blinking. HW is definitely more desirable as long as the HW supports it. Depending on the feedback we get from users with 1.8, which uses HW blink with no SW blink option, we'll decide which one we'll use by default. If only a small amount of users are experiencing problems, I'd keep HW as the default.

teckel12, 2017-11-02:
@fiam Sounds like a plan.

teckel12, 2017-11-09:
@fiam I can confirm that with development build 1.8.0 values would go invisible instead of flashing. With release 1.8.0, the values disappear instead of flashing.

Was `display_force_sw_blink` added to 1.8.0 or is it only in the 1.8.1 build you posted here? It seems like a software option should at least be an option as the problem still exists.

fiam, 2017-11-09:
@teckel12 it was merged after 1.8, so it will be in the next release. Not sure about its default value yet.

teckel12, 2017-11-10:
I'll merge it into my local build. I have a fairly common setup, so I have to imagine many others have this problem. Glad you've already created a fix for it! Thanks.

teckel12, 2017-11-10:
Oh, I'll report back after I try your branch so we can close this.

digitalentity, 2018-02-25:
@teckel12 I'm going to close this

PR #2452 (2017-11-07): Add support for SW blink
Allows displays which don't support blinking or with faulty blink implementations (like some MAX7456) to. Each display drivers declares the capabilities it supports and the display system handles blinking for the drivers that don't support it.

To workaround HW issues in MAX7456, a new setting has been introduced which makes the MAX7456 driver report its capabilities without support for blinking, so SW blink takes over.

Fixes #2419
closed 2018-02-25 by digitalentity