iNavFlight/inav #2425: Flying Wing, motor will not arm in iNav 1.8.0

Cattlebaron, 2017-10-29:
The motor will not arm in version 1.8.0. (The Configurator is version 1.8.1) All pre-arming checks are shown with a tick. In the modes tab Angle, Horizon and Passthru go active when selected but Arm and Navlaunch do not. I have set to  nav_extra_arming_safety = OFF
The fixed wing is a TekSumo and the flight controller is an Omnibus F3 AIO. This plane in not fitted with a GPS module as it is intended for LOS flying only. The CLI Diff dumps is as follows:

# version
# INAV/OMNIBUS 1.8.0 Oct 17 2017 / 12:05:26 (0e5cdd03)

# resources

# mixer
mixer FLYING_WING

# servo

# servo mix

# feature
feature -BLACKBOX
feature -OSD
feature MOTOR_STOP
feature AIRMODE
feature PWM_OUTPUT_ENABLE

# beeper

# map

# name

# serial
serial 0 32 115200 38400 0 115200
serial 2 64 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100
aux 1 1 1 1300 1700
aux 2 2 1 900 1300
aux 3 10 2 1700 2100
aux 4 14 3 1700 2100

# adjrange

# rxrange

# master
set gyro_sync = ON
set gyro_hardware_lpf = 188HZ
set acc_hardware = MPU6000
set acczero_x = 135
set acczero_y = -63
set acczero_z = -148
set accgain_x = 4075
set accgain_y = 4093
set accgain_z = 4007
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set receiver_type = SERIAL
set serialrx_provider = SBUS
set small_angle = 180
set nav_extra_arming_safety = OFF

# profile
profile 1

set fw_p_pitch = 15
set fw_i_pitch = 40
set fw_ff_pitch = 15
set fw_p_roll = 15
set fw_i_roll = 30
set fw_ff_roll = 15
set fw_p_yaw = 50
set fw_i_yaw = 45
set fw_ff_yaw = 0
set fw_i_level = 15
set roll_rate = 40
set pitch_rate = 15

#

digitalentity, 2017-10-29:
Sorry, can't reproduce. Configurator doesn't report all arming blockers. OSD will give you a better hint what's not allowing you to arm

Cattlebaron, 2017-10-30:
I for one only ask for help when I have exhausted all other avenues thus your terse reply is of no help. I don't know if the screen shot will show but it tells me nothing helpful.

[image]

I would be quite happy to post the flight controller to you (at my expense) if it would help you to solve the problem because it is not unique, it has been reported a number of times on GitHub.

digitalentity, 2017-10-30:
This is really odd issue and it surely needs to be investigated. I should have an original Omnibus in my box, will gid it out and test your settings there.

However, since you don't have GPS and don't use any navigation modes (they are useless w/o GPS anyhow), it's safe to set `nav_extra_arming_safety` to `OFF`.

Can you describe actions you do when it doesn arm? 
Which flight mode do you activate when it doesn't arm?
Is your throttle stick at zero throttle?

Cattlebaron, 2017-10-30:
I hope you can sort the problem. I am using the Omnibus F3 flight controller for my Quad also using iNav and have no problems. I want to use the FC for stability control for my plane to replace a gyro which does the job. I use a Taranis TX X9D and a X4R-SB RX connecting the two via Sbus. I switch my TX on and once it is booted I power the FC. No matter what flight mode I choose the FC will not arm and allow the motor to run. The system is set to Angle mode by default ARMING is by use of a switch (SF). I also have Horizon mode but the system will NOT arm in this mode or Passthru mode either. The only other setting I have is nav_launch but that will never activate if the motor will not arm. I observe if I enter the motors tab, switch the arm to on, I can simulate the motor running but physically the motor will not start. Further all settings in the Modes tab work but the arm one will not change to a blue background when the switch is activated, the small position indicator moves as expected. Finally the Taranis will not boot if the throttle stick is not set to the minimum position.

digitalentity, 2017-10-30:
Ok, thanks for detailed description. So you are trying to use `NAV LAUNCH` and it doesn't arm then. Oh, by the way, launch mode will block the motor until throw is detected, so the machine may in fact arm. Do you have a beeper connected?

Cattlebaron, 2017-10-30:
Thank you for the reply. I am not using nav_launch at present. I hope to fly the wing first and then start the adventure of nav_launch as I have to have the wing launched (thrown) by a club member, hence the appeal of nav_launch. I checked my Receivers tab in the configurator  and I had set the low throttle to 1000, mid point to 1500 and full throttle to 2000 are these the values needed?
I have been experimenting with iNav for fixed wing since the release of 1.7.1 and it was all fine for a short time but then arming became a haphazard thing and failed totally shortly after.  I purchase some new Omnibus F3 controllers but have not had any success. The problem for users like myself is we don't know if it is the controller, the firmware or the configurator. I am examining a change to the Paris Air3 controller sold by CarbonBird as I feel the problem lies with the OBF#.

Cattlebaron, 2017-11-02:
Very short piece of information. A friend has his OBF3 installed in a wing and the motor arms. The only difference I can see that his is also fitted with a GPS module.

snowblind64, 2017-11-02:
One thing that prevented arming for me before I figured out what was happening is not allowing the accelerometer/gyro enough time to calibrate after plugging in your battery. I thought there was a problem with the FC but it turned out I just needed to give it a couple seconds sitting still after powering up.

nav_extra_arming_safety = OFF should allow arming regardless of GPS (fix,no fix, not used)

I'd say the issue may lie elsewhere. Have you verified the hardware is functional. Perhaps try flashing another firmware (ie BetaFlight) to verify.

Cattlebaron, 2017-11-02:
Thank you for the hints, I will follow up on them. I have already set nav_extra_arming_safety = OFF

sniperxnl, 2017-11-06:
I am having the same problem, i can spin the motor in motor tab. But when i arm nothing happen. I have a switch for arming, and when i arm that switch stay grey.
I use a Orginal Omnibus F4 pro v3 from airbot, using a gps and mag meter. I have set nav_extra_arming_safety = OFF for testing, but then i can;t arm also

digitalentity, 2017-11-07:
@sniperxnl please give us your config dump so we ca review possible configuration issue here.

sniperxnl, 2017-11-07:
Found it, my compass did it, it lichts up red and blue everytime, then i can't arm

Cattlebaron, 2017-11-07:
I am pleased for you, sniperxn, I hope that your can't was a typo and should read can. This unfortunately does not solve my problem as I have not installed a GPS/Compass.

Cattlebaron, 2017-11-10:
Seems I am getting no more information. Can I assume that following sniperxnl comment above the only way forward is to fit the flight controller with a GPS/Mag combo? I have read a number of times that on a fixed wing aircraft a magnetometer is not needed is this still so or will it too have to defined for the flight controller to arm?

MW2Stambo, 2017-11-12:
Don't know whether this helps but I have an omnibus F4 with Inav.
It will not arm if you have RTH selected for failsafe unless it has enough sats.
Normal behavior I expect but just a thought.

Cattlebaron, 2017-11-12:
Thanks for the note

digitalentity, 2017-11-12:
@Cattlebaron "extra arming safety" will prevent you from ARMing if you don't have a GPS lock and any of the following is true:
1) You have failsafe mode set to RTH
2) You have WP, PosHold or RTH mode on a switch
3) You attempt to arm with RTH, WP or PosHold modes activated

So far I am unable to reproduce your issue on any of my setups (with or without GPS). Since you don't need (and don't have) GPS it's safe to keep "nav_extra_arming_safety = OFF"

Cattlebaron, 2017-11-13:
Thank you digitalentity, I have not resolved the problem I am now of the opinion that there must be a problem with the Omnibus F3 AIO. I will try another brand of flight controller as I like iNav.

digitalentity, 2017-11-13:
@Cattlebaron please report your findings with another board - I'm curious if this issue is hardware-related or there is some obscure bug in software.

Cattlebaron, 2017-11-13:
I will report but it will take some time as I have to order the board, which I will do tonight, but it takes a week or two to arrive at my place. Please be patient I will report back as soon as I have the new board. I intend purchasing a AIR3 PARIS Sirius™AirHERO32 F3 SPI - BetaFlight - Sbus - damped Drone autopilot - flight Controller board. I have saved a full dump of the Omnibus F3 board saved 30 minutes ago. I can send it if it would be help, I also have a diff dump but it may not ave all the information you need.

beachbreak, 2017-11-24:
I too am having same issue:  using DYS f4 board (which is an omnibus I presume) I cant get the motor to respond.  I’ve tried extra_arming_safety off, shitting off GPS completely, etc...only thing left for me is to swap out the esc.  Servos are moving fine.

Cattlebaron, 2017-11-24:
I have tested 3 ESCs without success. My new controller is due to arrive on Monday, it is flashed with iNav by the supplier (CarbonBird). I hope to be able to report back by next Wednesday

beachbreak, 2017-11-24:
Never mind...I knew better than to whine before going through EVERYTHING...figured it out....6 Point acc calibration is very important!

Cattlebaron, 2017-11-24:
Thank you, I calibrated the accelerometer and iNav reported it as OK, I  recalibrate later tonight and observe the result. I carried out the calibration as the very first step, maybe I should do it last thing after first set up.

MW2Stambo, 2017-11-26:
Another thought, you said in an earlier post the motor will not run from the motors tab.
Do you have the right ESC protocol selected?
Also, in the modes tab, remove nav launch completely, I believe this will not work without GPS and the FC will probably not arm if it is present. (even if it is not activated)
Perhaps for now remove everything but arm, you can always put it back later.

Cattlebaron, 2017-11-27:
Thanks MW2Stambo, I will check out all of your suggestions but first I need to repair a plane that landed vertically at speed on the weekend.

lancaster100, 2017-12-22:
Cattlebaron
I have the OBF4 V2.1
I could not get the motor to spin - I had two obstacles 
With no GPS fix I needed to disconnect GPS
I found that selecting 'extra arming disconnect' was not sufficient to fully disconnect GPS - it must also be deselected in the config tab
Secondly some makes/types of ESC will not function with the board even though they function normally without the board
Red Brick 30A did not work , HK 20A worked OK

Cattlebaron, 2017-12-22:
Thank you to all who contributed, I finally received my PARIS Sirius™AirHERO32 F3 which is flashed with iNav for fixed wing. It worked straight out of the box with only one change necessary  to accomodate my Taranis TX. I tested every suggestion but the Omnibus F3 refuses to cooperate, it works fine with my quadcopter. Thank you all agin.

closed 2017-12-22 by Cattlebaron

shellixyz, 2017-12-22:
@Cattlebaron I did not follow this thread but you now can see what is blocking arming in the CLI with the status command.

CircuitBender, 2017-12-27:
i have currently the same issue beta testing a betaflightf4 board on inav with an FT Spear Fixed Wing
- motor spins in motor tab
- servos are working correct 
things to do:
- deactivate all gps related stuff
- recalibrate accelerometer
- check ESC - it works as standalone ESC just fine, which does not mean it will with a FC

CircuitBender, 2017-12-27:
ok ... proper ACC Calibration is key :)

Cattlebaron, 2017-12-28:
I am a bit, no make that a lot, out of my depth here. I have given up on getting iNav to work for flying wings. I destroyed a TekSumo yesterday after I thought I had it all solved but then a turn to the right ended in a full power dive into the ground.  I think CircuitBender must have jinxed me LOL

CircuitBender, 2017-12-28:
@Cattlebaron the dive in the ground will come soon enough for me
weird thing is I have now a complete reversed smix with just once 100 and everything else at -100 which makes it now work with full range of the servos in passthrough mode, but now the angle and horizon modes are fucked up

btw it is now the 2nd flying wing not working ... i think the difference to quads is that there has to be the aviation part right from the beginning (Center of Gravity, Travel of Servos etc) then setting up inav works kinda way but I am now fiddeling around with custom airplane modes since the flying wing setup seems not to work (everything is 3 times reversed and then reacts unexpected)

my dream is still an UAV with a BananaPi :)

CircuitBender, 2017-12-28:
btw testing the setup in my living room, I get Telemetry loss now and then from my X8r TX ... could be the BetaflightF4 Board in Beta / unstable release for inav

Cattlebaron, 2017-12-28:
I would not mind corresponding via email but not so keen on using the forum

MW2Stambo, 2017-12-29:
@CircuitBender You will get telemetry lost if your plane and tx are too close together.

Grumpyrse, 2018-01-03:
I had the same problem, now solved.
In inav settings change esc frequency from 400hz to 50hz, worked for me!!
Hope this helps

Cattlebaron, 2018-01-03:
Thanks there is so much to learn. I also have a thread on  IntoFPV Forum › General Discussion › 350mm+ Multirotors & Aerial Filming Platform, called iNav help line. I think the comments for help would be very useful for iNav 'head scratchers' I would be quite happy to shift your comments over, I will acknowledge each author.

DzikuVx, 2018-01-10:
For the future reference: https://www.youtube.com/watch?v=7MAdgGkBkXs

lancaster100, 2018-01-10:
DzikuVx
Very useful video- thank you

Cattlebaron, 2018-01-11:
DzikuVx
Thank you, most helpful and explains why I have lost a lot of hair scratching my head. Now please another video of why the motor won't arm when using iNav for a fixed wing model, this was actually my first question and still defeats me. I do not use a GPS or Compass for my planes as I fly line of sight.

bgat, 2018-05-17:
I don't know about you guys, but I always have to set min_check=1150 (via CLI, and then saved) before my wing will arm.

I suspect it's because the default min throttle sent by my Futaba t18sz is 1100, but I can't say for sure. I had hunted up the code at some point, but didn't take good notes...

Cattlebaron, 2018-05-17:
Thanks Bill for the comment/hint. I have virtually given up because various things I have tried have end in failure. I tried only two days ago and thought I have finally got it going. I use an AIR32 flight controller as It was advertised as a working bundle. The last attempt I change the ESC to a SimonK because of its higher frequency, it seemed to have fixed the problem but on the next day it refused to work again. Thus I will now set min_check = 1150 and hope for the best.