iNavFlight/inav #2426: Add SOFTSERIAL support for MATEKF405 targets

digitalentity, 2017-10-29:
Possibly on S5/S6 pins

chrishbz, 2018-02-19:
This is a great idea for an enhancement! But I would **not** suggest to use S5/S6 but instead RX5/TX5 for the following reason:
- Matek F405-STD is great for airplane pilots, especially when combined with Matek FCHUB-W when you need 2 motors and 4 servos.
- S6 (PA8) is intended to be used for Rudder (see official document: http://www.mateksys.com/?portfolio=f405-std#tab-id-6 on the right column)
--> therefore S6 as SOFTSERIAL would break the whole benefit for this board combined with inav on an airplane.
- RX5/TX5 are one candidate for SOFTSERIAL. They are also available on FCHUB-W (see http://www.mateksys.com/?portfolio=fchub-w#tab-id-2 )
--> therefore RX5/TX5 would give possibility for example for S.PORT without any further soldering, which would be great.

digitalentity, 2018-04-16:
#2842 merged. Closing

PR #2842 (2018-04-04): Add SOFTSERIAL support on RX4/TX4 for MATEKF405 target
Pull request for the softserial mod provided by @MATEKSYS in #2841.

Closes #2426 and #2841.
closed 2018-04-16 by digitalentity