iNavFlight/inav #2427: RunCam Split control not working in 1.8-RC2

sergeysi779, 2017-10-29:
Camera is not responding to corresponding channel inputs. It works with the same hardware on BetaFlight 3.2.1 - it starts/stops recording, changes modes, and activates WiFi.

Board is Matek F405-AIO.
Camera is RunCam Split (v1), firmware version 1.1.0.

Could someone check if something is broken or I've just misconfigured something in iNav configurato?

One thing I notice is in BF configurator UART speed is set to AUTO while in iNav it is set to 115200. I checked RunCam manual and seems to be correct value.

fiam, 2017-10-29:
I’ve just finished installing an RC Split in my dev FC. Will test it tomorrow.

fiam, 2017-10-30:
Just tested it in a split 2 and it works fine. Are you sure you have selected the right port? Baud rate doesn’t matter for the RC device, since it will always use the correct one no matter what you choose in the configurator.

digitalentity, 2017-10-30:
Most likely this is due to RunCam protocol change for Split2. @sergeysi779 can you update Split firmware?
@azolyoung may have a better suggestion.

sergeysi779, 2017-10-30:
@fiam Yes, it is UART3 both in iNav and BF. It works fine with the same settings in BF 3.2.1.

@digitalentity 1.1.0 seems to be the latest. At least it doesn't find any updates in RunCam Android app.

azolyoung, 2017-10-30:
hi, just talking this issue with @fiam, we found the problem is the #2369 has not merged in to 1.8 rc2, so cause the old firmware can't work with iNav, just merge the PR in to it will work fine.

azolyoung, 2017-10-30:
[inav_1.8.0_MATEKF405_build_for_rcsplit_test.hex.zip](https://github.com/iNavFlight/inav/files/1426469/inav_1.8.0_MATEKF405_build_for_rcsplit_test.hex.zip)
@sergeysi779 , i build the newest code from development that including the rcsplit1.0 firmware  and rcsplit1.1 firmware   support

fiam, 2017-10-30:
@azolyoung Nice catch, I didn't realise it was merged after RC2.

sergeysi779, 2017-10-30:
@azolyoung thank you very much! It is working now.

azolyoung, 2017-10-30:
My pleasure :)

fiam, 2017-10-30:
Closing this, since it's fixed in the development branch.

closed 2017-10-30 by fiam