iNavFlight/inav #2428: All flight modes enabled, regardless of RX values with 1.8 RC2 on Matek F405-AIO

ShawnzAZ, 2017-10-29:
I should have grabbed a screen shot, and I will if it comes up again.    I went to launch today in passthrough and I noticed that the control surfaces were reacting the the model pitching and rolling like it was in angle mode while I still had it in hand.

I aborted the launch and plugged into the configurator.  Upon opening the Modes tab, I could see the pointers moving as I commanded arm / disarm etc, only every flight mode I had added a range for was active.   So regardless of switch positions all of these were active / lit up blue:  arm, angle, altitude hold, position hold, RTH.

I disconnected, unplugged the usb and unplugged the battery.   I plugged the battery back in, re-plugged the USB and launched the configurator.  The model was now back to normal.

I haven't seen it happen again but I figured I'd report it since this could present a very unsafe condition where the model would not disarm.

digitalentity, 2017-10-30:
Thanks for reporting. Let's keep this issue open and see if we'll be able to reproduce it.

DzikuVx, 2018-05-09:
Looks like stalled issue, closing since we had no other reports

closed 2018-05-09 by DzikuVx