iNavFlight/inav #2429: Current sensor value jumping around (on SP F3)

Dronek, 2017-10-29:
Not sure if this is a hardware or code issue but it goes as follows:
Me and two other guys have been experiencing heavily fluctuating current reading in iNav when motor is running. This has been going on for some iNav versions (I started with 1.4) up to 1.7.3 (had no chance to try 1.8 RC yet). The fluctuation shows up in the minimosd as well as in the configurator on the bench. However, the mAh consumption reading is quite accurate so the amperage value seems to be fluctuating around the correct value. 
We all use sp f3 deluxe boards on different fixed wing setups. I measured the output voltage of the current sensor and found very stable values at any given throttle position. Technically the board should therefore receive a non-fluctuating signal at the "current" adc. 

Are there known issues with this particular ADC and inav for this board? Could it be a software issue?

teckel12, 2017-10-30:
What are you using for your current sensor? When I flew the SP Racing F3 I used a PDB with current sensor. Once calibrated, it was very accurate. This was thru about 1.7.2 when I got rid of my last F3 FC.

Dronek, 2017-10-30:
It is a CJMCU 758 with an ACS758LCB hall effect sensor. I already tried both outputs of the sensor-board with similar results.
Again, the output signal of the sensor **does not** fluctuate when I measure it directly. And the mAh consumption calculated by inav seems to be accurate, too. I don't see a reason to troubleshoot the actual sensor futher. 
I will do a blackbox log of this behavior.

DzikuVx, 2017-10-30:
@Dronek this is likely due to ground loop effect. Try different ground. I had similar situation once and it was due to wrong ground used

Dronek, 2017-10-30:
Thanks for the hint @DzikuVx . I already paid attention to avoid groundloops when setting everything up but it's a good idea to check and try different grounds for this sensor.

Dronek, 2017-10-30:
So I checked everything for ground loops and disconnected most of the components.
For this test I also set iNav to defaults and enabled only the more or less needed things in order to get this bench test running (I could probably skip some more).

- FC is powered by a UBEC
- current sensor is powered by 3.3V pad on FC and ground is connected to the minus pad of uart3 (it doesn't seem to matter which ground I use)
- servos and ESC are connected to FC via signal wires only (servos powered by separate 5V step down)
- apart from the crossfire RX there was nothing connected to the FC at the time of logging

The log: [blackbox_log_2017-10-30_175713.TXT](https://github.com/iNavFlight/inav/files/1428115/blackbox_log_2017-10-30_175713.TXT)

And the diff:
```
# diff

# version
# INAV/SPRACINGF3 1.7.3 Sep  9 2017 / 10:20:09 (974440df)

# resources

# mixer
mixer FLYING_WING

# servo
servo 3 1000 2000 1551 -100 -1 
servo 4 1000 2000 1439 100 -1 

# servo mix

# feature
feature -RX_PPM
feature RX_SERIAL
feature CURRENT_METER
feature AIRMODE
feature PWM_OUTPUT_ENABLE

# beeper

# map

# name

# serial
serial 1 2 115200 57600 0 115200
serial 2 64 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 1 5 1300 1700
aux 1 10 7 900 1200
aux 2 17 6 1800 2100

# adjrange

# rxrange

# master
set acc_hardware = MPU6050
set acczero_x = 64
set acczero_y = -40
set acczero_z = -258
set accgain_x = 4071
set accgain_y = 4063
set accgain_z = 4062
set mag_hardware = HMC5883
set mag_declination = 230
set magzero_x = -64
set magzero_y = -31
set baro_hardware = MS5611
set pitot_hardware = NONE
set min_check = 1010
set max_check = 1990
set rssi_channel = 8
set serialrx_provider = CRSF
set min_throttle = 1005
set max_throttle = 2000
set motor_pwm_rate = 8000
set motor_pwm_protocol = MULTISHOT
set align_board_pitch = -50
set align_board_yaw = 1800
set vbat_scale = 111
set vbat_min_cell_voltage = 25
set vbat_warning_cell_voltage = 30
set current_meter_scale = 285
set current_meter_offset = 1650
set servo_pwm_rate = 330
set small_angle = 180
set auto_disarm_delay = 0
set nav_extra_arming_safety = OFF

# profile
profile 1
```

krzysztofmatula, 2017-10-31:
Is the RX transmitting telemetry?
Do you have RX antenna connected? Does it change something if you move the antenna?
Can you try without the RX?

Dronek, 2017-10-31:
> Is the RX transmitting telemetry?
Do you have RX antenna connected? 
Does it change something if you move the antenna?
Can you try without the RX?

- yes
- yes
- unable to test in actual state (receiver is built into one wing and antenna glued in at its base)
- yes. I just tried without RX connected by spinning the motor through iNav configurator. Without logging I can see on the "setup" page that current is fluctuating as before. I can probably rule out any influence of the RX.

In my previous post I forgot to comment on the actual log. It is the "amperageLatest" curve where you can see the fluctuation. Following screenshots show transitions where throttle is raised from zero to approx. 20% and lowered from approx. 50% to zero, respectively. Obviously the amplitude of fluctuation is correlated with throttle (or motor rpm) and negligible at zero throttle.

I'm going to try and measure the sensor signal again with a simple scope.

[image]

[image]

krzysztofmatula, 2017-10-31:
Checking with oscilloscope on the actual sensor output could say something.
ACS758 datasheet recommends using RC filter on the output. 
I guess CJMCU758 may have it, but it could be worth double checking.

Dronek, 2017-10-31:
Thanks for your reply. I'm narrowing it down actually. With the scope I noticed an oscillation on the signal. My previous fault was measuring the signal with a multimeter which showed me a clean voltage. Now I built an RC low pass for the signal and the oscillation is gone, hurray !! :)

However, when hooking all up to the FC, I now have a drift in the current reading. May probably have to tune the RC filter with different components. I'm closing the issue because it seems to be not iNav specific.

closed 2017-10-31 by Dronek

krzysztofmatula, 2017-10-31:
Once figured out please report here what RC filter worked best for you.

Dronek, 2017-10-31:
> Once figured out please report here what RC filter worked best for you.

Of course. I just finished putting everything together again and recorded a log for the sake of completeness: 
[blackbox_log_2017-10-31_154051.TXT](https://github.com/iNavFlight/inav/files/1431102/blackbox_log_2017-10-31_154051.TXT)

This is how it looks now:
[image]

Finally I went with a 10 uF capacitor and a 75 Ohm resistor which results roughly in a 220 Hz low pass filter. Actually I took whatever I had lying around and many different combinations worked out equally well, including low pass at some kHz. I'm using a BLHeli_S ESC on this wing, so the "noise" on the power should be at a relatively high frequency. This could be different with other ESCs where you can actually hear e.g. a 8 kHz tone when the motor is driven.