iNavFlight/inav #2434: mmix & smix can not change any motor output and servo output ordering

chan2328, 2017-10-31:
whatever mixer type (quadx, custom, customairplane) i used in beerotor F4.  mmix and smix can not change   any output pin ordering, for my case, i connected 4 motors to output 1 to 4, and 1 servo to output 7 showed as picture. i have tried a lot of times to change the value of mmix and smix. base on https://github.com/iNavFlight/inav/blob/master/docs/Mixer.md .  it does not change anything in the output ordering. 
[image]

digitalentity, 2017-10-31:
`mmix` and `smix` can't change output ordering, they are designed to map mixer input (roll, pitch, yaw, throttle) to outputs. Order and functionality of outputs is board-specific.

chan2328, 2017-11-01:
@digitalentity thanks for reply, so, at this moment, the connection showed in above picture is not able to work at current inav firmware.

digitalentity, 2017-11-01:
@chan2328 what do you want servos for? I believe we can allow 8th output to be a servo on multicopters.

chan2328, 2017-11-01:
@digitalentity the attached picture shows what i designed, which is a camera mounted on a servo.  i want to see the view by changing my camera's angle with the servo when using GPSHOLD, or for landing purpose. 
[image]

digitalentity, 2017-11-01:
Ok, I see. I'll change the name of the issue to reflect the the problem better.

chan2328, 2017-11-01:
@digitalentity thanks for help.

chan2328, 2018-02-24:
@digitalentity any chance to change pin 5,6 to be servo out in release 1.9?

shellixyz, 2018-02-24:
@chan2328 Please test this build enabling servos to be used for multirotors on PWM outputs 5 to 8 and report your results in #2829:
[inav_1.9.0_BEEROTORF4.hex.zip](https://github.com/iNavFlight/inav/files/1754559/inav_1.9.0_BEEROTORF4.hex.zip)

digitalentity, 2018-02-25:
Fix merged. Closing

PR #2829 (2018-02-25): Allow use of servos with multirotors on BEEROTORF4 FC
Should fix #2434. It also fixes the timers for use in FW. Testing required.
closed 2018-02-25 by digitalentity