iNavFlight/inav #2436: Functioning setup for Omnibus F7V2 requested

VoicOfReason, 2017-10-31:
To make a long story short, I have a TBS G2 frame with TBS Endurance motors and a 4-in-1 ESC (Lumenier BLHeli_S 35A 4-in-1 12v / 5v BEC DSHOT ESC +Current Sensor). In Betaflight 3.2.1 it flew perfectly. Switching to iNav, I tried several ESC protocols, none of which would work. The motors would twitch or else quad would lift off violently and flip over. I tried to revert back to Betaflight, but from that point onward, the quad continued to behave erratically even in betaflight. So, I re-calibrated the motors and one of them literally started smoking. 

I since replaced the motor and upgraded the ESC to a Lumenier Blheli_32 4-in-1. 

Now, it flies again perfectly in Betaflight 3.2.1. So, I crossed my fingers and again flashed iNav (latest). This time I started with Multishot protocol. Quad zips up in the air a few inches and does a violent flip 360 degrees. I'm not messing with protocols anymore until someone can give me settings that are verified to work with a setup like this. Just to be clear, I have installed iNav and flown with it on 2 other quads I built. 

Some will claim that software settings cannot ruin hardware. I disagree (https://oscarliang.com/esc-on-fire-burn-smoke/). But, at the very least I need to be cautious. I can't afford to have to replace another ESC and/or motor(s). Below is a dump of the latest setup I tried. Maybe someone can give me some workable settings. 

`# version
# INAV/OMNIBUSF7V2 1.8.0 Oct 24 2017 / 07:24:22 (0f1975ce)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_UIB
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature MOTOR_STOP
feature GPS
feature TELEMETRY
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
[... 382 lines cut ...]
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

# 
`

DzikuVx, 2017-10-31:
Try one of Presets that is close to your setup. ESC protocol is the least important thing

VoicOfReason, 2017-10-31:
I tried that once already but it didn’t help any. What I’m looking for is workable, verified settings on this flight controller.

DzikuVx, 2017-10-31:
Flight controller has nothing to do. Really. And hoping you will find anyone alse with the same setup are rather slim. I strongly suspect 5" GPS Racer preset will be suitable

VoicOfReason, 2017-10-31:
Ok ill try that setup.

digitalentity, 2017-10-31:
`set gyro_hardware_lpf = 42HZ` this is where your violent shakes come from. 

Please understand that Betaflight has an advantage of being pre-tuned for racing setups, because it's not being used on anything else. INAV doesn't have this advantage - this is why we have presets.

VoicOfReason, 2017-11-01:
Ok. I still don’t get why my other quad with an f4 works fine with iNav
whereas this one with f7 won’t even fly.

VoicOfReason, 2017-11-01:
What should I set gyro_hardware_lpf to?

digitalentity, 2017-11-01:
>Ok. I still don’t get why my other quad with an f4 works fine with iNav whereas this one with f7 won’t even fly.

@VoicOfReason if you will post a dump of your F4 quad settings, I could get a hint what's different.

> What should I set gyro_hardware_lpf to?

For miniquads a value of choice is `256HZ`. Preset will handle that for you.

VoicOfReason, 2017-11-01:
Nope nope nope nope. It will NOT fly. I did all of what you said. I did 5" GPS racer preset, and I even changed lpf to 256. With the smallest throttle it violently flips over onto its back. There is something seriously wrong here. I need help. I'm not going to ruin my quad again with this. Is there anyone how has ACTUALLY flown on the F7 with iNav who can give me their settings???

digitalentity, 2017-11-02:
Flies fine for me. Flipping over usually means that either motor order or board alignment is incorrect. Since the board doesn't have an alignment arrow, our default board orientation might be different from Betaflight's. Please verify that model in the Configurator follows actual copter movements.

For the record. Just did a hover test on Omnibus F7 V2.

VoicOfReason, 2017-11-02:
Yes! That did it. Betaflight's alignment was very different than iNav. So, a few quick questions:
1. LED is not working yet despite me trying to configure in GUI. Any workaround for this?
2. I know this is a tuning issue, but it's twitch a bit when hovering. Which PID should be adjusted?
3. Throttle seems a bit mushy. In betaflight, the throttle felt more crisp and responsive.

closed 2017-11-02 by VoicOfReason