iNavFlight/inav #2438: TRAMP with soft Serial not working

TimMW1, 2017-10-31:
Using a SPRacingF3 Deluxe and TRAMP enabled on softserial 1, it appears that the TX line is driven high after the MCU sends the command.  On the scope I see the command come out with full p-p V.  But when the Tramp drives the response back, it can only drive to about 4.5v.   In general this means that the MCU is still attempting to drive a 1 on the wire.  Tramp worked fine with current clean flight and beta flight SW using softserial.  With that code I see full rail drive coming form Tramp.  As time permits I will attempt this with a hard uart.

DzikuVx, 2017-10-31:
There is a chnce it will work in 1.9 since port of CF/BF SoftSerial is scheduled for that release. But stil, SS are not considered as best choice for anything

TimMW1, 2017-10-31:
I do see the correct command written on the TX wire.  Just not switched to RX for the receive from the TRAMP.   Maybe it is time for a new board.  the SPRacingF3 only has 3 UARTs.  Im already doubling up the telemetry with USB.

HJOGermany, 2017-11-01:
In the new Matek F405OSD is the same Problem.
Softserial doesn't work so you can use the LUA Script for VTX. 
I think it would be fixed in 1.9.

stronnag, 2017-11-01:
Not quite, soft serial is not yet implemented on Matek F405, see #2426

ShaunDwyer, 2018-01-17:
I haven't got a scope out to confirm, but im seeing "not working" behaviour on omnibus F4 Pro v2 board.. 

Oddly, when i enable 'pit' vtx appears to turn off, but I dont seem to be able to change any other settings.

Also the status line is: * -- ---- ----

Cheers!
-Shaun

closed 2018-05-27 by stale[bot]