iNavFlight/inav #2443: Brushed motor quad unfunctional in 1.7.3

OviV, 2017-11-01:
Setting motor_pwm_rate at value greater 8000 : surprisingly the min_throttle and min_command must be zero - please remove propeller before setup ! In the BF/CF is 1000. Works with slow fly and smoother control. With aggressive flying the PID regulation breaks and the copter falls uncontrolled to ground. After flying I observed, that in the CLI is motor_pwm_rate  fixed at 490 after save and reset operation.

digitalentity, 2017-11-01:
Please format the issue according to the template. Also provide CLI dump and blackbox log.

DzikuVx, 2017-11-26:
Since there is no progress with this ticket, I'm closing it. If any new data appears, feel free to reopen. Thanks

closed 2017-11-26 by DzikuVx