iNavFlight/inav #2444: Still some issue with zero throttle before RTH failsafe in 1.8 RC2

rb1205, 2017-11-01:
Condition: SP3 EVO board, fully functional quadcopter with RTH failsafe procedure

**Test**
Take props away, power on the quad and the TX, arm the quad, raise thrust to about 50%, then engage Failsafe either by aux command or by switching off the radio. Observe how motors behave. Disengage failsafe (make sure it is in fact disengaged), zero the thrust, engage Failsafe again. Observe how motors behave.

**What happens**
When the failsafe is engaged with some throttle, during failsafe the motors will change their overall RPM. Instead, if the throttle was idle then the motors are kept overall idle as well. Additionally, if you move the quad up down to simulate a loss of altitude, if the thrust was zero before failsafe then the engine won't rev up to increase lift, probably resulting in a crash.

**What should happen**
Failsafe behavior shouldn't be affected by throttle position at all (except for failsafe_throttle, which has nothing to do with this). Controller should definitely increase overall RPM if the quad is losing altitude.

**Hypotesis**
If you try to move around the quad while it's in failsafe, you can see how the attitude PIDs (both rate and angle) seem to work correctly with and without RC throttle, albeit with much different range of intervention on the actual RPM of the engine. IMHO the problem lies with the althold PID, which is forcing the overall thrust to be zero just like a idle thrust RC command would do.

NOTE: I don't know if it's relevant, but i use nav_use_midthr_for_althold.

digitalentity, 2017-11-01:
Please provide your settings. We can't guess how your machine is configured.

rb1205, 2017-11-01:
# version
# INAV/SPRACINGF3EVO 1.8.0 Oct 24 2017 / 06:58:38 (0f1975ce)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_UIB
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature LED_STRIP
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
[... 432 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_rate = 36
set pitch_rate = 36
set yaw_rate = 26
set tpa_rate = 30
set tpa_breakpoint = 1600

digitalentity, 2017-11-01:
@rb1205 thanks! I assume you test at home without GPS fix, right?
Blackbox logs of your described scenarious would also be very helpful.

rb1205, 2017-11-01:
Tested both with and without GPS fix, same results.

Sure, I will make a blackbox log in the afternoon.

rb1205, 2017-11-01:
Here you go, blackbox with [50% throttle](https://vps1.belloli.net/public/50p_throttle.txt) and [with no throttle](https://vps1.belloli.net/public/no_throttle.txt) . Notice how the roll PID I term is building up in the 50% throttle log, and is forced to 0 in the no throttle one.

I also set nav_use_midthr_for_althold to OFF, got same results.

rb1205, 2017-11-06:
[Here's](https://vps1.belloli.net/public/moving_throttle.txt) an even clearer log of the bug: I arm, trigger failsafe mode with some throttle, I let the I-terms slowly build up then, at t=37.88 seconds **still in failsafe mode** I zero the throttle on the radio. The PIDs I-terms are immediately zeroed. I also tried this with airmode enabled, got same results.

Am I wrong in assuming that failsafe should completely ignore rcCommand values?

closed 2018-05-27 by stale[bot]