iNavFlight/inav #2446: Strange problem with iNav 1.8.0 buzzer and Matek LED-Stip (with buzzer)

jstremmler, 2017-11-01:
I have strange hardware/software??  problem with iNav 1.8.0, my OmnibusF4 V3 FC, and a buzzer on the Matek LED-Stip (RGB LED BOARD WS2812B & buzzer)

Here is the fault matrix: 

- LED strip and buzzer both conneced correctly to the board: buzzer beeps permanently  (beeper -all, CLI command has no effect!)
- LED strip disconnected + buzzer (the one on the LED strip) connected: buzzer works correctly !!!
- LED strip connected + buzzer disconnected: LED strip works correctly !!!
- LED strip connected and different !!! buzzer (not the one on the LED strip) connected: everything works correctly

Do you guys have an idea what causes this strange behaviour of LED's and buzzer in combination when both are connected to the Omnibus F4 V3?

HJOGermany, 2017-11-04:
Please tell us what for a buzzer/LED you use.
I've got nearly the same Problem with an Omnibus F3 Clone an a Matek Buzzer/LED Board.

Bobrcnut, 2017-11-06:
For what it's worth: I also use the Matek Buzzer/LED combo on my quads, which are all running either 1.7.2 or 1.8.0, and this problem does NOT occur on any of them.  The Matek board, in my case, is this one:
https://hyperion-world.com/en/p2662991-hp-fpled2812alm

digitalentity, 2017-11-07:
Ok, this seems this is hardware/quality issue, not the firmware issue. I believe we can close this.

closed 2017-11-07 by digitalentity

jstremmler, 2017-11-12:
Yes it (was) a hardware quality issue. 

I finally solved it! 

The Matek guys had connected (GND) and (Buz -) on the WS2812 board with a buzzer. And that explaines to 100% the above failure matrix. 
- so I desoldered the buzzer. 
- then soldered the (Buz -) cable directly to the buzzer 
- and then soldered the buzzer back to the board with only (Buz +) connected. 

And then of course everythig worked.

How can these buggers from Matek do such things???