iNavFlight/inav #2455: (help) Servo tilt, channel forwarding OMNIBUS F4

assael07, 2017-11-03:
I try to enable servo tilting and channel forwarding to control servo on my omnibus f4 PWM6 but servo is still not working. 
I use INAV 1.7.2.
Any help would be very useful. 
Thanks

ins2d, 2017-11-07:
Jump here mate: https://github.com/iNavFlight/inav/issues/2216
A number of people patiently waiting for a miracle to happen one day ;)

DzikuVx, 2018-05-09:
INAV 2.0 will solve this problem as long as target will have usable servo outputs. 
Channel forwarding and servo_tilt will be done on mixer level, not specially configurable, target dependent features. 

See https://github.com/iNavFlight/inav/blob/development/docs/Channel%20forwarding.md and  https://github.com/iNavFlight/inav/blob/development/docs/Servo%20Gimbal.md

Because so, closing this one. Thanks

closed 2018-05-09 by DzikuVx