iNavFlight/inav #2456: Softserial on the f4 omnibus for V3 does not work

lumiror, 2017-11-03:
Wanted to ask, does the softserial work on the F4 omnibus for V3? I tried to turn it on but it does not work. It does not appear in ports.

stronnag, 2017-11-03:
As pointed out on RGC, not (obviously) supported on V3 firmware:

````
#if defined(OMNIBUSF4V3)
#define SERIAL_PORT_COUNT       4 //VCP, USART1, USART3, USART6
#else
#define USE_SOFTSERIAL1
#define SOFTSERIAL_1_RX_PIN     PC8
#define SOFTSERIAL_1_TX_PIN     PC9

#define SERIAL_PORT_COUNT       5 //VCP, USART1, USART3, USART6, SOFTSERIAL1
#endif
````

lumiror, 2017-11-03:
That is why I would like to ask for support.

stronnag, 2017-11-03:
I know, so a more appropriate title would be "Please consider supporting soft serial on Omnibus F4V3"; then we don't repeat all the discussion on RCG about whether it's a bug, a configuration issue or a feature request.

lumiror, 2017-11-03:
O.K.

OloTka, 2017-11-05:
Hello,  I've got f4 pro v3 from airbot and soft serial on ch5 & ch6 works perfectly,  so it must be regular f4 I'd suggest.
Olo

lumiror, 2017-11-06:
Hello,if you allow softserial you got them in ports? I have the original
board.

http://www.readytoflyquads.com/flip-32-f4-omnibus-v2-pro

OloTka, 2017-11-06:
Looks like the same board, although I bought it from Airbot. This weekend, I added bluetooth to softserial and it worked perfectly. The pads I used ( V3 picture from your link) are above Moto_VCC ("Silk Prints") - bottom pad is CH5, upper CH6. 
Hope that helps.

lumiror, 2017-11-06:
I know where are ch5,6 items, i have the problem that i can not see softserial in ports

OloTka, 2017-11-06:
Which firmware did you load omnibusF4V3 or omnibusf4pro? I've got pro. Maybe that's the case.

lumiror, 2017-11-06:
So it helped I already have serialsoft in ports. thank you. Do you have ch5 bridged 1k resistor with ch6 as it is recommended? One more question I have attached to UART 6 and UART 3 GPS with magnetometer UART 1 is serialRX ch5 and ch6 will telemetry but i want to connect bluetooth to know where to connect it.Thank you

OloTka, 2017-11-07:
Glad it worked :) I'm using softserial for bluetooth, so no need for single wire. I don't have separate telemetry as don't need it. Try RCgorup to find solution.

OloTka, 2017-11-08:
Hello, I think that one could be closed as SS worked on your board finally.

lumiror, 2017-11-08:
Yes and once again thank you

closed 2017-11-08 by lumiror

DoctorFPV, 2017-11-20:
Not the case for me. I have omnibus nano V2 and can't get the soft serial to show up in ports

VoicOfReason, 2018-01-06:
I know this issue is closed but it isn't obvious to me how the issue was solved. Is the solution to flash the "pro" firmware on the non-pro board and/or add the resister? Does adding the resistor suddenly make the extra soft serial channels appear on the ports page of iNav, because right they don't appear at all on the ports page for me (Omnibus F4 V3 w/baro).

This post was ambigous:
"So it helped I already have serialsoft in ports. thank you. Do you have ch5 bridged 1k resistor with ch6 as it is recommended? One more question I have attached to UART 6 and UART 3 GPS with magnetometer UART 1 is serialRX ch5 and ch6 will telemetry but i want to connect bluetooth ,you can advise where to connect it.Thank you"

lumiror, 2018-01-07:
Yes you need to upload "PRO" firmware and between CH5 and CH6 resistor 1K. https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20Omnibus%20F4.md
By Bluetooth I wanted  to set PID parameters but eventually solved it over the transmitter so I did not need bluetooth