iNavFlight/inav #2457: Suggestion: quick switch while flying between RTH normal and RTH tail first

VoicOfReason, 2017-11-03:
Sometimes when I'm flying line of sight (with LED's mounted in the rear) I like to be able to RTH tail first. Also, when my kids are flying, it's disorienting for them if the quad turns around. In addition, without LED's visible, makes it hard to see it returning. However, when flying FPV, I definitely want RTH normal (for obvious reasons). It would be nice if there were a way to map this change to a RC switch. Will iNav support this currently? If not, I'd like to suggest that it would (maybe a new mode??)

teckel12, 2017-11-09:
When flying FPV, if I do a RTH I always take off my goggles. If not, it may land on my head if the home position is slightly off.

I like the tail first RTH for the reasons you identified, but in my opinion you shouldn't be flying FPV if you're doing a RTH anyway. I did a RTH and left my goggles on one time and I landed on the roof of my house about 20 feet from the actual launch position.

VoicOfReason, 2017-11-09:
I'm not saying we should leave our goggles on all the way to the ground. Even with a Phantom or Mavik, that's risky. But, if I'm out beyond LOS and I'm doing a RTH, of course I want to leave my goggles on in case there is something in the RTH path like a high tree or something (or a hill). Anyway, I just think that a mode switch would be helpful the reasons listed above regardless of how long a person ought to leave their goggles on.

VoicOfReason, 2017-11-09:
> That makes sense, where I live it's totally flat, so I can set the RTH
> height to 75' and it will never hit anything from any direction. It seems
> like a flight mode modifier could be incorporated.

Yeah! It would be great to be able to assign a RTH altitude to an RC (e.g.
Taranis) control knob. Just spin the knob to adjust based on location.
Maybe I should start a new thread.

closed 2018-05-27 by stale[bot]