iNavFlight/inav #2458: PPM / S-BUS on F1 (Naze32) not working

Ginganinja82, 2017-11-04:
### Board and Version
_Afroflight Naze32 Rev6.
_1.7.3 - NAZE - 2017-9-10 10:53 _

# version
# INAV/NAZE 1.6.1 Mar  9 2017 / 00:36:58 (6c878ae)

### Behavior
* PPM signal not responding on FlySky receivers. (Tested with FS-IA6B and FlySky X6B)
* BetaFlight and Cleanflight works fine (not a hardware problem)
* Tested with Two rev6 boards. 

Vanilla Firmware load no additional config.

stronnag, 2017-11-04:
1. F1 boards are no longer supported
2. PPM and Serial RX work for me on F1 
3. 1.6.1 is obsolete and no one is going to support it

sergeysi779, 2017-11-05:
I think there was a bug either in iNav or iNavConfigurator. PPM and PWM were mistakenly swapped. Try changing the setting to PWM or upgrade configurator and iNav.

digitalentity, 2017-11-07:
@fiam @DzikuVx can you look into it? I'll try to confirm with latest configurator as well.

DzikuVx, 2017-11-07:
Check what? INAV/NAZE 1.6.1 or 1.7.3? but ok, i will try to verify on 1.7.3

Ginganinja82, 2017-11-08:
Thanks guys. 

DzikuVx - I have installed the latest on both. Configurator(1.8.1) and firmware on 1.7.3. I was not aware that the Naze boards or F1 chips are not supported anymore. quite sad as it it still very cheap on ebay for budget builds.

Attachments for reference though... (this on Vanilla. No additional changes made) 

[image]
[image]

Schalonsus, 2017-11-09:
I think i saw some configuration issues with 1.8 configurator and 1.7.3 too. Use the 1.7 configurator

Ginganinja82, 2017-11-09:
Thanks! I'll try 1.7.2. Cannot find 1.7 on iNavFlight/inav-configurator release page..

danarrib, 2017-11-11:
I can confirm: On Naze32, INAV 1.7.3 (last version available for this FC), the Configurator 1.8.1 do not work. Using the Standalone 1.7.3 version of Configurator everything works fine.

jstremmler, 2017-11-12:
If I look on the screenshot of your configurator above you did not configure a PPM RX but a serial RX with IBUS. 

Did you then connect the IBUS signal of your RX correctly to pin 4 of the NAZE instead of pin 1? 

- signal of ppm RX is connected to pin 1 of the NAZE32
- signal of a serial RX is connected to pin 4 of the NAZE32

Ginganinja82, 2017-11-13:
@danarrib - THANK YOU ! Confirmed, 1.7.3 standalone works on PPM... happy days ! 

@jstremmler - Thank I tried both.. the screenshots are for the "missing" sliders when selecting i-bus as the serial input..

fiam, 2017-11-18:
Just submitted a PR for the configurator that fixes this problem: https://github.com/iNavFlight/inav-configurator/pull/303

closed 2018-02-05 by DzikuVx

asm7100, 2018-02-10:
In 1.8.1 just set it to "PWM RX INPUT" if you want to use PPM