iNavFlight/inav #2459: iNav 1.7.3 FW maintains a turn even after sticks are released in Angle mode

ArxangelRC, 2017-11-05:
Hey guys, I may have a possible problem on an iNav build, and would appreciate some info on the matter, as I did not get any answers in RCGroups, so decided to post here as well.

I recently flew a 3.5kg V-tail plane with an Omnibus F3 Pro on it, running latest iNav at the time I setup the board, it was 1.7.3. Did the V-tail mix from Painless360 and only had to reverse one aileron to make everything work and stabilize in the correct directions.

Had to switch off the pitch2thr mix as it was constantly nosing down the plane and trying to plant it in the ground, so if I may suggest something to the devs - make it so that this mix is switched off when a non-flying wing type of plane is selected, as it could cause all sorts of issues! Nearly crashed a few times because of it!

Anyway, having sorted a lot of the programming I got to relax and fly the plane for a bit to test it out, and noticed a weird behavior in Angle mode! My trims on the radio are 0'ed out, and I have adjusted the servo mid positions on the Servos tab in the GUI to allow level flight in passthru mode. When in Angle mode the plane flies straight and true if I don't touch anything, or if I make small changes in the course! However, when I do a long 180 degree turn using ailerons and rudder, and then let go of the sticks, the plane doesn't return to level, but keeps turning in that same direction, and I actually have to push the rudder and aileron sticks in the other direction a good deal and stay like this for a few seconds, and then when I let go of the sticks it would get to level again!

Any chance this is normal Angle mode behavior, which I somehow doubt, or is there something wrong somewhere in the code, or possibly with my plane?

Any insight into this would be much appreciated! Thanks in advance.

brat002, 2017-11-05:
I also have such issue on all my planes. Would be great to know how to fix it.

digitalentity, 2017-11-05:
This is a known issue and currently there's no way to fix it (I am working on it though).

ArxangelRC, 2017-11-05:
Thanks for the quick reply! Will be waiting for 1.9 then! :)

fiam, 2017-11-05:
@digitalentity is this caused by the horizon drift?

digitalentity, 2017-11-05:
@fiam yes. During a balanced turn acceleration vector is ~ equal to G but is not pointing downwards which confuses the firmware into thinking that "banked is actually "level".

OlivierC-FR, 2017-11-05:
I don't get this, is that only on planes with a rudder ?
I never noticed such behaviour but I only have flying wings with iNav.
Well if it's also affecting planes with only elevons what worries me is if I never noticed this, because I fly in angle mode a lot.

Dronek, 2017-11-05:
> I don't get this, is that only on planes with a rudder ?

No, I have been experiencing this on my first and second flying wing with inav.
I fly ACRO most of the time so I don't really care about ANGLE mode but it is annoying to have the plane drifting in ALTHOLD after doing a constant turn.

digitalentity, 2017-11-06:
It depends on the setup and flying style actually. It's most evident if the turn is slow and long and the plane sinks a bit during the turn. I notice it on some setups and don't notice on others. If you accelerometer calibration is off it may also make things worse (or better). It's really complicated that's the main reason it's tricky to fix.

digitalentity, 2017-11-06:
You can make the effect less severe by lowering `imu_dcm_kp` quite a bit. But bear in mind that it will also make the firmware take longer to recover from the effect.

ArxangelRC, 2017-11-06:
I'd rather wait for a proper fix than mess with params I don't understand! :)

ArxangelRC, 2017-11-06:
Is it know if this behaviour also occurs in RTH or the other auto modes?

OlivierC-FR, 2017-11-06:
I find this one puzzling, I've never noticed this horizon drift issue, I fly iNav on a large flying wing in angle mode, most of the time, maybe like 90%. I only switch to acro mode for few stunts and landings.
I also have iNav on a small wing, mostly flown in acro mode, so it's unlikely I would notice an angle issue on this aircraft, but for sure she comes back in RTH mode really well.
Next time I fly the large wing I'll make tests, what is worrying me is not the issue itself, but the fact that I never noticed it :)
If it happen on this plane.

Dronek, 2017-11-06:
> It's most evident if the turn is slow and long and the plane sinks a bit during the turn.

In such a condition the resulting absolute g-force can quite possibly be equal to standard gravity for the sensor itself while not pointing towards ground but towards the bottom of the banked plane. 

This most likely results in a similar ACC sensor reading as in leveled flight. But there should be a difference in GYRO reading. In leveled flight, all gyro axes should read more or less zero deg/s while in a turn one or more gyro axes should read non-zero values. This fact might be helpful to rule out false interpretation of leveled state.

digitalentity, 2017-11-06:
@Dronek actually this might work if we have a pitot sensor. Knowing airspeed and angular rates it would be fairly easy to calculate and cancel out the centrifugal acceleration. I need to give it a thought.

ArxangelRC, 2017-11-06:
The ArduPlane firmware for certain does not suffer from this issue, and I have quite a few planes equipped with that, and none have a pitot tube! Since you calibrate the ACC once so it knows what is level, isn't it easier just to have that calibration stored as a constant, and to have it revert to that the moment you let go of the sticks?

Dronek, 2017-11-06:
> isn't it easier just to have that calibration stored as a constant, and to have it revert to that the moment you let go of the sticks?

This could work in angle mode but autonomous modes would be still affected unless you substitute "let go of the sticks" by something else. In acro mode letting go of the sticks doesn't necessarily mean leveled flight, so there would be no mechanism to revert a slanted AHI back to normal unless doing a constant turn in the opposite direction (as it works now).

It's a good idea to compare with arduplane. I still have a gut feeling that "early" versions of iNav (like 1.4) didn't exhibit this issue but I could be wrong.

ShikOfTheRa, 2017-11-06:
I was looking at this with original multiwii which had similar issues. I'm guessing the code is significantly different now but I had some degree of success. I was using change in MAG heading  over time to create a value that varied the trust of gyro data vs acc data and length of validity of acc data. It didn't stop the drift when in turn as I had no compensation, but recovery was improved. 
I didn't appreciate at that time how much current was affecting the mag sensor. And I had significant variation from sensors being impacted by vibration. Some appeared really bad - the vibrations were increasing the perceived gravitational force in the turn. 
Lost interest as migrated to mainly fly acro but it's always bugged me. Mainly because OSD horizon accuracy stands out. 

Agree - worth having a look at what other Open Source projects do.

digitalentity, 2017-11-07:
@ArxangelRC you are getting it a bit wrong. Accelerometer measures *acceleration* not *levelness*. Acceleration is composed of gravity (which is constant and indicates level) and motion acceleration (speeding up, turning etc). Acceleration from motion can't be easily calculated. 

APM is using GPS to attempt to compensate for this acceleration from motion. #1481 is an attempt to do something similar, but it's not working yet.

OlivierC-FR, 2017-11-07:
I get why it's so hard to fix, if the turn is really slow and smooth all 3 gyros are almost at 0°/s2.
What about "detecting" the return of the roll stick to center, that'd trigger a reset, as in, cancel the gyros part in the attitude estimation ? (I'm probably not using the correct words here, plus it's been a long time since I looked at the PID code, was also Multiwii era...)

ArxangelRC, 2017-11-09:
I really hope you fix it guys! ArduPlane has it solved and the planes are a dream to fly in FBWA mode! :)

compuatic, 2017-11-21:
Agree, I also encounter this issue almost every flight now. Now I know why the plane has always had this "slightly drunk" behavior. Never quite flying level after letting go of the sticks. Sometimes it's quite violent and needs full counter stick for a few seconds. When it's bad you can clearly see the artificial horizon crossing the real horizon by 20-45 degrees.

compuatic, 2017-11-21:
Could this happen during a long rth command?

Feldsalat, 2018-04-23:
Is there a solution to this old  horizon drift problem in sight? Sometimes it's almost impossible to fly in Horizon Mode... :-(
17.04.18:
[With Inav 1.7.3](https://youtu.be/Zg_YElIAMsM)
(POSHOLD always + ALTHOLD)

Sometimes in addition to the drift ALTHOLD works very badly. Otherwise, ALTHOLD usually works better.
18.04.18 same model updated firmware:
[With  Inav 1.9.0](https://youtu.be/fpWqB5Lpu_M)
(POSHOLD always + ALTHOLD)

Ety-T, 2018-04-25:
I'm pretty sure that this issue was almost unnoticeable in 1.6.1 version. Now I moved to 1.9 and I feel very uncomfortable with this issue. Now thinking about returning to 1.6.1

Kaffiflight, 2018-05-08:
Hello, I guess it´s not just the horizon that is affected. I guess it´s all axis. I think the artificial horizon is just where you can see it obviously first. When going nose down for more than a few seconds, the plane also tends to go to the ground, after the sticks are released to level flight. And vice versa. You can notice that behaviour in Horizon flightmode. I guess it´s always happening when the board wants to level the plane but "does not know" where level is. It seems always the latest flying position changes the zero position. After some time of counteracting manually in the opposite direction, it slowly drifts back. I don´t use ANGLE, I have this in Horizon mode.

IvanBiv, 2022-04-24:
I'm puzzled by the horizon drift issue.
I looked at the INAV source code and it seems to me that the error is accumulating at the level of the wind estimate calculation functionality.
I have a plane on Matek F411 controller. I constantly observe the deviation of the horizon - the problem of horizon drift.
I want to help solve the problem. I am a software developer. To quickly dive into the subject area I need a developer who has experience with INAV source code, its debugging, testing. Together we are more likely to solve the problem.
Who can I work with?

DzikuVx, 2022-12-04:
done with INAV 6 new attitude computation

closed 2022-12-04 by DzikuVx