iNavFlight/inav #2463: SBUS doesn't work at Omnibus F4 Pro Corner

MightyFlights, 2017-11-07:
### Board and Version
_Omnibus F4 Pro Corner from Airbot
_Version of INAV used goes here_1.8

_Use CLI command `version` and put its output here_
INAV/OMNIBUSF4PRO 1.8.0 Nov  1 2017 / 06:34:32 (912d1315)

### Behavior
* _Describe the problem_

SBUS FRSKY X4R all channels are 1500 and not moving
Same config and settings with Cleanflight FW works very well
SBUS connected to UART3 

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

https://pastebin.com/bsQFMijC

digitalentity, 2017-11-07:
On OMNIBUSF4PRO board inverted S.Bus is only supported on UART1

MFlghts, 2017-11-07:
This is strange. There is no inverter on UART1. But there are invertors at both UART3 and 6. Please check documentation for Pro Corner.

Anyway as I mentioned it works with Cleanflight FW at Uart3. Why can't you make it also work with Inav?

digitalentity, 2017-11-07:
Then we need a separate target definition for Pro Corner which is apparently a different design than other Omnibus Boards.

MFlghts, 2017-11-07:
Will be this issue fixed in a new FW, or you can share a patched FW now?

If you will support this board can you please also add I2C support? It's not mentioned in the manual. 
Thank you.

digitalentity, 2017-11-07:
It's hard to tell. We've just discovered the issue with S.Bus on UART3 thanks to you.

MFlghts, 2017-11-07:
It says that inverter at UART3 can be switched off by pulling PC9 as low. Can this help with I2C support?

digitalentity, 2017-11-07:
No, since for I2C pin has to be bi-directional. Inverter is an uni-directional device regardless of being turned on or off

MFlghts, 2017-11-07:
I see. Thank you for the explanation. They also suggest to desolder inverter from the board if it's not needed. Will that help?

ShikOfTheRa, 2017-11-07:
Just checked diag. can confirm UART3 and 6 have inverter. UART1 does not. PC9 is inversion control for UART3. PC8 for 6
No inverter for UART1
I2C breakout is I2C2 hence is is shared with UART3
Also note PPM1 input is connected to UART6 RX prior to inversion.

digitalentity, 2017-11-07:
OMG this Pro Corner board is so messed up!

DzikuVx, 2017-11-07:
BTW, I've changed title of this issue since apparently it was refering to F3, not F4

digitalentity, 2017-11-07:
@DzikuVx thanks, I probably did a typo without noticing

MFlghts, 2017-11-14:
Hi, any news on fixing SBUS?
I've tested Naza GPS on UART1 - works fine.
What only bothers me that there are a lot (22715) errors at GPS statistic window and 3 timeouts. What does it mean?

MFlghts, 2017-12-02:
Hi, any news? Will it be fixed?

BFMav, 2017-12-21:
Hi I just got this board, need to use Sbus on Uart 3, I have an external telem on uart 1, I want to add Naza GPS on uart 6 but I need to disable the inverter, how do I do this ?

MFlghts, 2018-02-20:
It looks like developers doesn't want to support this board. Nothing changed for many months. This is strange attitude.

DzikuVx, 2018-02-20:
@MFlghts and how do you think developers should get few pieces of those boards? You think we should invest both our time and money buying things we do not really need?

fiam, 2018-02-20:
@MFlghts It's not that we don't want to support it, but time is limited and none of us have this board. That means that we'd need to write the code to support it "blindly" and ask users to test it on our behalf. Unfortunately, that's very time consuming and a pretty ineffective use of developer resources.

On the other hand, if you have the board and you want to add support for it in INAV, PRs for it would be welcome.

digitalentity, 2018-02-20:
@MFlghts None of the active developers has the board to test the fixes. As @fiam wrote developing blindly is a waste of developer time that could be used more effectively. 

Corner was designed for simple wiring for racing quads. Given that Corner board is most likely incapable of supporting I2C it will only be usable on airplanes where it won't be of much convenience because of form factor.

Where is this board listed as supported?

MFlghts, 2018-02-20:
@digitalentity 
it's listed here https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20Omnibus%20F4.md
I agree that this is not the best board, and it's better to leave it alone, and there is no any problem with this. The only problem is that some people bought the board, because it is in the list of the supported boards and now they have to trash their money because it's not going to be supported. This is quite tricky situation. Not very nice in fact.
I hope that it's not gonna happen with the other boards like Kakute F4 V2 or else.

stronnag, 2018-02-20:
It is not on the list of supported boardst; https://github.com/iNavFlight/inav/wiki/Supported-boards

That there is a board document is no endorsement of "supported".

MFlghts, 2018-02-20:
You are wrong, because your link leads to a text that refers to the file I mentioned, and it means that the board is supported. 

**See the docs folder for additional information regards to many targets in INAV, to example help in finding pinout and features.**

If it's not supported why the hell you have this string in docs??!
**Uses target OMNIBUSF4PRO**

Please don't make an idiot of me. If board is not supported it should not be mentioned anywhere in the docs with such a statement.

digitalentity, 2018-02-21:
Omnibus F4 Pro Corner **is** supported. Board boots and is flyable.

The only things that don't work are I2C (due to hardware design issues) and inverted S.Bus. Works fine with non-inverted S.Bus, Crossfire or IBUS.

If you feel that S.Bus support is mandatory - please open an issue specifically for that

MFlghts, 2018-02-21:
Without I2C and SBUS this board is useless for me. 
I think the board DOC file must be updated with the resent info, and this issue can be closed.
Thank you.

digitalentity, 2018-02-21:
@MFlghts While we might try fixing S.Bus inversion, I2C support is not possible on this board. Thus as I previously said - this board is not suitable for INAV anyway.

Indeed, we should amend the docs stating a warning that "Corner" doesn't have S.Bus inversion and I2C.

MFlghts, 2018-02-21:
In fact this board can be used for fixed wing, and even for multis with Naza GPS. But it also has some bad issues with noisy onboard dc-dc that affects imu, so I believe it's better to forget it.

digitalentity, 2018-02-21:
I'll submit a PR to amend the docs with a BIG WARNING about this board.

MFlghts, 2018-02-22:
Good. Thanks.

closed 2018-05-27 by stale[bot]

jstremmler, 2018-05-27:
Why should the Omnibus F4 Pro corner not work properly with iNav? I left SCL and SDA where iNav definies them to be (at UART3), connected the SBUS RX to UART 6 (works fine) and have connected the GPS to UART1......

jstremmler, 2018-06-09:
Try SBUS on UART1 and GPS on UART6!

Frank-Seider, 2018-09-17:
I have GPS on UART 1 and S-Bus on UART 6 and it works, but the UART 3 which I want to use for telemetry do nothing. Is it because I2C is on UART 3?  I try to use Softserial but no success. What can I do?