iNavFlight/inav #2464: BN-880 auto configuration issues

giacomo892, 2017-11-07:
##Goals
Enabling UBLOX7 protocol to obtain 10Hz update rate.

##Results
UBLOX7 is configured only "randomically", most of the time the unit is configured using the standard UBLOX configuration with POSLLH instead of PVT sentence.

#Used Hardware:
I'm testing a BN-880 GPS+GLONASS+GALILEO (3.01) on an OMNIBUS F3 board.

#First setup
UART @115200bps UBLOX7 chosen in the configurator.

Plug the battery, BN-880 start blinking at 1Hz, after seconds the buzzer kicks in. GPS ERROR (red icon) and timeouts (4) are reported in the GPS tab.
 
INAV then reconfigures it and then the unit starts to blink faster at 5Hz and no PVT sentence enabled. 

At this point to simulate a warm startup, I enter CLI and let the FC reboot. Again 90% of the time PVT sentence is not configured and UBLOX at 5Hz is set.

#Second setup

UART @57600bps UBLOX7 chosen in the configurator.

With this setup the buzzer no more kicks in and 40% of the times UBLOX7 with PVT sentence is enabled on boot again with the same delay.

At this point to simulate a warm startup, I enter CLI and let the FC reboot. Again 90% of the time PVT sentence is not configured and UBLOX at 5Hz is set.

#Tests done

I've edited to code and tried to stretch the timeouts in the unit initialization and I've to say that the rate of success has increased but nevertheless this increase init time of the unit and if it happens during flight can pose a serious threat to GPS modes.

#Possibile "solutions"

I've seen that the code does not check if the settings have being properly applied on the gps unit, maybe we should check them just to be sure.

P.S:

Just to say, I have two BN-800 (no mag) and they have being working flawlessy on inav with UBLOX7.

stronnag, 2017-11-07:
I have seemingly 100% success at 38400 baud on my BN-880; at least I've never noticed it not being at 10Hz.

giacomo892, 2017-11-07:
I still get 2 timeouts errors at 38400 but the success rate seemed increased. Tried 3 times in a row and at boot I got 10Hz.
Tried to enter CLI two times. First time FC rebooted and still was 10Hz. Second time it went back to 5Hz.
Would you mind done some trivial test in changing baud rate and entering/leaving CLI and check GPS tab? Thanks

stronnag, 2017-11-07:
Sure, I'll do some tests later this evening.

stronnag, 2017-11-07:
100% success for 10Hz at 38400, 56700, 115200 (5 power cycle tests at each speed). Excursions to the CLI have no effect (nor would I expect them to, setting UBLOX7 is a one-off boot process). 

Maybe you have marginal hardware somewhere in the loop, or some other setting on the GPS that makes it less reliable. I cannot fail it here.

giacomo892, 2017-11-07:
I have reset to default the module settings beside enabling GALILEO and disabling NMEA sentences which are anyway disabled by INAV.
Maybe something changed in newer BN-880? This one arrived 2 days ago from BG

For sure we can implement something to check if the configuration is really applied or not after the configuration process.

stronnag, 2017-11-07:
But so what? Navigation still works with UBLOX and 5Hz.

giacomo892, 2017-11-07:
Yes, at least it is working. But it is not the expected behaviour . INAV should check if the selected config (UBLOX7) is really applied. I will take a better look at the code and let's see what I can do.

stronnag, 2018-04-04:
Closing on no further activity / PR.

fiam, 2018-04-04:
I’m tempted to reopen this one and assign it to myself. I’m building a small wing specifically for testing/experimenting and I think the current code for GPS autoconfig is a bit fragile.