iNavFlight/inav #2467: No servo output from Omnibus f4 V2 with Inav

JonjoeMWalton, 2017-11-10:
I have iNav all set up in a new project plane.
 All wired up ect no problems. Plugged the lipo in, no magic pixy dust brilliant!
Board,VTX,Camera & ESCs all fire up.

Armed the model increase the throttle, both motors spin up! Yes.... but then I noticed something NONE of the servos plugged into the board are moving at all, only the flap servos that are plugged directly into the rx on ch7+8 bypassing the board are working.

Its strange! Iv'e double checked all the wiring its fine. The board arms fine. (The servos work as well I've checked)

I checked the motors tab in I nav and it shows the bars for the servos moving up and down when I am moving the sticks but nothing from the servos not even a peep!

I've checked the refresh rate and thats at 50Hz

Nothing at all from the servos! Anyone got any ideas! I'm using Omnibus F4 V2 pro

The 2 Aileron servos are powered from the ESC BEC and not the boards to reduce the load on the boards BEC but share the same battery and so have a common ground.

The Elev and Rud servos are powered directly from the board.

The Flaps are powered from the board via the rx they work fine
None respond

The ESC's respond so the PWM outputs must be working on atleast PWM1 and PWM2 and the flaps, RX & VTX work so the 5v and RAM rails all work so im lost.

Im thinking it must be somewhere in Inav I have a setting wrong. Any ideas?

JonjoeMWalton, 2017-11-10:
UPDATE:

If I power all the servos from the ESC BEC they all work, the board isn't putting 5v out to the PWM pins, any idea why? It is giving the rx 5v why not the pins. Have I missed something in Inav?

skaman82, 2017-11-10:
It is not safe to use the FC to power you servos anyway - power them through a separate 5v (3A) BEC.
Otherwise you risking a FC reboot in the air due to a brownout.

JonjoeMWalton, 2017-11-10:
Thabks for that :) Yes I have now. I didn't know the FC bec wasnt the best. It does show a noticable dip in voltage if used. But now all servos are on a 6A BEC.

Just having the problem with PWM 6 output theres voltage and ground but the servo just moves slowly in one direction no matter what input is given. (Its not the servo as if I swap the servo to the pwm3 it works fine)

skaman82, 2017-11-10:
What is your mixer type in the config tab? Can you post a screenshot?

skaman82, 2017-11-10:
If it is not clearly a bug with INAV, you should use support forums instead and ask there for help.

JonjoeMWalton, 2017-11-10:
Good evening,

As the power is sorted with BECs. 
I think it is something to do with INAV as PWM 6 just will not work, its just very slowly pulses in one direction no matter the input, Inav is telling me to connect it to the rudder to PWM6 however I have been told that PWM6 is programed for LED strip use? 

Any idea how I can get around this. Its driving me crazy! 
Is there a way I can prehaps create a custom mix and use PWM2 for rudder output instead?
Or disable LED output in the CLI on PWM6 so it can be used for the rudder as it should be.
People use these boards for fixed wing so there must be a way!

JonjoeMWalton, 2017-11-10:
So, I need to figure out how to remap PWM6 form LED strip to servo output.
I am unsure how to do this any ideas?

jstremmler, 2017-11-12:
Flash the correct firmware where the resource for LED_strip is on B6 and resource for PWM6 is on A8.

KayCox, 2018-02-04:
Same Problem on my side. Got Banggood Omnibus F4 Pro V2 and use it on an Mini Talon where PWM5 is used for the V-Tail Servo Mix. Need to use PWM6 for LED. Is there anything done? Please help.

jstremmler, 2018-02-05:
My Omnibus F4 Pro V3 has the LED-strip where it has been designed forfrom the board designers, on B06 and on pin 1 of connector J1. 
So PWM6 is on A8 and PWM5 is on A1 and therefore PWM5 can be used for a V-Tail Servo Mix......

gregbiv, 2018-05-07:
Hello there, I am having the same issue. The servo connected to PWM6 and I can see servo 6 moving on the motors page. However, the servo is not responding. I've connected 5v to PDB and also tested with servo tester and it's working. Might be related to #1587

vavisimo, 2018-09-29:
+1

i could need a servo output - pwm7 instead of the led on Airplane for flaps. As M1 motor, M2 cant be edited, M3 M4 M5 M6 for Left+Right A,E,R. And no flaps can be used by now when using x4sr as it has no pins like x8r, where you can directly connect the servo...to solve... would be nace to have one more PWM output

pediegoivan, 2019-04-11:
At least it would be great to move the channels AETR from channels 1-4 to 5-8 so you can free up the pwm outputs of your receiver for pwm servo controls and use the serial protocol for the AETR

wx4cb, 2019-04-11:
@vavisimo you can still use the pads on the x4rsb if you're using sbus, just bind it in 9-16 mode. that's exactly how i did it on one of my planes

pediegoivan, 2019-04-12:
I will look into that. Ah, you again! Thanks Bud.

closed 2020-02-19 by DzikuVx