iNavFlight/inav #2472: 3DR-Radio and Omnibus F4 Pro

jstremmler, 2017-11-12:
I have configured a 433MHz 3DR-radio with 57600 baud, configured MSP of UART1 on a SPRACINGF3 to 57600 baud, connected a 3DR-Radio ground station to my tablet with 57600 baud, establish the connection with EZGui and the FC and everything works perfectly.

**If I do exactly the same with the same !!! 3DR-radios and an Omnibus F4 Pro then both radios don't connect with EZGui. What the hell is the Omnibus F4 Pro doing differently compared with the SPRACINGF3 when connecting a 3DR-radio?** 

( I have the GPS on UART6 and the magnetometer with SCL and SDA on UART3 connected to the Omnibus which has the advantage that there is only one 6pin JST connector from the GPS to J10 on the Omnibus F4 Pro)

DzikuVx, 2017-11-12:
This board  by default shares PPM input with UART1 RX pin. You have to remove one jumper to separate them. https://quadmeup.com/uart1-and-ppm-on-airbot-f4-flip32-f4-flight-controllers/

closed 2017-11-12 by DzikuVx

jstremmler, 2017-11-12:
I tried that, remove the jumper and it did not work. Then I disconnected the GPS and tried it on UART 6... Did not work either....

DzikuVx, 2017-11-12:
The only thing I belive can be happeing in stat you have 3dr radio with inverted UART logic

Od: Dr. Juergen Stremmler
Wysłano: niedziela, 12 listopada 2017 18:09
Do: iNavFlight/inav
DW: Paweł Spychalski; State change
Temat: Re: [iNavFlight/inav] 3DR-Radio and Omnibus F4 Pro (#2472)

I tried that, remove the jumper and it did not work. Then I disconnected the GPS and tried it on UART 6... Did not work either....