iNavFlight/inav #2475: Sonar module I2C

Hydrologisst, 2017-11-12:
Hello evrybody,
Many new FC boards does not have 8 motor output pins.
So I'm searching sonar with I2C output.
And I'm found this one: [http://bit.ly/2ACnhzw]
But this module designed for PIXHAWK or APM controllers. 
One question it will work with INAV F4 FC (Matek F405-AIO) ?
Thanks for help.

digitalentity, 2017-11-13:
I'm not sure if we support this module. @DzikuVx I wonder if it's compatible with what you designed.

DzikuVx, 2017-11-13:
No it is not compatible . You might use this https://github.com/iNavFlight/INAV-Rangefinder-I2C-interface but you will need to program Attiny85 or Arduino to n between Inav and us100

Hydrologisst, 2017-11-13:
I'm not very good at programing, maybe it's possible to get prepared program code in notepad file?

DzikuVx, 2017-11-13:
Code is the link I gave you

sammichael25, 2017-11-13:
Would that code work on an arduino nano or pro micro? @DzikuVx

DzikuVx, 2017-11-13:
Pins would have to be modified, but besides that should be working

sammichael25, 2017-11-14:
Any chance you can add an edited version that could work with atmega 328p and 128p arduinos. :)

DzikuVx, 2017-11-14:
Rather slim in the next few months, busy on few other projects

Od: sammichael25
Wysłano: wtorek, 14 listopada 2017 18:52
Do: iNavFlight/inav
DW: Paweł Spychalski; Mention
Temat: Re: [iNavFlight/inav] Sonar module I2C (#2475)

Any chance you can add an edited version that could work with atmega 328p and 128p arduinos. :)

teckel12, 2017-11-21:
I have a US 100 on order from China. When I get it I'll interface it via i2c with an ATmega 328. While the chip is larger, it's easier to program (from an Arduino board) and more available than the ATtiny 85.

I've done quite a bit of Arduino programming specifically with audio and ultrasonic sensors.  So this dovetails well with my experience and INAV.

As the sensor is coming from China, I probably won't be able to start this project for 2-3 weeks.

Pokornz, 2017-11-24:
Hello, I just wired up an Arduino Nano and a bidirectional logic level shifter between the FC and US-100. I replaced the TinyWireS library with Wire library. I also took the watchdog and sleepmode stuff out of the code and put there delay(96) in the beginning of the loop. The Arduino seems to work (the LED lights up when distance < 10cm), but I am not able to get the iNav end to work. No matter which sonar I choose (HCSR04 or SRF10), the sonar icon is always red. I tried swapping the I2C wires on the FC, I even tried the alternative SCL/SDA position (I have an Omnibus F3).  Am I missing something in the configurator? Here is the code:
`/*
 * Decide on rangefinder type here. Only one type should be uncommented
 */
 #define USE_US100
 // #define USE_HCSR04

/*
 * Set I2C Slave address
 */
#define I2C_SLAVE_ADDRESS 0x14

#define DEBUG;

#define TRIGGER_PIN 2
#define ECHO_PIN 3
#define LED_PIN LED_BUILTIN

#define STATUS_OK 0
#define STATUS_OUT_OF_RANGE 1

#include <Wire.h>

#ifndef TWI_RX_BUFFER_SIZE
#define TWI_RX_BUFFER_SIZE ( 16 )
#endif

/*
 * Configuration magic, do not touch
 */
#ifdef USE_US100
  #define PULSE_TO_CM 59 //Multiplier pulse length to distance in [cm]
  #define MEASUREMENT_PERIOD_RATIO 6 // measurement rate = MEASUREMENT_PERIOD_RATIO * 16, 96ms in this case
  #define MAX_RANGE 300 //Range of 4 meters
#endif

#ifdef USE_HCSR04
  #define PULSE_TO_CM 58 //Multiplier pulse length to distance in [cm]
  #define MEASUREMENT_PERIOD_RATIO 5 // measurement rate = MEASUREMENT_PERIOD_RATIO * 16, 80ms in this case
  #define MAX_RANGE 300 //Range of 4 meters
#endif

#define PULSE_TIMEOUT (MAX_RANGE * PULSE_TO_CM) //this is an equivalent of 4 meters range

volatile uint8_t i2c_regs[] =
{
    0, //status
    0, //older 8 of distance
    0, //younger 8 of distance
};

const byte reg_size = sizeof(i2c_regs);

/*
 * 0=16ms
 * 1=32ms
 * 2=64ms
 * 3=128ms
 * 4=250ms
 * 5=500ms
 * 6=1 sec,
 * 7=2 sec,
 * 8=4 sec,
 * 9= 8sec
 */

volatile byte reg_position = 0;

/**
 * This function is executed when there is a request to read sensor
 * To get data, 2 reads of 8 bits are required
 * First requests send 8 older bits of 16bit unsigned int
 * Second request send 8 lower bytes
 * Measurement is executed when request for first batch of data is requested
 */
void requestEvent()
{  
  if (reg_position >= reg_size) {
    reg_position = 0;
  }
  
  Wire.write(i2c_regs[reg_position]);

  reg_position++;
}

void receiveEvent(uint8_t howMany) {

    if (howMany < 1) {
        // Sanity-check
        return;
    }

    if (howMany > TWI_RX_BUFFER_SIZE)
    {
        // Also insane number
        return;
    }

    reg_position = Wire.read();

    howMany--;

    if (!howMany) {
        // This write was only to set the buffer for next read
        return;
    }

    // Everything above 1 byte is something we do not care, so just get it from bus as send to /dev/null
    while(howMany--) {
      Wire.read();
    }
}

void setup() {
  pinMode(TRIGGER_PIN, OUTPUT);
  pinMode(ECHO_PIN, INPUT);
  pinMode(LED_PIN, OUTPUT);

  /*
   * Setup I2C
   */
  Wire.begin(I2C_SLAVE_ADDRESS);
  Wire.onRequest(requestEvent);
  Wire.onReceive(receiveEvent);

}

long microsecondsToCentimeters(long microseconds){
//  return microseconds / PULSE_TO_CM;
  return (microseconds * 34 / 100 / 2) / 10;
}

void loop() {

  delay(96); 

  /*
   * Measurement is done every 6th wakeup, that gives more less 10Hz update rate (96ms)
   */
    digitalWrite(TRIGGER_PIN, LOW);
    delayMicroseconds(2);
    digitalWrite(TRIGGER_PIN, HIGH);

    delayMicroseconds(10);
    digitalWrite(TRIGGER_PIN, LOW);

    long duration = pulseIn(ECHO_PIN, HIGH, PULSE_TIMEOUT);

    if (duration > 0) {
      i2c_regs[0] = STATUS_OK;
    } else {
      i2c_regs[0] = STATUS_OUT_OF_RANGE;
    }
    
    uint16_t cm = (uint16_t) microsecondsToCentimeters(duration);

#ifdef DEBUG
    if (cm <10) {
      digitalWrite(LED_PIN, HIGH);
    } else {
      digitalWrite(LED_PIN, LOW);
    }
 #endif
    
    i2c_regs[1] = cm >> 8;
    i2c_regs[2] = cm & 0xFF;
    
  
}`

teckel12, 2017-11-24:
@Pokornz In general, this code and your conversion is not the best way to approach this. I've created the NewPing sonar library which would work better and be more suited to this task.

I'll be reworking this code to work better and on the ATmega 328/Arduino. Waiting for my US100 sensor.

My guess is that your problem is the voltage. The sensors you're using are 5v but the FC is 3.3v. You really should be using logic level converters for this task. Also, you may need resistors for i2c communication. 

In any case, this should all be sorted out when I upgrade this code to use better sonar control. I would suggest getting a US100 sensor as that will make things MUCH easier once my code update happens.

Pokornz, 2017-11-24:
As I wrote, I am using logic level shifters and I am using US-100.
What is not good about the conversion? What can I improve? (except for deleting the comments that are not relevant)
And is iNav supposed to automatically recognize that there is an I2C sonar connected, or do you have to enable something? As I wrote, I tried both sonar options in the configuration tab and neither of those worked.

teckel12, 2017-11-24:
@Pokornz Sorry, your message was hard to follow as half the code was shown as text.

The code in general (not what you added) could be greatly improved, some just doesn't make much sense. But I've been the author of the sonar NewPing library for Arduino for many years, so it's evolved quite a bit over this fairly simplistic and prone to errors methods. 

In any case, I need to do a refresh of this code,  but without the US100 I'm not even going to bother because it's not very realistic to write the code and perfect the hardware connection to be easily reproducible without a sensor that can be connected without multiple power supplies and logic level converters.

Like I said, give me a couple weeks, I need the hardware to arrive. If you get it working, that's great, but it will be a lot easier and cleaner once I give it a work over.

Pokornz, 2017-11-25:
Ok, good. But I am sort of time pressed (my bachelors thesis is based on the quadcopter) :D. Do you know, if some settings in iNav needs to be changed in order for the FC to know that there is an I2C sonar instead of a regular one? I can't find that information and it is missing in the documentation.
In the inav/src/main/drivers, there are files rangefinder_hcsr04 and rangefinder_hcsr04_i2c.

DzikuVx, 2017-11-25:
@Pokornz @teckel12 US100 is 5V hardware, our FCs are 3.3V. 
When I was designing https://github.com/iNavFlight/INAV-Rangefinder-I2C-interface I decided that keeping the logic on 3.3V and protecting uC inputs with inline resistors is the best option. It can be either ATtiny running 3.3V or Arduino with 3.3V  logic. That is just cheaper and smaller than logic level shifters since you can not use resistors to interface I2C (Problems are pullups, if they can not be easily disabled on 5V device, it will cause problems.)

You can buy 3.3V arduinos for cheap. Like here: https://goo.gl/Dq8wC3 The best option, period ;)

CLI `rangefinder_hardware` `HCSR04I2C` is the answer to how to configure INAV to use I2C interface

As long as you will keep interface consistent with https://github.com/iNavFlight/INAV-Rangefinder-I2C-interface you can connect any rangefinding hardware this way:
3 bytes for each packet
byte 0: status, 0 = OK, 1 = out of range
byte 1: 8 older bits of range in cm
byte 2: 8 younger bits of range in cm

Pokornz, 2017-11-25:
Thank you @DzikuVx . The command should be mentioned in the notes of README.md I think ;)

I understand that getting a 3.3V microcontroller is the best, but I have bunch of Arduino Nanos laying around and I don't have a month to spare for waiting for China shipments.

I changed the rangefinder_hardware, but it still doesn't recognize the sonar.
Here is the scheme: 
[image]
Is there something wrong, or am I missing something in the code? I know that this may not be an issue with iNav, but any help is very appreciated :)

DzikuVx, 2017-11-25:
Ammm... Looks good, hard to tell. Maybe 5here is something with i2c slave implementation. Never tested it with atmega32u4

Pokornz, 2017-11-26:
Strange thing is that the LED on the Arduino also turns on when the distance is about 70 cm and higher..

teckel12, 2017-11-27:
@DzikuVx The US-100 and the ATmega 328 will run at 3.3 or 5v, so no logic level conversion is needed.

After I get my US-100 I'm going a different route with the ultrasonic control and wiring to keep it simple, faster, and more reliable. At least that's the plan.

Pokornz, 2017-11-27:
@teckel12  Are you telling me that the Arduino Nano (5V) can be directly connected to the FC (Omnibus F3) without any danger for the STM32F3? You should look at the diagram, the conversion is NOT between ATmega 328 and US-100.

teckel12, 2017-11-27:
@Pokornz I'm not speaking of the Nano at all. If I used any pre-made ATmega board it would be the mini due to the size, cheapness and ability to run at 3.3v.

As far as your problem, as you sure you're connecting to the I2C pins on your FC? Aren't they next to the RSSI and current pins?

Pokornz, 2017-11-27:
@teckel12 The iNav documentation says it is on the PWM7/8 pins: https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20OMNIBUS.md
I tried the ones next to the RSSI pin as well but to no avail.

Pokornz, 2017-11-29:
@DzikuVx  So I wanted to make a super simple test code to see if the communication between the FC and Arduino works. I took a look at[ what you wrote](https://github.com/iNavFlight/inav/blob/master/src/main/drivers/rangefinder_hcsr04_i2c.c) and at [how the i2cRead works](https://github.com/cleanflight/cleanflight/blob/1fccf48d64b31ca0f6686ad97da434ebc806e7cb/src/main/drivers/bus_i2c_stm32f30x.c) and came to this very specific code (it works till you change the rangefinder_hcsr04_i2c.c :D) that should be reporting 50cm of altitude. Could you please take a look at it to confirm that I understand it correctly?
~~~~c
#include <Wire.h>

volatile uint8_t i2c_regs[] ={0,0,50};
volatile byte reg_position = 0;

void receiveEvent(uint8_t howMany) {
  reg_position = 0;
  while(howMany--) 
  {
    Wire.read();
  }

}

void requestEvent()
{
  Wire.write(i2c_regs[reg_position++]);
}

void setup() {
  Wire.begin(0x14);
  Wire.onRequest(requestEvent);
  Wire.onReceive(receiveEvent);

}

void loop() {}
~~~~

Pokornz, 2017-12-04:
So I tried it (on Arduino Mega). I added a line of code to light up the built-in LED in the beginning of the receiveEvent, so if the Arduino EVER gets ANY data, the LED stays on and I know that at least that works. 
Unfortunately, it didn't do anything. Never lit up, nothing.

teckel12, 2017-12-04:
I got my US-100 today.  So I'll try interfacing with it in the next couple days.

DzikuVx, 2017-12-04:
I will test if original design using ATtiny85 still works with latest INAV.

DzikuVx, 2017-12-06:
OK, I know why it is not working for you. This hardware is enabled only on Ombus F4 and Airbot F4 boards. I will make a PR to enable it on other tagets too

DzikuVx, 2017-12-06:
Any specific target you request @Pokornz ?

Pokornz, 2017-12-06:
I have Omnibus F3, so OMNIBUS. Thank you.

sammichael25, 2017-12-06:
I’d like to test it on matek f405 aio target

DzikuVx, 2017-12-06:
@Pokornz @sammichael25 take a look at #2560 Matek F405 also added to the list

Pokornz, 2017-12-06:
I'm new to this, so I'm not sure how it works. Do I have to wait for the next version (release) of iNav? Or can I somehow compile it on my own?

DzikuVx, 2017-12-07:
@Pokornz there are developer docs in repository that pretty nicely descibes how to setup everything:

https://github.com/iNavFlight/inav/tree/master/docs/development

nerezw, 2017-12-16:
I came across one of these in my pile of un-finished projects http://wiki.sunfounder.cc/index.php?title=I2C-GPS_NAV_Navigation_Board .

DzikuVx, 2017-12-16:
Ah, good I was reminded. Just tested https://github.com/iNavFlight/INAV-Rangefinder-I2C-interface with INAV 1.8 on Omnibus F4 board and it is working perfectly fine

gstyle97, 2018-02-22:
Hi,
I'm new and have a lot of questions.

I will connect sonar with FC Matek F405 STD with iNav 1.9 downloaded from 

nav_1.9.0_MATEKF405-Softserial   Softserial1_TX on TX4 pad, softserial1_RX on RX4 pad

If  sda/scl and rx4/tx4 are used by GPS bn880, 
how can i connect the sonar HC SR04 

I found this: https://github.com/cleanflight/clean.../docs/Sonar.md

Now i need to know if and where i can do the same on the f405std board

And if it need this too : 
https://github.com/iNavFlight/INAV-R...ster/README.md

Can i use softserial? And if i can on what sort of pins?

Thanks

DzikuVx, 2018-02-23:
I2C allows to connect multiple devices in parallel, this is not a problem. SoftSerial can not be used and matek 405 does not have pads to connect HCSR04/US100 directly

Pokornz, 2018-02-23:
I'm not sure how the development process works, but I assumed that since @DzikuVx did PR to enable this hardware on other targets, the changes will show up in the next release. I just checked 1.9 and it is not there..
[image]
[image]

gstyle97, 2018-02-23:
Thank for so quick replay.
So if i have understand, i can use the sda /scl pad, in parallel with the compass of my gps.
I assume that i have to use 1k ohm over trigger and echo without attiny85?
If yes i will test it this morning

gstyle97, 2018-02-27:
Ok, that don't work without attiny85 mod, now i'm waiting for it and the usbasp. Locking at the mod i'm asking me of i had to inserto the 16v 4,7uf when i use the 3,3v from the flightcontroler directrly

gstyle97, 2018-03-10:
hi, i've programmed the tiny85, i've make like it's proposed in the doc, but it's not work in inav the sonar icon is red.

I've changed only the tiny libraries and this is the ardinuo output

`Warning: Board tiny:avr:attiny84at16 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY84AT16
Warning: Board tiny:avr:attiny85at128 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY85AT128
Warning: Board tiny:avr:attiny2313at8 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY2313AT8
Warning: Board tiny:avr:attiny4313at1 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY4313AT1
Warning: Board tiny:avr:attiny44at8 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY44AT8
Warning: Board tiny:avr:attiny84at1 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY84AT1
Warning: Board tiny:avr:attiny45at1 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY45AT1
Warning: Board tiny:avr:attiny2313at1 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY2313AT1
Warning: Board tiny:avr:attiny85at16p doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY85AT16P
Warning: Board tiny:avr:attiny24at16 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY24AT16
Warning: Board tiny:avr:attiny4313at8 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY4313AT8
Warning: Board tiny:avr:attiny44at1 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY44AT1
Warning: Board tiny:avr:attiny25at8 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY25AT8
Warning: Board tiny:avr:attiny85at1 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY85AT1
Warning: Board tiny:avr:attiny25at1 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY25AT1
Warning: Board tiny:avr:attiny45at8 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY45AT8
Warning: Board tiny:avr:attiny85at16e doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY85AT16E
Warning: Board tiny:avr:attiny84at8 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY84AT8
Warning: Board tiny:avr:attiny85at8 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY85AT8
In file included from C:\Users\PhotoStudio\AppData\Local\Arduino15\packages\tiny\hardware\avr\1.0.6\cores\tiny/Stream.h:24:0,

                 from C:\Users\PhotoStudio\AppData\Local\Arduino15\packages\tiny\hardware\avr\1.0.6\cores\tiny/TinyDebugSerial.h:31,

                 from C:\Users\PhotoStudio\AppData\Local\Arduino15\packages\tiny\hardware\avr\1.0.6\cores\tiny/WProgram.h:18,

                 from C:\Users\PhotoStudio\AppData\Local\Arduino15\packages\tiny\hardware\avr\1.0.6\cores\tiny/Arduino.h:4,

                 from sketch\inav_i2c_sonar.ino.cpp:1:

C:\Users\PhotoStudio\AppData\Local\Arduino15\packages\tiny\hardware\avr\1.0.6\cores\tiny/Print.h:37:0: warning: "BIN" redefined [enabled by default]

 #define BIN 2

 ^

In file included from c:\users\photostudio\appdata\local\arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5\avr\include\avr\iotn85.h:38:0,

                 from c:\users\photostudio\appdata\local\arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5\avr\include\avr\io.h:428,

                 from c:\users\photostudio\appdata\local\arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5\avr\include\avr\interrupt.h:38,

                 from C:\Users\PhotoStudio\AppData\Local\Arduino15\packages\tiny\hardware\avr\1.0.6\cores\tiny/WProgram.h:8,

                 from C:\Users\PhotoStudio\AppData\Local\Arduino15\packages\tiny\hardware\avr\1.0.6\cores\tiny/Arduino.h:4,

                 from sketch\inav_i2c_sonar.ino.cpp:1:

c:\users\photostudio\appdata\local\arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5\avr\include\avr\iotnx5.h:55:0: note: this is the location of the previous definition

 #define BIN     7

 ^

Lo sketch usa 2340 byte (28%) dello spazio disponibile per i programmi. Il massimo è 8192 byte.
Le variabili globali usano 62 byte (12%) di memoria dinamica, lasciando altri 450 byte liberi per le variabili locali. Il massimo è 512 byte.
C:\Users\PhotoStudio\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/bin/avrdude -CC:\Users\PhotoStudio\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/etc/avrdude.conf -v -pattiny85 -cusbasp -Pusb -Uflash:w:C:\Users\PHOTOS~1\AppData\Local\Temp\arduino_build_931789/inav_i2c_sonar.ino.hex:i 

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\PhotoStudio\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/etc/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : ATtiny85
[... 35 lines cut ...]
Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e930b (probably t85)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: auto set sck period (because given equals null)
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
[... cut ...]
avrdude: verifying flash memory against C:\Users\PHOTOS~1\AppData\Local\Temp\arduino_build_931789/inav_i2c_sonar.ino.hex:
avrdude: load data flash data from input file C:\Users\PHOTOS~1\AppData\Local\Temp\arduino_build_931789/inav_i2c_sonar.ino.hex:
avrdude: input file C:\Users\PHOTOS~1\AppData\Local\Temp\arduino_build_931789/inav_i2c_sonar.ino.hex contains 2340 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 1.31s

avrdude: verifying ...
avrdude: 2340 bytes of flash verified

avrdude done.  Thank you.`

gstyle97, 2018-03-10:
perfect, i've written tiny85 in 8mhz, all perfect, inav will reconozise the sonar, but in sensor the sonar is not work or better there are no input, what can it be?

Pokornz, 2018-03-11:
Did you check that the communication between attiny and sonar works?

gstyle97, 2018-03-11:
How can i do it?

Pokornz, 2018-03-11:
I'd suggest to find a way to report the measured distance back to the compute, so you can see if the data is relevant, or you can attach an LED and set a condition that it lights up for ex. when the distance is <10cm.

gstyle97, 2018-03-11:
Good idea but i don't know how i can attach the led, figure if i know how Maker the condition for the distance 😭😭😭

gstyle97, 2018-03-12:
Can anybody help me with this problem?
The tiny chip is reconocized by inav 1.9 for matek405, but the hc sr 04 don't vive any out put, in sensor tap it is a flat line on 0. The sensor is new.

Pokornz, 2018-03-12:
As I said above, first check that the attiny+sonar part is working. Do a test code (without the I2C communication) to make sure that the problem is not there. If you don't know how to attach an LED or a serial monitor to attiny, look it up.

gstyle97, 2018-03-19:
i've only USBASP progrmmer, so i can't do nothing else as program the attiny85, now my question is if i 've to change the original file like PIN : inav_i2c_sonar ?
i use th attinycore 1.1.6 and wire.s library
i new in this, and so i think that without any help i'm wasted

Sylvania2, 2018-03-30:
Hi

Can't get I2C Rangefinder to work with OMNIBUS F7 V2.  So i used a second  arduino to test the i2c sonar. Serial port output is distance in cm, and the rangefinder does work, just not on OMNIBUS F7 V2.
Any ideas ? 

`#include <Wire.h>

void setup() {
  Wire.setClock(400000L); 
  
  Wire.begin();        // join i2c bus (address optional for master)
  Serial.begin(9600);  // start serial for output
  Serial.print("I2C TEST");
}

void loop() {

  Wire.beginTransmission(0x14);
  Wire.write(2);              // sends one byte
  Wire.endTransmission();    // stop transmitting 
  
  Wire.requestFrom(0x14, 1);    // request 1 bytes from slave device #14

  while (Wire.available()) { // slave may send less than requested
    unsigned char c = Wire.read(); // receive a byte as character
    Serial.print(c, DEC);         // print the character
    Serial.print(" ");
  }
     Serial.print(" \n\r");
  

  delay(500);
}`

Pokornz, 2018-03-31:
@Sylvania2 It has to be set up as a slave with a specific address. Read the code carefully: https://github.com/iNavFlight/INAV-Rangefinder-I2C-interface/blob/master/inav_i2c_sonar.ino

Sylvania2, 2018-03-31:
@Pokornz yes the sonar i a slave, i made a master to test the slave. And my sonar i working, replying xxx cm to master as it should. Problem is that sonar i2c is not detected when connected to OMNIBUS F7 V2. SONAR icon stays red. Sda/scl have pull Up resistors to 3.3v, I2C clock changed to 100KHZ but No luck.

Pokornz, 2018-03-31:
Have you enabled it in iNav? CLI: rangefinder_hardware HCSR04I2C

Sylvania2, 2018-03-31:
@Pokornz i only enabled it in user interface :-/ but in CLI i get: 

# rangefinder_hardware HCSR04I2C
Unknown command, try 'help'
#

Sylvania2, 2018-03-31:
@Pokornz sorry, cli: set rangefinder_hardware = HCSR04I2C returns:
rangefinder_hardware set to HCSR04I2C

Sylvania2, 2018-03-31:
Did some measurements with oscilloscope.
Disable Mag (HMC5883 dont have it anyway) and Enable HCSR04I2C, no sck pulses.
enable Mag HMC5883 and Disable HCSR04I2C, sck pulses for a while, no hardware of cource.
enable Mag HMC5883 and enable HCSR04I2C, sck pulses for a while, Rangefinder not found, so is only trying to find MAG, not HCSR04I2C.
So could there be some bug in code not setting correct i2c port for the Rangefinder ?

Pokornz, 2018-03-31:
@DzikuVx The clock should be always ticking when HCSR04I2C is set as the rangefinder_hardware, right?

Sylvania2, 2018-03-31:
i guess so. I dont have the Mag HMC5883 yet so of cource FC stops asking but no scl / data with HCSR04I2C mounted and enabled tells me there is a bug in OMNIBUSF7V2

juanquy, 2018-04-02:
Hi, guys, I have this old board,
https://drotek.com/shop/en/home/335-i2c-gps-nav-module.html
It is basically a Atmel 328p, wondering if it can be used to connect a us-100 to a SP Racing F3 via I2C , I have the sonar already working on pin 7,8 on my SP Racing but I need those pins cuz I'm using pwm not sbus or cppm. Anyone with the code so I can flash it and try?
[image]
l

regards 
JC

Pokornz, 2018-04-02:
@juanquy You will have to take the [code](https://github.com/iNavFlight/INAV-Rangefinder-I2C-interface/blob/master/inav_i2c_sonar.ino) and change it in a similar way as I already did in this thread (the post with a looong code).

[33 more replies]