iNavFlight/inav #2476: Pendulum swing with GPS alt/pos hold on

VoicOfReason, 2017-11-14:
I apologize if this issue would be better addressed in a different forum, but I trust the expertise of those here. I've tried many different options but I'm getting a pendulum type swing the slowly comes to a stop when moving around and then stopping with GPS pos/alt hold both turned on. I first noticed this last summer on an iNav quad I built and never fixed, giving up on solving the issue for awhile. Now, I have a FC without an internal mag (Omnibus F4 V3). It's not really a toilet bowl, but it just sort of swing leisurely back and forth a bit before coming to a stop. Once it stops, it hovers in place fairly well. I'm wondering if this is a bug or if there is a specific PID setting that needs to change since I've been messing with all of them with no success. Thanks.

KenImhof, 2017-11-14:
I had a similar problems with the F4 FC affecting the Magnetometer in the GPS module.  I would test the Mag operation on a bench by checking proper Mag readings at 0,90, 180, and 270 degrees.  Also make sure when rotating the quad, that the mag reads and properly tracks the quad yaw rotation and immediately stops when the rotation stops.   I was able to fix this in both of my quads by isolating the Mag from the FC.  In one of the quads I was able to isolate the GPS using a brass shield. I believe this is the best approach as brass has no magnetic properties.  In the other quad I was able to isolate the FC by placing a mumetal shield over the FC, but since mumetal is magnetic the GPS with MAG has to be located about 15mm above the shield.

VoicOfReason, 2017-11-14:
Could you show me where to find a brass shield? Picture?

VoicOfReason, 2017-11-15:
Let's postulate that it isn't a mag issue since my mag is pretty far from the FC. What specific inav PID settings would I look at to get my quad to lock in its position quicker during hover (RTH is working fine)?

digitalentity, 2017-11-15:
Toilet-bowling effect is in 99% cases a compass issue. Please verify that when you point your quad North heading shows 0 deg, East - 90 deg, South 180 deg, West 270 deg and values don't change more than 5 deg when you pitch/roll the machine up to 45 deg.

VoicOfReason, 2017-11-15:
Yes, that is all correct in terms of degrees based on direction. It's actually NOT a toilet bowl. It is just sluggish in terms of stopping itself when moving around with alt/mag hold both turned on. At worst, it is a pendulum (left and right swinging until coming to rest). I had the toilet bowl last summer and I never solved that problem until switching FC's. However, in this case, I'm getting a solid hover in all directions even in wind, but if I MOVE around with alt/pos hold on, then it sways before coming to rest at its new position. Hope this is clear now what my issue is. I'm looking for specific settings to look at, e.g. set nav_mc_vel_xy_i = 45, etc.

digitalentity, 2017-11-15:
@VoicOfReason Ah, I see. How big is your machine? Swinging might be an indication that the code is expecting the machine to do very rapid corrections which it is unable to satisfy. In that case you may try reducing PosXY P and VelXY P

VoicOfReason, 2017-11-15:
Excellent. Will try.

rennster12, 2017-11-22:
I'm having the same symptoms. iNAV ona 250 sized quad with NEO8N GPS good sat lock 18-19 sats. In fast forward flight it flares, goes back to approximate where I let off on the pitch and swings around to the point. It isn't as solid and rock steady as a Phantom 3 pro. I read the doc on PosPIDs and Vel PIDs but I don't understand the relationship and what effects they have. I'd appreciate any insight. Mahalo!

VoicOfReason, 2017-11-22:
I’m still struggling to fix this but I’ve made a little progress adjusting Various P values. Let me know what progress your making...

giacomo892, 2017-11-22:
What looptime are you running? @VoicOfReason @rennster12

rennster12, 2017-11-23:
looptime at 2000
My hardware is a 250 sized quad with a FLIP32 AIO. I believe it is an F1 chip. I understand it is dated.
Here is the complete dump:
# INAV/NAZE 1.6.1 Mar  9 2017 / 00:36:58 (6c878ae)
mixer QUADX

mmix reset

servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1
smix reset

feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature RX_PPM
feature VBAT
feature GPS
feature CURRENT_METER
feature DASHBOARD
feature PWM_OUTPUT_ENABLE
beeper GYRO_CALIBRATED
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACTION_SUCCESS
beeper ACTION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
beeper ON_USB
beeper LAUNCH_MODE
[... 256 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 120
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 65
set pitch_rate = 65
set yaw_rate = 45
set tpa_rate = 0
set tpa_breakpoint = 1500

I noticed significant improvement when I used the 250 preset and set GPS to 50 delay. So far it is better than the large toilet bowling when I released the pitch.

danylam, 2018-04-12:
Has anybody found a perfect solution for this issue? I have this with a 330 quad and a kakutef4. 
Thanks

VoicOfReason, 2018-04-12:
I'm still having the issue.

geodetizca, 2018-04-26:
hiy guys, I have a pretty good news for this so called pendulum swing or toilet bowl effect or circular drift etc... I got my quad with these specs:
- Eachine Falcon 250 Pro frame
- MT2204 2300kv motor
- LittleBee20A OPTO esc
- SP F3 DELUXE (completed with Baro and Compass)
- GPS UBLOX Neo M8N (I just take GPS data)
- Flysky i6 Rx-Tx
- 5" props

FIRMWARE: INAV 1.8

I also struggled to make it stable on PosHold and RTH. I recalibrate compass and acc just to see what progress I have but it stays the same, :( then I realize that I really need to pay attention to my PID.

I input these PID settings:
Roll 30 20 70
Pitch 30 20 70
Yaw 100 40 0

PosXY  65 120 -
VelXY 180 15 5
Surface 10 5 8

ROLL, PITCH and YAW rate: 65

Modes: ANGLE for POSHOLD and RTH. And ALTHOLD combined with POSHOLD.

And all of the sudden, I got what I'm dreaming for. :) A rock steady GPS 250 quad for both Poshold and RTH. Now you try guys and please let me know.