iNavFlight/inav #2477: OSD switch not working Omnibus F4 pro (V2)

MW2Stambo, 2017-11-14:
# INAV/OMNIBUSF4PRO 1.7.3 Sep 9 2017 / 10:35:55 (974440df)

I have OSD enabled on the configuration tab and I have set up the OSD SW in the modes tab and see it change blue when i flick the switch on my tx but it makes no difference, the OSD stays on in either switch position.
Am I doing something wrong?

I will update to latest build if that is what is needed but I am a bit new to INav and worry about losing settings and having to start again.

digitalentity, 2017-11-15:
I believe we don't handle `BOXOSD` at all. @fiam I think we should look at using `BOXOSD` for turning OSD on/off. What do you think?

ShawnzAZ, 2017-11-18:
Could we have the OSD switch just switch to another profile like how MWOSD does it?  Then you could choose to either disable all the screen elements or switch to a minimal interface.   Other options could be those map / radar modes.

csurf, 2017-12-08:
+1 vote for alternate OSD layouts and alternate map/radar.

Would be ideal to implement many of the MWOSD features which can be useful for long-range/'nav' FPV (versus short-range 'racing/acro' FPV which doesn't really require a complex OSD or alternate layouts).

Do we have more than enough resources on F3 & F4 for this?

digitalentity, 2018-02-12:
Fixed.

PR #2676 (2018-02-01): Implement BOXOSD mode used to disable OSD display
Fixes #2477 while waiting for OSD layouts to be implemented.
closed 2018-02-12 by digitalentity