iNavFlight/inav #2481: Revolution : how to configure Servo output

ValCx, 2017-11-15:
Hi!
I'm trying to enable pwm servo output 5 or 6 to control pitch of my gimbal. Is it possible to enable it on the revolution board or it still not working?

Thank you

Valentin

ricardovalmeida, 2017-12-29:
hi dear friend.
I also have the same problem with my cc3d revolution.
I can not power the servo motor by pwm, I want to use two servos for gimbal and they are not working, I'm using the 1.8 inav version.

digitalentity, 2017-12-31:
You want to use S5 & S6 outputs as servo outpus on a quad?

ricardovalmeida, 2017-12-31:
HI dear friend.
Yes, I want to control a servo gimbal.
same as this link.
https://pt.aliexpress.com/store/product/700-Line-CCD-HD-Portable-Mini-Camera-Dual-Axis-Servo-Gimbal-All-In-One-Machine-include/2008011_32669499086.html
thank you.

CopterFail, 2018-01-24:
Hi,
I've tried a Tricopter with two yaw servos. Servo output was not possible. (The internal magnetometer was also not detected - already mentioned in another thread)

digitalentity, 2018-03-01:
@ricardovalmeida @ValCx It may be possible to add servo outputs to S5 and S6 of REVO. Could you please test the attached hex? If it work for you - we'll make this change official 😄 
@CopterFail unfortunately there's only 1 servo supported for TRI/CUSTOMTRI mixer.

ricardovalmeida, 2018-03-03:
hi dear friend
Thanks for your help.
I just did the test with the servo, and this is working the signal output for the servo motor, gimbal
I did not do other tests of the general functioning of the board. Only the servo gimbal. (it's raining a lot in my city today Ponta Grossa - Parana - Brazil)
Thanks for your help.

closed 2018-03-04 by digitalentity